zephyrproject-rtos/zephyr #27843: spi_nor.c: Wrong buffers for rx_set

yashi, 2020-08-27:
**Describe the bug**
`spi_nor_access()` in `spi_nor.c` is [currently](https://github.com/zephyrproject-rtos/zephyr/blob/master/drivers/flash/spi_nor.c#L256) like this
```C
static int spi_nor_access(const struct device *const dev,
			  uint8_t opcode, bool is_addressed, off_t addr,
			  void *data, size_t length, bool is_write)
{
	struct spi_nor_data *const driver_data = dev->data;

	uint8_t buf[4] = {
		opcode,
		(addr & 0xFF0000) >> 16,
		(addr & 0xFF00) >> 8,
		(addr & 0xFF),
	};

	struct spi_buf spi_buf[2] = {
		{
			.buf = buf,
			.len = (is_addressed) ? 4 : 1,
		},
		{
			.buf = data,
			.len = length
		}
	};
	const struct spi_buf_set tx_set = {
		.buffers = spi_buf,
		.count = (length) ? 2 : 1
	};

	const struct spi_buf_set rx_set = {
		.buffers = spi_buf,
		.count = 2
	};
[...]
	return spi_transceive(driver_data->spi,
		&driver_data->spi_cfg, &tx_set, &rx_set);
}
```
Because `spi_buf` is given to `rx_set` with `count` == 2, `spi_transceive()` is expected to use both `buf` and `data` to receive data.  But, this is wrong because callers are expecting to have the received data in `data`, which is `spi_buf[1]`.  I'm testing with out-of-tree board and fixing it like the following made it work on my boards.

```diff
diff --git a/drivers/flash/spi_nor.c b/drivers/flash/spi_nor.c
index c83ef2b071..a34dc0f498 100644
--- a/drivers/flash/spi_nor.c
+++ b/drivers/flash/spi_nor.c
@@ -163,8 +163,8 @@ static int spi_nor_access(const struct device *const dev,
 	};
 
 	const struct spi_buf_set rx_set = {
-		.buffers = spi_buf,
-		.count = 2
+		.buffers = &spi_buf[1],
+		.count = 1
 	};
 
 	if (is_write) {
```

What I don't understand is that there seems to be several boards using this device driver.  With this bug, I assume non should be working.

Am I reading it wrong?

pabigot, 2020-08-27:
> Because `spi_buf` is given to `rx_set` with `count` == 2, `spi_transceive()` is expected to use both `buf` and `data` to receive data.

Yes.

>  But, this is wrong because callers are expecting to have the received data in `data`, which is `spi_buf[1]`.

No.

The way Zephyr's SPI transceive works is rather confusing.  In short `spi_transcieve` takes two buffer sets, each of which must contain fragments that combine to be the total length of the transfer.  It simultaneously transmits data from the `tx_bufs` while reading data into the `rx_bufs`, since SPI is a full-duplex protocol.

`spi_buf` is used to encode the two fragments necessary for a SPI NOR read or write:
* the 4 bytes of the opcode and offset in the stack `buf`
* the `length` bytes from/to the user at `data`

When `is_write` is set a helper function is used that transmits the data from both fragments, and discards whatever is simultaneously read.

When `is_read` is cleared then the transcieve operation transmits the address from `buf`, and reads the responses (which are probably zeros or at least are not data) back into `buf`.  This works because the data we needed was already used so the overwrite doesn't matter.

Then transceive writes out whatever was in `data` while storing what's received into `data`.  This works because the SPI NOR device doesn't care what gets written to it when it's executing a read, and the actual data it's providing gets stored into `data` as expected.

Does this make sense?  If you had to make the change you show to get the right behavior it sounds like the SPI driver for your out of tree board is implemented incorrectly.

yashi, 2020-08-28:
> Does this make sense?

Yes

> If you had to make the change you show to get the right behavior it sounds like the SPI driver for your out of tree board is implemented incorrectly.

That sounds right. I'll check my SPI driver.  Thanks!

yashi, 2020-08-28:
BTW, does it mean the doc is rather wrong?

It reads:

```
tx_bufs: Buffer array where data to be sent originates from, or NULL if none.
rx_bufs: Buffer array where data to be read will be written to, or NULL if none.
```

https://docs.zephyrproject.org/latest/reference/peripherals/spi.html#_CPPv414spi_transceiveP6devicePK10spi_configPK11spi_buf_setPK11spi_buf_set

pabigot, 2020-08-28:
No, it's not wrong, it's just ambiguous.  It could be more clear that every byte transfer bidirectional: the output data comes from `tx_bufs` (or NULL to send zeros) and the input data is stored in `rx_bufs` (or NULL to discard).

closed 2020-09-01 by yashi

rweickelt, 2022-02-05:
I am glad I found this thread and @pabigot 's explanation. The actual implementation is sensible, but the documentation of this function is (still) misleading.