zephyrproject-rtos/zephyr #27845: Support other kind of SPI NOR flash in spi_nor.c

yashi, 2020-08-27:
**Is your enhancement proposal related to a problem? Please describe.**
`spi_nor.c` currently expect spi_nor_data.spi_config.operation to be `SPI_WORD_SET(8)`.  How should we go by supporting other kind of SPI Flash like 16 bits or transfering MSB first?

**Describe the solution you'd like**
These properties are device dependent.  So they should be in DTS?

BTW, I have set the parameter `operation` to 

```diff
+	data->spi_cfg.operation = SPI_OP_MODE_MASTER |
+				  SPI_TRANSFER_MSB | SPI_WORD_SET(16) |
+				  SPI_LINES_SINGLE | SPI_LOCK_ON;
```
And my device is working.

pabigot, 2020-08-27:
I don't understand why byte ordering is an issue: doesn't the flash present as a sequence of bytes, and the order depends on what's written?  What specific flash device are you using?

yashi, 2020-08-28:
It's S25FL256LAGMFM000

pabigot, 2020-08-31:
I think byte ordering is more likely to be a controller issue.

I have the S25FL512S flagged as a problematic device, but that's because it requires a 4-byte address.  I believe S25FL256 will as well, in order to use the whole memory.  We don't currently support that, but if that's the problem please open an issue and I will take a look, or you can prepare a PR.    "Other kind of SPI NOR" is too generic for an enhancement.

yashi, 2021-11-09:
I'll close this for now.  I'll create a new PR when I need to support a new deivice.

closed 2021-11-09 by yashi