zephyrproject-rtos/zephyr #27846: Weird ADC outliers on nrf52

mfiumara, 2020-08-27:
**Describe the bug**
When using the nrfx_saadc in advanced mode as per https://github.com/NordicSemiconductor/nrfx/wiki/SAADC-Advanced-mode-with-IRQs with continuous mode enabled, some strange outliers are observed when oversampling.

**To Reproduce**
Steps to reproduce the behavior:
1. Build the aforementioned sample with the following adc configuration:
```
const nrfx_saadc_channel_t channel_config = {
{
	.resistor_p = NRF_SAADC_RESISTOR_DISABLED,
	.resistor_n = NRF_SAADC_RESISTOR_DISABLED,
	.gain       = NRF_SAADC_GAIN1_6,
	.reference  = NRF_SAADC_REFERENCE_INTERNAL,
	.acq_time   = NRF_SAADC_ACQTIME_3US,
	.mode       = NRF_SAADC_MODE_SINGLE_ENDED,
	.burst      = NRF_SAADC_BURST_ENABLED,
  	},
.pin_p          = ACOUSTIC_CHANNEL_ID,
.pin_n          = NRF_SAADC_INPUT_DISABLED,
.channel_index  = 0,
};

err = nrfx_saadc_channels_config(&channel_config, 1);
if (NRFX_SUCCESS != err) {
	LOG_ERR("Failed initializing acoustics (err %x)", err);
	return (int) err;
}

const nrfx_saadc_adv_config_t advanced_config = {
	.oversampling = NRF_SAADC_OVERSAMPLE_8X,
	.burst = NRF_SAADC_BURST_ENABLED,
	.internal_timer_cc = ACOUSTIC_TIMER_CC,
	.start_on_end = true,
};
```

**Expected behavior**
I would not expect any outliers when sampling continuously.

**Impact**
For my application this is a showstopper, as we're relying on maximum / minmum values and with outliers these cannot be used.

**Logs and console output**
Example where the peak to peak is being calculated on each buffer of 1024 samples, a sudden outlier of 224 is observed:
```
[00:18:34.547,332] <dbg> acoustic.process_window: current_ptp: 1                                                                                                                               
[00:18:34.593,566] <wrn> acoustic: NRFX_SAADC_EVT_DONE size 1024                                                                                                                               
[00:18:34.593,811] <dbg> acoustic.process_window: current_ptp: 1                                                                                                                               
[00:18:34.640,075] <wrn> acoustic: NRFX_SAADC_EVT_DONE size 1024                                                                                                                               
[00:18:34.640,319] <dbg> acoustic.process_window: current_ptp: 1                                                                                                                               
[00:18:34.686,553] <wrn> acoustic: NRFX_SAADC_EVT_DONE size 1024                                                                                                                               
[00:18:34.686,798] <dbg> acoustic.process_window: current_ptp: 1                                                                                                                               
[00:18:34.733,001] <wrn> acoustic: NRFX_SAADC_EVT_DONE size 1024                                                                                                                               
[00:18:34.733,245] <dbg> acoustic.process_window: current_ptp: 1                                                                                                                               
[00:18:34.779,510] <wrn> acoustic: NRFX_SAADC_EVT_DONE size 1024                                                                                                                               
[00:18:34.779,754] <dbg> acoustic.process_window: current_ptp: 1                                                                                                                               
[00:18:34.826,019] <wrn> acoustic: NRFX_SAADC_EVT_DONE size 1024                                                                                                                               
[00:18:34.826,019] <wrn> acoustic: raw: 2683                                                                                                                                                   
[00:18:34.826,263] <dbg> acoustic.process_window: current_ptp: 224
`

**Environment (please complete the following information):**
 - Zephyr 2.3.0
 - Toolchain gnuarmembedded

**Additional context**
Related question that I posted on devzone: https://devzone.nordicsemi.com/f/nordic-q-a/63067/nrfx_saadc-advanced-mode-oversampling

mfiumara, 2020-08-27:
@pabigot

mfiumara, 2020-09-02:
This looks resolved when lowering the oversampling from 8x to 2x, is there any reason for this?

carlescufi, 2020-09-08:
@anangl please take a look

closed 2020-11-22 by github-actions[bot]