zephyrproject-rtos/zephyr #27849: tests: lib: cmsis_dsp: transform: malloc out of memory

ABOSTM, 2020-08-27:
**Describe the bug**
When running tests/lib/cmsis_dsp/transform/ on nucleo_f207zg,
with config: libraries.cmsis_dsp.transform.cf64 ,
`malloc()` failed as it runs out of memory.
Please note that nucleo_f207zg has 128k RAM (whereas there is a min_ram of 64)

**To Reproduce**
Steps to reproduce the behavior:
1. sanitycheck -N --device-testing --hardware-map ../map.yaml -c -T tests/lib/cmsis_dsp/transform/ -p nucleo_f207zg
2. See error

**Expected behavior**
Test passed

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v2.3.0-1075-g4e12dfb5ebda  ***
Running test suite transform_cf64
===================================================================
START - test_arm_cfft_f64_cmplx_noisy_16
 PASS - test_arm_cfft_f64_cmplx_noisy_16
===================================================================
START - test_arm_cifft_f64_cmplx_noisy_16
 PASS - test_arm_cifft_f64_cmplx_noisy_16
===================================================================
START - test_arm_cfft_f64_cmplx_noisy_32
 PASS - test_arm_cfft_f64_cmplx_noisy_32
===================================================================
START - test_arm_cifft_f64_cmplx_noisy_32
 PASS - test_arm_cifft_f64_cmplx_noisy_32
===================================================================
START - test_arm_cfft_f64_cmplx_noisy_64
 PASS - test_arm_cfft_f64_cmplx_noisy_64
===================================================================
START - test_arm_cifft_f64_cmplx_noisy_64
 PASS - test_arm_cifft_f64_cmplx_noisy_64
===================================================================
START - test_arm_cfft_f64_cmplx_noisy_128
 PASS - test_arm_cfft_f64_cmplx_noisy_128
===================================================================
START - test_arm_cifft_f64_cmplx_noisy_128
 PASS - test_arm_cifft_f64_cmplx_noisy_128
===================================================================
START - test_arm_cfft_f64_cmplx_noisy_256
 PASS - test_arm_cfft_f64_cmplx_noisy_256
===================================================================
START - test_arm_cifft_f64_cmplx_noisy_256
 PASS - test_arm_cifft_f64_cmplx_noisy_256
===================================================================
START - test_arm_cfft_f64_cmplx_noisy_512
 PASS - test_arm_cfft_f64_cmplx_noisy_512
===================================================================
START - test_arm_cifft_f64_cmplx_noisy_512
 PASS - test_arm_cifft_f64_cmplx_noisy_512
===================================================================
START - test_arm_cfft_f64_cmplx_noisy_1024
 PASS - test_arm_cfft_f64_cmplx_noisy_1024
===================================================================
START - test_arm_cifft_f64_cmplx_noisy_1024
 PASS - test_arm_cifft_f64_cmplx_noisy_1024
===================================================================
START - test_arm_cfft_f64_cmplx_noisy_2048
 PASS - test_arm_cfft_f64_cmplx_noisy_2048
===================================================================
START - test_arm_cifft_f64_cmplx_noisy_2048
 PASS - test_arm_cifft_f64_cmplx_noisy_2048
===================================================================
START - test_arm_cfft_f64_cmplx_noisy_2048
 PASS - test_arm_cfft_f64_cmplx_noisy_2048
===================================================================
START - test_arm_cifft_f64_cmplx_noisy_2048
 PASS - test_arm_cifft_f64_cmplx_noisy_2048
===================================================================
START - test_arm_cfft_f64_cmplx_noisy_4096
Assertion failed at WEST_TOPDIR/zephyr/tests/lib/cmsis_dsp/transform/src/cf64.c:27: test_arm_cfft_f64_cmplx_backend: (output is NULL)
buffer allocation failed
 FAIL - test_arm_cfft_f64_cmplx_noisy_4096
```

**Environment (please complete the following information):**
 - OS: Linux and Windows)
 - Toolchain Zephyr SDK
 - Commit SHA: 70ff1d73c42f2e174c49b1e217712b52ad6ae78b

**Analysis**
`malloc()` fails as it runs out of memory when it reaches step: `test_arm_cfft_f64_cmplx_noisy_4096`

&nbsp;&nbsp;&nbsp;&nbsp;`DEFINE_TEST_VARIANT4(arm_cifft_f64_cmplx, step_4096, &arm_cfft_sR_f64_len4096,`
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;`	in_cifft_step_4096, in_cfft_step_4096,  `**8192**);

Analysing further I faced 2 issues:

1) testcase.yaml has a min ram property for this config
`    min_ram: 64`
But while executing step `test_arm_cfft_f64_cmplx_noisy_4096` there is a malloc() of length:
8192 * sizeof(float64_t) = 8192 * 8 = 64k
but we need more RAM for variables, stack, ...
so 64K RAM is no enough.

2) I don't know exactly how malloc() algorithm from newlibc is working,
but it appears during my debug session that even if memory is freed between each step,
each time the size of the malloc is increased, the previous memory size is 'lost' (probably due to heap fragmentation).
So with current implementation before reaching step 4096, we loose 256 + 512 + 1024 + 2048 + ... + 16k + 32k, which is almost 64k lost. And  then it is no more possible to request 64k besides. Finally my 128k RAM are not enough.
If I reverse the order of the test steps : starting from 4096 and decreasing to 16, then test is passed.
Probably malloc() algorithm is able to reuse big memory chunk for smaller malloc which reduce fragmentation.

So 128k is  sufficient to run the test, but it needs changes.

Note: there is no problem with config libraries.cmsis_dsp.transform.rf64 because the max malloc is for `test_arm_rifft_f64_real_noisy_4096`
4096 * sizeof(float64_t) = 4096 * 8 = 32k

ABOSTM, 2020-08-27:
Implementation proposal: #27850

ABOSTM, 2020-08-27:
^^ @stephanosio

stephanosio, 2020-08-31:
Left some comments in the PR #27850.

PR #27850 (2020-09-10): tests: lib: cmsis:transform: cf64 min_ram 128k
tests: lib: cmsis:transform: cf64 min_ram 128k

tests/lib/cmsis_dsp/transform/ with config:
   * libraries.cmsis_dsp.transform.cf64
Needs more than 64k RAM.
128k is sufficient if tests steps are reordered
(decreasing malloc size)

Fixes #27849
closed 2020-09-10 by ioannisg