zephyrproject-rtos/zephyr #27855: i2c bitbanging on nrf52840

sviraaj, 2020-08-27:
Hi,
I am trying to use i2c_gpio (bitbanging) on nrf52840. I am unable to make it work yet! Did anyone test bitbanging i2c on nrf52840?

Following DT settings used.
```
     gpio_i2c0: gpio_i2c {                                                                                         
         compatible = "gpio-i2c";
         status = "okay";
         clock-frequency = <I2C_BITRATE_STANDARD>;
         sda-gpios = <&gpio0 26 (GPIO_OPEN_DRAIN)>;
         scl-gpios = <&gpio0 27 (GPIO_OPEN_DRAIN)>;
         label = "GPIO_I2C_0";
         #address-cells = <1>;
         #size-cells = <0>;
     }
```

pabigot, 2020-08-27:
Are you trying to use I2C, or are you trying to specifically use the bit-bang I2C driver instead of the `nordic,nrf-twi` I2C device that already uses those pins?  If the latter, you need to disable the existing driver:

```
&i2c0 {
    status = "disabled";
};
```

sviraaj, 2020-08-27:
@pabigot Thanks for the answer. I am trying to use the bit-bang I2C driver. Is there any reason why I need to disable the existing driver? Could both the drivers not work simultaneously?

pabigot, 2020-08-27:
If the TWI driver is active it controls the pins, so attempts to use the bitbang code will not work.  The devices aren't designed to be turned on and off, transferring control from one driver instance to another..  Your best option is to use two different sets of pins, one for the I2C and another for the bitbang.  Alternatively if the slow I2C devices would work at 100 kHz then use that (slowest) I2C speed with the TWI driver.

sviraaj, 2020-08-28:
@pabigot Sorry, I should have probably mentioned that I was using different pins for different drivers (I2C-TWIM and GPIO-I2C). The issue seems to be something else for me. Any idea what else might be causing the trouble? Thanks.

pabigot, 2020-08-28:
Yes: the pins in the devicetree fragment you gave are the same pins normally used by the `i2c0` device.  Did you try adding the fragment I suggested?

sviraaj, 2020-08-28:
@pabigot Yes, I have disabled i2c0. And also, the board I am using is custom, so i2c0 dts has different sda and scl pins assigned than the one mentioned above. Just in case, I have still disabled the i2c0 as you suggested. I still have the same issue. The i2c reads all '0' on the bus.

pabigot, 2020-08-28:
Ok, I can reproduce this.  Looks like the GPIO driver doesn't work on nrf52840.  Oddly, the line signals are good, so it's something in the API.  I'll look into it.

pabigot, 2020-08-28:
Try #27875.

pabigot, 2020-08-28:
You might also need this, or you might not.  But it was enabling input that fixed it for me.
```
     sda-gpios = <&gpio1 1 (GPIO_OPEN_DRAIN | GPIO_PULL_UP)>;
     scl-gpios = <&gpio1 2 (GPIO_OPEN_DRAIN | GPIO_PULL_UP)>;
```

sviraaj, 2020-08-28:
@pabigot  Wow. This worked -> #27875 . I wonder why? Maybe it has something to do with the behavior in nrf52840 GPIO controller, since SDA needs to be bidirectional? Anyways, Thanks a lot for this patch. :)

closed 2020-08-31 by ioannisg

jpdoyon, 2021-03-22:
@sviraaj I'm trying to use the gpio-i2c too but I can't seem to get the binding from the call to device_get_binding.  How are you getting it?  By calling device_get_binding("GPIO_I2C_0"); ??

sviraaj, 2021-03-22:
Hi,
Yes, device_get_binding("GPIO_I2C_0")  will work. Make sure that you are
enabling it in prj.conf by setting CONFIG_I2C_GPIO=y and the corresponding
node in the device tree.

cyberjupiter, 2023-07-06:
> Hi, Yes, device_get_binding("GPIO_I2C_0") will work. Make sure that you are enabling it in prj.conf by setting CONFIG_I2C_GPIO=y and the corresponding node in the device tree.
> […](#)

HI @sviraaj. I am sure I probably need to open an issue but just to confirm first, can I see your dts and the application code (just the part where you used gpio-i2c).

Thanks.

PR #27875 (2020-08-31): i2c: bitbang: fix SDA pin configuration
The driver must be able to read from this pin as well as write to it.

Fixes #27855.