zephyrproject-rtos/zephyr #27867: up_squared: couldn't get test result from serial of each test.

chen-png, 2020-08-28:
**Describe the bug**
for up_squared, when using zephyr.strip file and usb disk to run tests on this board, couldn't get test result from serial for each test.

the good commit is 4cf1832fa8, but for the latest, it's broken.

**To Reproduce**
Steps to reproduce the behavior:
1.  west build -b up_squared -p auto samples/hello_world/
2.  west flash ~/up_squared.sh
3.  see error

**Logs and console output**
there is nothing in the console, it seems that the test doesn't boot up.

**Environment (please complete the following information):**
 - OS: fedora28
 - Toolchain: zephyr-sdk-0.11.4
 - CommitID: 6a8c622bcca

jhedberg, 2020-08-31:
@chen-png did you try `git bisect` to narrow this down, since you have a fairly recent "good" commit?

chen-png, 2020-09-07:
> @chen-png did you try `git bisect` to narrow this down, since you have a fairly recent "good" commit?

the recent good commit is e9d1545, and the first bad commit is 38e17b6.

MaureenHelm, 2020-09-14:
@wentongwu @jhedberg any update on this issue? will it be fixed soon?

andrewboie, 2020-09-16:
@MaureenHelm I'm taking a look at this.

andrewboie, 2020-09-16:
Also there appears to be a workaround via BIOS settings to use PIO for serial, in which case we might consider lowering severity. I'll follow up on this.

tbursztyka, 2020-09-16:
@andrewboie I'll try your fix on my up² (so far, nor efi nor grub boot works, even on commit before your paging ones...)

nashif, 2020-09-16:
so just to document this here, building with -DCONFIG_PCIE_MMIO_CFG=n, get this back to a working state, if this is a workable solution, we could downgrade priority while working on a  fix.

PR #28446 (2020-09-17): x86: memory map BIOS Data Area
Changes to paging code ensured that the NULL virtual page is
never mapped. Since RAM is identity mapped, on a PC-like
system accessing the BIOS Data Area in the first 4K requires
a memory mapping. We need to read this to probe the ACPI RSDP.

Additionally check that the BDA has something in it as well
and not a bunch of zeroes.

It is unclear whether this function is truly safe on UEFI
systems, but that is for another day.

Fixes: #27867
commit 8240d7d0 (2020-09-17): x86: memory map BIOS Data Area

Changes to paging code ensured that the NULL virtual page is
never mapped. Since RAM is identity mapped, on a PC-like
system accessing the BIOS Data Area in the first 4K requires
a memory mapping. We need to read this to probe the ACPI RSDP.

Additionally check that the BDA has something in it as well
and not a bunch of zeroes.

It is unclear whether this function is truly safe on UEFI
systems, but that is for another day.

Fixes: #27867
closed 2020-09-17 by nashif