zephyrproject-rtos/zephyr #27870: drivers: modem: create subfolders

chris-makaio, 2020-08-28:
I'm not sure it's okay to suggest this, but wouldn't it be better to move modem drivers in subfolders?
I'm once again back to porting a modem driver and find it really hard to follow the code in those single files. I'd prefer to create "my" driver in a subfolder and split several parts across multiple files (power management, socket handling, command handling to name a few)

jukkar, 2020-09-04:
There has been some talks in the past that we should have proper modem APIs but so far no one has stepped up and send patches. Atm the modem directory is just a bunch of files and the actual drivers need to use the facilities directly. So this needs more TLC and patches are welcome in this area.
cc @mike-scott

chris-makaio, 2020-09-04:
I absolutely agree, but I honestly refrain from starting with patching w/o further discussions before.. I want to do it right and not submit a finished concept which gets declined immediately ;) 
Well, I could start by publishing a fork to check out for everyone involved... happy to discuss