zephyrproject-rtos/zephyr #27876: TCP2: Apaprent issues with server-side connections (>1 connection doesn't work properly)

pfalcon, 2020-08-28:
**Describe the bug**

It looks that new TCP2 doesn't properly handle server-side connections (i.e. Zephyr acting as a listening socket): of several consecutive connections, only 1st suceeds. 2nd and following have the data delivered, but the connection isn't closed properly from the Zephyr server side (and thus client hangs  waiting for more data - either until timeout or forever).

**To Reproduce**
Steps to reproduce the behavior:
1. Make a standard qemu networking setup per Zephyr docs.
2. cd samples/net/sockets/dumb_http_server
3. west build --board qemu_x86 -t run
4. Open another console, following steps are in it.
5. (Using telnet to show bad behavior in detail). `telnet 192.0.2.1 8080`
6. Enter `GET / HTTP/1.0`, followed by 2 newlines.
7. Watch the HTML output appear and connection being closed (`Connection closed by foreign host.`). This is the expected behavior, happening on the 1st connection.
8. Re-run the telnet command, paste the same input.
9.  Watch that HTML output appear (the last line is `</body></html>`), but the connection not being closed, the local telnet session waits for more data from the Zephyr server. It can stay like that indefinitely (assumed; tested for 5 mins).
10. Press Ctrl+], then Ctrl+D to terminate the session from local side.
11. Re-run the telnet command and paste the same input, to confirm that 3rd and consecutive connections are also not closed properly (but no other changes in the behavior are apparent, at least the data is delivered).

With samples/net/sockets/dumb_http_server_mt, there's a similar issue. (But this sample doesn't seem to work with telnet, and with `curl http://192.0.2.1:8080/`, even the 1st connection appears to hang).

With CONFIG_NET_TCP1=y enabled, both samples work as expected (it being `ab -n1000 http://192.0.2.1:8080/` running without errors).

**Expected behavior**

There should be no hanging connections. `ab -n1000 http://192.0.2.1:8080/` should run without errors.

**Impact**

I would humbly consider this to be a grave issue, a serious regression comparing to TCP1 stack. As it stands, it raises a question whether TCP2 is ready to be enabled by default. (OTOH, this looks like an issue with delivering a FIN TCP packet. Hopefully, this is easy to fix to developers of TCP2.)

**Environment (please complete the following information):**
 - OS (client-side): Linux
 - Toolchain: Zephyr SDK
 - Commit SHA: master 7efe988a5436024020925

jukkar, 2020-08-31:
As there is a workaround (using the legacy stack) for this, lowering the priority.

pfalcon, 2020-09-01:
To confirm, the same issue happens with BOARD=frdm_k64f (tested with the same revision 7efe988 for consistency). Actually, with the `dumb_http_server` sample, even 1st connection hangs.

PR #27960 (2020-09-02): net: tcp2: Fix connection close seq values
When connection is closed and we send ACK flag, use proper seq
values so that any data that is still in flight will get acked too.
Currently this assumes that window is still open.

Fixes #27876
commit 4f6145a4 (2020-09-02): net: tcp2: Fix connection close seq values

When connection is closed and we send ACK flag, use proper seq
values so that any data that is still in flight will get acked too.
Currently this assumes that window is still open.

Fixes #27876
closed 2020-09-02 by jukkar