zephyrproject-rtos/zephyr #27880: build errors for some samples/ on lpcxpresso55s69_cpu1

jenmwms, 2020-08-28:
**Describe the bug**
This PR https://github.com/zephyrproject-rtos/zephyr/pull/27549 encountered build errors for the platform `lpcxpresso55s69_cpu1` on some samples. No build errors occured for lpcxpresso55s69_cpu0.

These 10 are from the buildkite report https://buildkite.com/zephyr/zephyr/builds/5366#4f785229-ffdf-4f11-b5c3-e8984a6ff795 
- samples/mpu/mpu_test/sample.mpu.mpu_test
- samples/drivers/led_apa102c_bitbang/sample.drivers.led.apa102c_bitbang
- samples/drivers/flash_shell/sample.drivers.flash.shell
- samples/drivers/entropy/sample.drivers.entropy
- samples/drivers/crypto/sample.drivers.crypto.mbedtls.micro
- samples/drivers/crypto/sample.drivers.crypto.mbedtls
- samples/basic/threads/sample.basic.threads
- samples/basic/button/sample.basic.button
- samples/application_development/out_of_tree_driver/sample.drivers.out_of_tree
- samples/application_development/external_lib/sample.app_dev.external_lib

**To Reproduce**
For example,
` west build -p -b lpcxpresso55s69_cpu1 samples/mpu/mpu_test`

**Expected behavior**
No cmake build error using this board identifier.

**Impact**

**Logs and console output**
```
$ west build -p -b lpcxpresso55s69_cpu1 samples/mpu/mpu_test   -- west build: making build dir /home/jennife2/zephyrproject/zephyr/build pristine
-- west build: generating a build system
Including boilerplate (Zephyr base): /home/jennife2/zephyrproject/zephyr/cmake/app/boilerplate.cmake
-- Application: /home/jennife2/zephyrproject/zephyr/samples/mpu/mpu_test
-- Zephyr version: 2.3.99 (/home/jennife2/zephyrproject/zephyr)
-- Found Python3: /usr/bin/python3.7 (found suitable exact version "3.7.7") found components:  Interpreter
-- Found west: /home/jennife2/.local/bin/west (found suitable version "0.7.2", minimum required is "0.7.1")
-- Board: lpcxpresso55s69_cpu1
-- Found toolchain: zephyr (/home/jennife2/zephyr-sdk-0.11.3)
-- Found dtc: /home/jennife2/zephyr-sdk-0.11.3/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: /home/jennife2/zephyrproject/zephyr/boards/arm/lpcxpresso55s69/lpcxpresso55s69_cpu1.dts
-- Generated zephyr.dts: /home/jennife2/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/jennife2/zephyrproject/zephyr/build/zephyr/include/generated/devicetree_unfixed.h
Parsing /home/jennife2/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/jennife2/zephyrproject/zephyr/boards/arm/lpcxpresso55s69/lpcxpresso55s69_cpu1_defconfig'
Merged configuration '/home/jennife2/zephyrproject/zephyr/samples/mpu/mpu_test/prj.conf'

error: ARM (defined at arch/Kconfig:27) is assigned in a configuration file, but is not directly
user-configurable (has no prompt). It gets its value indirectly from other symbols. See
http://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_ARM.html and/or look up ARM in the
menuconfig/guiconfig interface. The Application Development Primer, Setting Configuration Values,
and Kconfig - Tips and Best Practices sections of the manual might be helpful too.

CMake Error at /home/jennife2/zephyrproject/zephyr/cmake/kconfig.cmake:230 (message):
  command failed with return code: 1
Call Stack (most recent call first):
  /home/jennife2/zephyrproject/zephyr/cmake/app/boilerplate.cmake:529 (include)
  /home/jennife2/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:24 (include)
  /home/jennife2/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:35 (include_boilerplate)
  CMakeLists.txt:5 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /home/jennife2/bin/cmake/cmake-3.13.1-Linux-x86_64/bin/cmake -DWEST_PYTHON=/usr/bin/python3 -B/home/jennife2/zephyrproject/zephyr/build -S/home/jennife2/zephyrproject/zephyr/samples/mpu/mpu_test -GNinja -DBOARD=lpcxpresso55s69_cpu1
```
**Environment (please complete the following information):**
 - OS: Fedora 30
 - Toolchain; Zephyr SDK
 - d0f5253

nashif, 2020-08-28:
This is blocking other PRs fixing other bugs from going so raise priority.

agansari, 2020-09-01:
@nashif @jenmwms @MaureenHelm I've created a pull to fix the build problems, but we need to blacklist this target from some of the samples/tests: 
 - right now running this is not possible, this target does not run without cpu 0 enabling it
 - samples/mpu/mpu_test/sample.mpu.mpu_test - core has no MPU
 - samples/drivers/entropy/sample.drivers.entropy - no entropy device
 - samples/basic/threads/sample.basic.threads - led0 is not allocated to this core
 - samples/basic/button/sample.basic.button - no button allocated

nashif, 2020-09-15:
>     * samples/drivers/entropy/sample.drivers.entropy - no entropy device

however, the soc is selecting this...

see soc/arm/nxp_lpc/lpc55xxx/Kconfig.series

PR #27933 (2020-09-01): boards: lpcxpresso55s69_cpu1: fix build errors
Remove bad configurations added full gpio configuration.

Partially fixes #27880
PR #28420 (2020-09-16): boards: arm: Add entropy dts chosen node for lpcxpresso55s69_cpu1
Adds an entropy dts chosen node for the lpcxpresso55s69_cpu1 board
configuration. Commit 5083038 added
this for the _cpu0 and _ns board configurations, but missed the _cpu1
board configuration.

Fixes sanitycheck build errors for network socket samples.

Fixes #27880
closed 2020-09-16 by galak