zephyrproject-rtos/zephyr #27881: Zephyr requirements.txt fails to install on Python 3.9rc1

mbolivar-nordic, 2020-08-28:
**Describe the bug**

Python 3.9rc1 has been released and is available for testing. Zephyr currently fails to install with Python 3.9rc1, at least on Windows 10, due to various missing requirements.

The [expected release date](https://www.python.org/dev/peps/pep-0596/) for 3.9.0 final is Monday, 2020-10-05, so we can expect users to be unable to install Zephyr with the latest Python on that date.

**To Reproduce**

1. [Install 3.9rc1 from python.org](https://www.python.org/downloads/release/python-390rc1/) (downloads at bottom of page)
2. Follow Zephyr getting started guide 
3. Get to 'pip3 install -r scripts/requirements.txt' part
4. Get a ton of errors

If you are installing this in parallel with another Python, I recommend a virtual environment.

Make sure to run the right `pip` using something like this:

```
py -3.9 -m pip install -r scripts/requirements.txt
```

**Expected behavior**

Works good plz

**Impact**

Will be a showstopper when 3.9 is out on officially supported platforms, e.g. on Windows whenever 3.9 lands in chocolatey

**Logs and console output**

As best I can tell, the packages which we cannot install are:

- `windows-curses`
- `Pillow`
- `lxml`

Some of these are direct dependencies; `lxml` is indirect. I used the (excellent) [pipdeptree](https://pypi.org/project/pipdeptree/) on my Linux machine to investigate where the dependencies were coming from.

The below resulting diff was enough to get me through installing Python requirements and led to a successful `west build` + `west flash` on a Nordic board in Windows on 3.9rc1:

```diff
diff --git a/scripts/requirements-base.txt b/scripts/requirements-base.txt
index 5f10ed2538..4e626cd372 100644
--- a/scripts/requirements-base.txt
+++ b/scripts/requirements-base.txt
@@ -11,7 +11,7 @@ pyelftools>=0.26
 PyYAML>=5.1

 # used by west_commands
-canopen
+# canopen
 packaging
 progress

@@ -26,4 +26,4 @@ west>=0.7.2

 # used for windows based 'menuconfig'
 # "win32" is used for 64-bit Windows as well
-windows-curses; sys_platform == "win32"
+# windows-curses; sys_platform == "win32"
diff --git a/scripts/requirements-build-test.txt b/scripts/requirements-build-test.txt
index 6ef9d03676..5cc50d9af3 100644
--- a/scripts/requirements-build-test.txt
+++ b/scripts/requirements-build-test.txt
@@ -13,7 +13,7 @@ ply>=3.10
 pykwalify

 # used for code coverage
-gcovr>=4.2
+# gcovr>=4.2
 coverage

 # used for west-command testing
diff --git a/scripts/requirements-extras.txt b/scripts/requirements-extras.txt
index 90200ee02f..11a98f7b02 100644
--- a/scripts/requirements-extras.txt
+++ b/scripts/requirements-extras.txt
@@ -13,4 +13,4 @@ junit2html
 lpc_checksum

 # used by scripts/gen_cfb_font_header.py - helper script for user
-Pillow
+# Pillow
```

From pipdeptree, `gcovr` is pulling in `lxml`. I had to rip out `canopen` because it also depends on `windows-curses`.

All of the errors seem to be due to missing wheels. This sometimes causes a fallback to a source install, which bombs out because the required development headers are absent on the Windows machine.

Make 100% sure cmake uses the right python like this:

```
py -3.9 -m west.app.main build -b nrf52840dk_nrf52840 -s samples/hello_world -- -DPYTHON_PREFER=C:\Users\Marti\AppData\Local\Programs\Python\Python39\python.exe
```

**Environment (please complete the following information):**
 - OS: Windows 10
 - Toolchain: N/A
 - Commit SHA: today's master, f75f8bec2046cab122ff050390f6fdb0716d2e35

nashif, 2020-08-31:
dont see this as a bug, it is probably something to monitor but I do not see how to fix this now at least (based on python 3.9 being not released yet)

mbolivar-nordic, 2020-08-31:
> I do not see how to fix this now at least (based on python 3.9 being not released yet)

1. The zephyr project (@carlescufi  in particular IUC) owns the `windows-curses` package and we need to update it ourselves. It will not magically get fixed on its own.
2. We can decide what to do about the others. Work with the project maintainers to make sure wheels will be available or remove them from requirements.txt, for example.

The lxml documentation, for example, seems vague about whether we can expect a wheel: https://lxml.de/installation.html. This seems like a potential issue that should be understood and handled ahead of time. Perhaps we need to remove projects that depend on it out of the main requirements.txt tree of files.

nashif, 2020-08-31:
> Perhaps we need to remove projects that depend on it out of the main requirements.txt tree of files.

or we could also say that we will not support python 3.9 the minute it is released...

andrewboie, 2020-08-31:
we can promote this to a bug at some point when this version of Python is released, right now this is just messing with our bug counts

carlescufi, 2020-09-01:
> > Perhaps we need to remove projects that depend on it out of the main requirements.txt tree of files.
> 
> or we could also say that we will not support python 3.9 the minute it is released...

This is relatively easy on Linux but harder on Windows and macOS, where chocolatey and brew automagically update all packages to the latest available versions, including Python major versions.

@mbolivar-nordic will rebuild the `windows-curses` wheel for 3.9 so we are ready when it's released. For the rest, I suggest we wait until a couple of weeks after release to see how the dependencies adapt.

nashif, 2020-09-01:
> This is relatively easy on Linux but harder on Windows and macOS, where chocolatey and brew automagically update all packages to the latest available versions, including Python major versions.

I still can do

```brew install python@3.7```

and I imagine you should be able to do the same thing with any package manager.

mbolivar-nordic, 2020-09-02:
> we can promote this to a bug at some point when this version of Python is released, right now this is just messing with our bug counts

Very well.

> and I imagine you should be able to do the same thing with any package manager.

Not really on bleeding edge distros whose package repositories delete old versions, but I will grant you that those users should know how to build their own Python if necessary.

On windows etc., though, where users need more hand-holding, I would argue that means we should document the supported versions of Python, especially if it's easy to guess that the current one is going to break immediately after the next release. So there is still something to do here.

The Python RC period is long exactly to allow community projects like ours that depend on it to get their ducks in a row and build their wheels before the release.

Kicking the can down the road by changing this to an enhancement rather than a bug until it actually bites people feels a little bit like shenanigans just to meet the release criteria, similarly to how sometimes medium priority bugs magically become low priority right before the release sometimes. But I get it, we've got deadlines to meet.

Note that Python is moving to a yearly release cadence starting with 3.9, so this is going to happen more often than before.

mbolivar-nordic, 2020-09-04:
Initial python 3.9 support for windows-curses is up here: https://github.com/zephyrproject-rtos/windows-curses/pull/15

Some open questions in my mind about how we should release that, since I've never done a release of this package (or any binary wheel).

mbolivar-nordic, 2020-09-25:
> Initial python 3.9 support for windows-curses is up here: [zephyrproject-rtos/windows-curses#15](https://github.com/zephyrproject-rtos/windows-curses/pull/15)

Ping

mbolivar-nordic, 2020-11-04:
A windows-curses build for 3.9 is now available on PyPI. Everything else installs now as well, so I'm closing this issue.

Nobody is stepping up to maintain windows-curses, though, so I'm proposing we transition Windows users to guiconfig in time for 3.10: https://github.com/zephyrproject-rtos/zephyr/pull/29770

closed 2020-11-04 by mbolivar-nordic