zephyrproject-rtos/zephyr #27885: Question regarding to some issues when work with OpenThread in Zephyr

oubotong, 2020-08-30:
I tried to implement MQTT-SN protocol with OpenThread and everything works normally within OpenThread. When I tried to build the same program in Zephyr, my board doesn't send out the CONNECT unicast message to the broker(I built up a sniffer, and it can't detect the CONNECT message).

While I tried to debug it, I couldn't find out the final function to send out the radio. So I want to seek advice on what could be the possible reason for my issue.

Platform: NRF52840-DK
Test successfully by running: make -f examples/Makefile-nrf52840 MQTTSN=1 JOINER=1
Test failure in zephyr with the following configuration file prj.conf and overlay-ot.conf:
prj.conf:

# Generic networking options
CONFIG_NETWORKING=y
CONFIG_NET_UDP=y
CONFIG_NET_TCP=y
CONFIG_NET_IPV6=y
CONFIG_NET_IPV4=y
CONFIG_NET_SOCKETS=y
CONFIG_NET_SOCKETS_POSIX_NAMES=y
CONFIG_NET_SOCKETS_POLL_MAX=4
CONFIG_NET_CONNECTION_MANAGER=y

# Kernel options
CONFIG_MAIN_STACK_SIZE=2048
CONFIG_ENTROPY_GENERATOR=y
CONFIG_TEST_RANDOM_GENERATOR=y
CONFIG_INIT_STACKS=y

# Network buffers
CONFIG_NET_PKT_RX_COUNT=16
CONFIG_NET_PKT_TX_COUNT=16
CONFIG_NET_BUF_RX_COUNT=80
CONFIG_NET_BUF_TX_COUNT=80
CONFIG_NET_CONTEXT_NET_PKT_POOL=y

# IP address options
CONFIG_NET_IF_UNICAST_IPV6_ADDR_COUNT=3
CONFIG_NET_IF_MCAST_IPV6_ADDR_COUNT=4
CONFIG_NET_MAX_CONTEXTS=10
overlay-ot.conf:

CONFIG_NEWLIB_LIBC=y

CONFIG_OPENTHREAD_COAP=y
CONFIG_OPENTHREAD_MQTTSN=y

# Disable TCP and IPv4 (TCP disabled to avoid heavy traffic)
CONFIG_NET_TCP=n
CONFIG_NET_IPV4=n

CONFIG_NET_IPV6_NBR_CACHE=n
CONFIG_NET_IPV6_MLD=n
CONFIG_NET_CONFIG_NEED_IPV4=n

CONFIG_SYSTEM_WORKQUEUE_STACK_SIZE=2048
CONFIG_SHELL_STACK_SIZE=3072

CONFIG_NET_L2_OPENTHREAD=y

# mbedTLS tweaks
CONFIG_MBEDTLS_SSL_MAX_CONTENT_LEN=768

# A sample configuration to enable Thread Joiner, uncomment if needed
CONFIG_OPENTHREAD_JOINER=y
CONFIG_OPENTHREAD_JOINER_AUTOSTART=y
Source code:
```
#include <string.h>
#include <unistd.h>
#include <stdlib.h>
#include <stdio.h>
#include <stdbool.h>

#include "openthread/cli.h"
#include "openthread/instance.h"
#include "openthread/thread.h"
#include "openthread/tasklet.h"
#include "openthread/ip6.h"
#include "openthread/mqttsn.h"
#include "openthread/dataset.h"
#include "openthread/link.h"

#define NETWORK_NAME "OpenThread"
#define PANID 0xFACE

#define GATEWAY_PORT 47193
#define GATEWAY_ADDRESS "fd11:22::1637:7871:d626:bf3e"

#define CLIENT_ID "NRF52840"
#define CLIENT_PORT 47193

static void HandleConnected(otMqttsnReturnCode aCode, void* aContext)
{
    OT_UNUSED_VARIABLE(aCode);
    OT_UNUSED_VARIABLE(aContext);
    printf("Connected\n");
    otInstance *instance = (otInstance *)aContext;
    const char *clientStateString;
    otMqttsnClientStateToString(otMqttsnGetState(instance), &clientStateString);
    printf("MQTT state:%s\n",clientStateString);

}

static void HandleDisconnected(otMqttsnDisconnectType aType, void* aContext)
{
    OT_UNUSED_VARIABLE(aType);
    OT_UNUSED_VARIABLE(aContext);
    printf("Disconnected\n");
}

static void StateChanged(otChangedFlags aFlags, void *aContext)
{
    otInstance *instance = (otInstance *)aContext;
    if ((aFlags & OT_CHANGED_THREAD_ROLE) != 0)
   {
       otDeviceRole changedRole = otThreadGetDeviceRole(aContext);

       switch (changedRole)
       {
       case OT_DEVICE_ROLE_LEADER:
           printf("Leader\n");
           break;

       case OT_DEVICE_ROLE_ROUTER:
           {
           printf("Router\n");
         if(otMqttsnGetState(instance) == kStateDisconnected){
            printf("begin to connect\n");
            otIp6Address address;
            uint16_t port=47193;
            otIp6AddressFromString(GATEWAY_ADDRESS, &address);
            otMqttsnSetConnectedHandler(instance, HandleConnected, (void *)instance);
            otMqttsnSetDisconnectedHandler(instance, HandleDisconnected, (void *)instance);
            otMqttsnConnectDefault(instance,&address,(uint16_t)port);
	 }  
           }
           break;
       
       case OT_DEVICE_ROLE_CHILD:
           {
           printf("Child\n");
           }
           break;
       
       case OT_DEVICE_ROLE_DETACHED:
           printf("Detached\n");
	   break;
       case OT_DEVICE_ROLE_DISABLED:
           printf("Disable\n");
           break;
        }
    }
}

int main(int aArgc, char *aArgv[])
{
    otError error = OT_ERROR_NONE;
    otInstance *instance;

    otSysInit(aArgc, aArgv);
    instance = otInstanceInitSingle();

    // Set default network settings
    // Set network name
    error = otThreadSetNetworkName(instance, NETWORK_NAME);
    // Set PANID
    error = otLinkSetPanId(instance, PANID);
    // Set channel
    error = otLinkSetChannel(instance, DEFAULT_CHANNEL);

    // Register notifier callback to receive thread role changed events
    error = otSetStateChangedCallback(instance, StateChanged, instance);

    // Start thread network
    error = otIp6SetEnabled(instance, true);
    error = otThreadSetEnabled(instance, true);
    error = otMqttsnStart(instance, CLIENT_PORT);

    while (true)
    {
        otTaskletsProcess(instance);
        otSysProcessDrivers(instance);
    }
    return error;
}
```
Thanks for the help in advance!
If someone is interested in using MQTT-SN with OpenThread in Zephyr application, I'd like to share the library and we can debug the issue together.

oubotong, 2020-08-31:
@rlubos Can I have a quick online discussion with you about my issue? Thanks a lot.

closed 2020-09-01 by oubotong

oubotong, 2020-09-01:
@rlubos Could you tell me your email address? That would be great! Thank!

rlubos, 2020-09-01:
@oubotong Just ping me on slack, same username.

rlubos, 2020-09-01:
Hi, you shouldn't set up the OpenThread manually in the main() function - in Zephyr we initialize OpenThread during system startup (in the default configuration), and have a separate thread to process OpenThread:
https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/net/l2/openthread/openthread.c#L410
https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/net/l2/openthread/openthread.c#L262

In your case, as you need to register state handler, you'd need to start the OpenThread manually though (as it has to be done before the network is running). The following should be enough in your main:
```
int main(int aArgc, char *aArgv[])
{
	openthread_set_state_changed_cb(StateChanged);
	openthread_start(openthread_get_default_context());
}
```

You'd need to set some extra configs as well:
```
CONFIG_OPENTHREAD_MANUAL_START=y # to disable thread autostart
CONFIG_OPENTHREAD_CHANNEL=11 # to set the channel
CONFIG_OPENTHREAD_PANID=64206 # to set the panid
CONFIG_OPENTHREAD_NETWORK_NAME="OpenThread" # to set the network name 
```

I assume you have a border router in your setup as well, so you might need to set either of the following options, in order to be able to configure global IPv6 address:
```
CONFIG_OPENTHREAD_SLAAC=y
CONFIG_OPENTHREAD_DHCP6_CLIENT=y
```