zephyrproject-rtos/zephyr #27915: Samples:LoRa  send;sx126x with NRF52840dk ,no data from SPI miso

holmes81, 2020-09-01:
**Describe the bug**
I use the NRF52840DK board with sx126x lora board,and then try to run **samples/drivers/lora/send**
But from the log, I can’t get any data from MISO, but everything else is normal, such as CLK, MOSI, NSS.
I am pretty sure that the hardware is ok.
Here is my prj.conf:

**CONFIG_LOG=y
CONFIG_GPIO=y
CONFIG_LORA=y
CONFIG_LORA_SX12XX=y

CONFIG_LORA_LOG_LEVEL_DBG=y

CONFIG_RTT_CONSOLE=y
CONFIG_USE_SEGGER_RTT=y
CONFIG_UART_CONSOLE=n

CONFIG_SPI=y
CONFIG_SPI_1=y
CONFIG_SPI_NRFX=y**

My dts:
&spi1 {
        compatible = "nordic,nrf-spi";
        status = "okay";
        sck-pin = <47>;
        mosi-pin = <45>;
        miso-pin = <46>;
        cs-gpios = <&arduino_header 13 GPIO_ACTIVE_LOW>;

        lora: sx1261@0 {
                compatible = "semtech,sx1261";
                reg = <0>;
                label = "sx1261";
                reset-gpios = <&gpio0 3 0>;
                busy-gpios = <&gpio1 4 1>;
                dio1-gpios = <&gpio1 6 1>;
                spi-max-frequency = <4000000>;
        };
};

**To Reproduce**
Steps to reproduce the behavior:
1.west build -b nrf52840dk_nrf52840 lora/send/ --pristine
2.nrfjprog --program build/zephyr/zephyr.hex --sectorerase
3.nrfjprog --reset
4. Observe the data through a logic analyzer.See error

**Expected behavior**
SPI work normal 

**Impact**
SPI can't get work normally

**Logs and console output**
No error obeserved from the text log:

00> [00:00:00.000,000] <inf> NRFX_SPI: Function: nrfx_spi_init, error code: NRFX_SUCCESS.
00> [00:00:00.000,000] <dbg> sx126x.sx126x_lora_init: Initializing sx1261
00> [00:00:00.000,000] <dbg> sx126x.SX126xReset: Resetting radio
00> [00:00:00.032,928] <inf> NRFX_SPI: Function: nrfx_spi_xfer, error code: NRFX_SUCCESS.

00> [00:00:12.041,809] <inf> NRFX_SPI: Function: nrfx_spi_xfer, error code: NRFX_SUCCESS.
00> [00:00:12.041,961] <inf> NRFX_SPI: Function: nrfx_spi_xfer, error code: NRFX_SUCCESS.
00> [00:00:12.041,992] <dbg> NRFX_SPI.irq_handler: Event: NRF_SPI_EVENT_READY.
00> [00:00:12.042,022] <inf> NRFX_SPI: Function: nrfx_spi_xfer, error code: NRFX_SUCCESS.
00> [00:00:12.042,022] <dbg> NRFX_SPI.irq_handler: Event: NRF_SPI_EVENT_READY.
00> [00:00:12.042,053] <dbg> NRFX_SPI.irq_handler: Event: NRF_SPI_EVENT_READY.
00> [00:00:12.042,053] <dbg> NRFX_SPI.irq_handler: Event: NRF_SPI_EVENT_READY.
00> [00:00:12.042,083] <inf> NRFX_SPI: Function: nrfx_spi_xfer, error code: NRFX_SUCCESS.
00> [00:00:12.042,114] <inf> lora_send: Data sent!

[image]
[image]

**Environment (please complete the following information):**
 - OS: Linux
 - Board:nrf52840DK+sx126x

**Additional context**
Add any other context about the problem here.

anangl, 2020-09-02:
The modem's reset line seems to be active-low, so you probably need to use:
```
reset-gpios = <&gpio0 3 GPIO_ACTIVE_LOW>;
```
If the modem is kept in reset, that could explain the silence on the MISO line.

holmes81, 2020-09-03:
@anangl You are right,this solved my problem, thank you very much

GABS9718, 2020-09-14:
Hi @holmes81, were you able to receive the data sent? I'm using the same board, I can get the message ''Data sent'' in the log output, but I dont receive in my gateway. I think that this is related with the settings of OTAA protocol, ''Device EUI'', ''Application EUI'' and ''App key''. Do u know how to help me?