zephyrproject-rtos/zephyr #27928: Settings api hangs

CodingGhost, 2020-09-01:
**Describe the bug**
I have something like the following in my code:

`void main(void)
{
	settings_subsys_init();
	settings_register(&my_conf);//name is system
        int a = 2;
        settings_save_one("system/test",&a,sizeof(a));
}`

this will work completly normal. But as soon as I move the settings_save_one out of the main, it will just hang until my wdt resets the chip. Debugging looks like its hanging in idle.

also, If I first save the key in the main function, and then save it again somewhere else, it will work flawlessly. Sometimes it does not hang but I will get error -116.

Im not 100% sure if this really is a bug, but i think its super strange and I cannot imagine what is happening.

martinjaeger, 2020-09-01:
@CodingGhost thanks for the potential bug report. Can you please post a minimum working example that has this issue so that others can easily compile the code to reproduce it?

CodingGhost, 2020-09-03:
Ok, I made a small test: https://pastebin.com/32Ufzgy7
but now, as soon as I try to use the testfunc anywhere I get an MPU fault?!

adding to that, I just tried the zephyr supplied settings sample, and I get nothing but errors...?! I used the same config and the same code, so how is that possible?? only thing I added was
CONFIG_USE_SEGGER_RTT=y
CONFIG_RTT_CONSOLE=y
CONFIG_UART_CONSOLE=n

snippet of my output of the example:
`save all key-value pairs using registered handlers
00>
00> load all key-value pairs using registered handlers
00>
00> =================================================
00> loading subtree to destination provided by the caller
00>
00>   direct.length = ERR
00>   direct.length_1 = 828
00>   direct.length_2 = 536871624
00>
00> =================================================
00> Delete a key-value pair
00>
00> delete <alpha/length>: fail (err -2)
00>   Can't to load the <alpha/length> value as expected`

am I just stupid or is there something strange going on?

CodingGhost, 2020-09-03:
`	LOG_INF("--ZepyrOS booted succesfully--");
	int e = settings_subsys_init();
	LOG_DBG("setting init err=%d",e);
	e = settings_register(&my_conf);
	LOG_DBG("setting register err=%d",e);

	// k_timer_init(&periodictimer, timerfunc, NULL);
	// k_timer_start(&periodictimer,K_SECONDS(1),K_SECONDS(1));

	settings_load();
	int a; 
	settings_save_one("system/test_a",&a,sizeof(a));`

`00> [00:00:00.000,000] <inf> main: --ZepyrOS booted succesfully-- 00> [00:00:00.001,556] <inf> fs_nvs: 8 Sectors of 4096 bytes 00> [00:00:00.001,556] <inf> fs_nvs: alloc wra: 0, ff0 00> [00:00:00.001,586] <inf> fs_nvs: data wra: 0, 0 00> [00:00:00.001,586] <dbg> main.main: setting init err=0 00> [00:00:00.001,586] <dbg> main.main: setting register err=0 00> [00:00:00.001,647] <err> os: ***** MPU FAULT ***** 00> [00:00:00.001,647] <err> os:   Data Access Violation 00> [00:00:00.001,647] <err> os:   MMFAR Address: 0xf8000 00> [00:00:00.001,678] <err> os: r0/a1:  0x000f8000  r1/a2:  0xe000ed00  r2/a3:  0x4001e000 00> [00:00:00.001,678] <err> os: r3/a4:  0x80000000 r12/ip:  0xffffffff r14/lr:  0x00000467 00> [00:00:00.001,678] <err> os:  xpsr:  0x81000000 00> [00:00:00.001,678] <err> os: Faulting instruction address (r15/pc): 0x0000040a`

CodingGhost, 2020-09-03:
OK. I figured it out, its a mix of stupidity and bad documentation:

the first thing I was missing was CONFIG_NVS=y.
but this wasnt enought, I also needed to add CONFIG_ARM_MPU=n, which was not easy to find out.

then everything worked except for that I couldnt get it running inside a timer function. That is because the timer function executes in ISR context, which I didnt know and is described nowhere I looked. Maybe this is clear as day for everyone deeply involved with zephyr, but Im coming from mbed, and there are timer functions that execute on the main thread and not on ISR context. 
maybe add this info to https://docs.zephyrproject.org/1.12.0/kernel/timing/timers.html

after I changed my code to submit work items from the timer ISR, and execute my code inside the system workqueue, everything now works as expected!

closed 2020-09-03 by CodingGhost

arvid-r, 2020-09-11:
Is  `CONFIG_ARM_MPU=n` really needed?
Maybe `CONFIG_MPU_ALLOW_FLASH_WRITE=y` will work?