zephyrproject-rtos/zephyr #27934: Tests ignore custom board config overlays

rerickson1, 2020-09-01:
**Describe the bug**
If you place a boards/my_board.conf in a test folder it is ignored if the test case calls out extra_args: CONF_FILE=come_config.conf.

**To Reproduce**
Steps to reproduce the behavior:
1. Use `sensors.build.a_h:` in `tests/drivers/build_all/testcase.yaml` as an example.
2. place `boards/my_board.conf` file along side `tests/drivers/build_all/testcase.yaml`, your board config will be ignored and not merged.

**Expected behavior**
Board config overlays should never be ignored.

**Impact**
showstopper

**Environment (please complete the following information):**
CI system on Github.

chrta, 2020-09-01:
This is the documented behaviour, see https://docs.zephyrproject.org/latest/guides/kconfig/setting.html?highlight=conf_file#the-initial-configuration

rerickson1, 2020-09-01:
> This is the documented behaviour, see https://docs.zephyrproject.org/latest/guides/kconfig/setting.html?highlight=conf_file#the-initial-configuration

Thanks for pointing that out.  How can we deal with this use case though? One option I see is editing the sample or test cmake file to include a board specific config...

carlescufi, 2020-09-01:
@tejlmand is this something you could look into?

rerickson1, 2020-09-01:
In #27755 I came across a lot of samples or tests where I just need to increase `CONFIG_MAX_THREAD_BYTES` for my board.

tejlmand, 2020-09-02:
@rerickson1 Thanks for raising this.
Last week I actually noticed this myself, but there is a risk in simply changing current behavior.

There might be downstream users of Zephyr who relies on current behaviour, so that when they simply do:
```
cmake -DBOARD=my_board -Bbuild my_sample
```
then they know any board overlays will be included, but when they do:
```
cmake -DBOARD=my_board -Bbuild -DCONF_FILE=test_settings.conf my_sample
```
then they know that no overlays are included, so if we suddenly start including board overlays, then it might create a lot of issues to downstream users.

What I have been considering, is to extent the current functionality, so that if you specify a conf file following this pattern `prj_<confname>.conf`, then any board overlay of the form `<board>_<confname>.conf` will be considered.

That way you could have:
```
prj.conf
prj_debug.conf
boards/pinnacle_100_dvk.conf
boards/pinnacle_100_dvk_debug.conf
```

and then when using:
```
cmake -DBOARD=pinnacle_100_dvk -Bbuild <sample>
```
then `prj.conf` and `boards/pinnacle_100_dvk.conf` are used, and when doing:
```
cmake -DBOARD=pinnacle_100_dvk -Bbuild -CONFFILE=prj_debug.conf <sample>
```
then `prj_debug.conf` and `boards/pinnacle_100_dvk_debug.conf` are used.

tejlmand, 2020-09-02:
Note, this is more an enhancement request than a bug.

rerickson1, 2020-09-02:
Thanks, that makes sense. I agree that extending the functionality as you describe is probably the best approach.

tejlmand, 2020-09-03:
Fixed here: https://github.com/zephyrproject-rtos/zephyr/pull/28039

PR #28039 (2020-09-04): cmake: extend support for conf file and board overlays
Fixes: #27934

This commit introduces additional naming support of config files.

It is now possible to create a config file of the form
`prj_<build>.conf` and automatically have a corresponding
`boards/<BOARD>_<build>.conf` config file merged.

As example, one can create a structure as:
```
|-- prj.conf
|-- prj_debug.conf
|-- boards
      |-- nrf52840dk_nrf52840.conf
      |-- nrf52840dk_nrf52840_debug.conf
```

when building: (existing behavior)
`cmake -DBOARD=nrf52840dk_nrf52840 ...`
prj.conf is merged with nrf52840dk_nrf52840.conf

whe [... cut]
commit 212c3364 (2020-09-04): cmake: extend support for conf file and board overlays

Fixes: #27934

This commit introduces additional naming support of config files.

It is now possible to create a config file of the form
`prj_<build>.conf` and automatically have a corresponding
`boards/<BOARD>_<build>.conf` config file merged.

As example, one can create a structure as:
|-- prj.conf
|-- prj_debug.conf
|-- boards
      |-- nrf52840dk_nrf52840.conf
      |-- nrf52840dk_nrf52840_debug.conf

when building: (existing behavior)
cmake -DBOARD=nrf52840dk_nrf52840 ...
prj.conf is merged with nrf52840dk_nrf52840.conf

when buildin [... cut]
closed 2020-09-04 by carlescufi