zephyrproject-rtos/zephyr #27943: tests/kernel/sched/schedule_api fails on nsim_hs_smp

andrewboie, 2020-09-01:
```
$ sanitycheck -T tests/kernel/sched/schedule_api/ --emulation-only -a arc
Deleting output directory /dev/shm/sanity-out
INFO    - JOBS: 48
INFO    - Selecting emulation platforms per test case
INFO    - Building initial testcase list...
INFO    - 96 test configurations selected, 68 configurations discarded due to filters.
INFO    - Adding tasks to the queue...
INFO    - Total complete:   24/  28  85%  skipped:    0, failed:    0
ERROR   - nsim_hs_smp               tests/kernel/sched/schedule_api/kernel.scheduler.multiq_no_timeslicing FAILED: Failed
ERROR   - see: /dev/shm/sanity-out/nsim_hs_smp/tests/kernel/sched/schedule_api/kernel.scheduler.multiq_no_timeslicing/handler.log
INFO    - Total complete:   25/  28  89%  skipped:    0, failed:    1
ERROR   - nsim_hs_smp               tests/kernel/sched/schedule_api/kernel.scheduler   FAILED: Failed
ERROR   - see: /dev/shm/sanity-out/nsim_hs_smp/tests/kernel/sched/schedule_api/kernel.scheduler/handler.log
INFO    - Total complete:   26/  28  92%  skipped:    0, failed:    2
ERROR   - nsim_hs_smp               tests/kernel/sched/schedule_api/kernel.scheduler.multiq FAILED: Failed
ERROR   - see: /dev/shm/sanity-out/nsim_hs_smp/tests/kernel/sched/schedule_api/kernel.scheduler.multiq/handler.log
INFO    - Total complete:   27/  28  96%  skipped:    0, failed:    3
ERROR   - nsim_hs_smp               tests/kernel/sched/schedule_api/kernel.scheduler.no_timeslicing FAILED: Failed
ERROR   - see: /dev/shm/sanity-out/nsim_hs_smp/tests/kernel/sched/schedule_api/kernel.scheduler.no_timeslicing/handler.log
INFO    - Total complete:   28/  28  100%  skipped:    0, failed:    4
INFO    - 24 of 28 tests passed (85.71%), 4 failed, 68 skipped with 0 warnings in 37.52 seconds
INFO    - In total 484 test cases were executed on 24 out of total 282 platforms (8.51%)
INFO    - 28 tests executed on platforms, 0 tests were only built.
```

nashif, 2020-09-16:
is this test result valid given that mdb is required here?

ruuddw, 2020-09-16:
yes, it is a valid issue. See #28178

PR #28178 (2020-09-16): soc: arc: Increase cpu frequency for nsim_hs_smp
0.5 Mhz with 100 ticks per sec leaves 5000 cycles per tick,
which broke some tests that assumed more work within 1 tick.
Set to 1 Mhz: balance multi-core simulation speed and tick duration.

Fixes #27943
commit 0076313b (2020-09-16): soc: arc: Increase cpu frequency for nsim_hs_smp

0.5 Mhz with 100 ticks per sec leaves 5000 cycles per tick,
which broke some tests that assumed more work within 1 tick.
Set to 1 Mhz: balance multi-core simulation speed and tick duration.

Fixes #27943
closed 2020-09-16 by MaureenHelm