zephyrproject-rtos/zephyr #27956: CI: Add support to run external programs necessary for a test

ceolin, 2020-09-02:
**Is your feature request related to a problem? Please describe.**

In order to test gdbstub implementation is necessary to run the test itself and another process with gdb to attach the test. Currently there is no such way to do this in Zephyr's CI. Although this is describing the need for testing gdbstub there are possible several other cases where this can be helpful like, for example, an application that tests the network stack send unusual or malformed tailored packet to check corner cases.

**Describe the solution you'd like**
Allow an application run together with a test

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
https://github.com/zephyrproject-rtos/zephyr/pull/27314

nashif, 2020-09-02:
@jocelyn-li  FYI

PerMac, 2023-05-10:
I linked #57117 as a fix since it basically fulfils the request from the title.

PerMac, 2023-05-10:
e.g. we already are working on using mcumgr during pytest tests. Other prototype I saw at work with our tools was end2end tests for llwm2m, where a dut (can be qemu/hw/etc) runs client app and in addition a virtual server is run in the test. Test is verifying if the commands sent to the client are correctly addressed by the server.

closed 2023-05-26 by nashif