zephyrproject-rtos/zephyr #27975: [Thingy52_nrf52832 board] - Working with other led than led0

tlr-id, 2020-09-02:
# Context

I'm testing ZephyrRTOS with a Nordic Semiconductor dev kit : ```thingy52```.
The board is already supported by ZephyrRTOS and some samples/basic examples are working fine.

However there's an issue with samples/basic/blinky. It works fine with led0, but it doesn't work with led1 or led2.

# Explanation

This is working : 

```C
/* The devicetree node identifier for the "led0" alias. */
#define LED0_NODE DT_ALIAS(led0)
```

But this isnt (and it's not related to the fact that I defined my led as LED0_NODE) : 
```C
/* The devicetree node identifier for the "led1" alias. */
#define LED0_NODE DT_ALIAS(led1)
```
# Solution found : 

It appears that in ```zephyrproject/zephyr/boards/arm/thingy52_nrf52832/thingy52_nrf52832.dts```, the aliases for led1 and led2 aren't declared.
Just adding the next lines should make blinky work fine for every led.
```C
led1 = &led1;
led2 = &led2;
```

nashif, 2020-10-01:
@tlr-id  can you submit a PR with this change please?

gmarull, 2022-08-11:
led1/2 exists as aliases in the board now, closing.

closed 2022-08-11 by gmarull