zephyrproject-rtos/zephyr #27982: TCP2: Apparent issues with client-side connections (hangs when server (apparently) closes connection).

pfalcon, 2020-09-02:
This is a kind of dual to #27876. While that dealt with Zephyr-as-server case, this is about Zephyr-as-client case, and `samples/net/sockets/big_http_download` sample.

When running that sample (frdm_k64f tested so far), I get:

~~~
Iteration 1 of INF
sock = 0
[00:00:03.000,000] <inf> net_config: Interface 1 (0x20000ef0) coming up
[00:00:03.001,000] <inf> eth_mcux: ETH_0 enabled 100M full-duplex mode.
[00:00:03.001,000] <inf> net_config: IPv4 address: 192.0.2.1
[00:00:03.001,000] <inf> net_config: Running dhcpv4 client...
Download progress: 7015896 Bytes; 6851 KiB; 6 MiB
~~~

At which point it hangs. Looks like another issue with FIN handling.

I tested this with and without https://github.com/zephyrproject-rtos/zephyr/pull/27960, the behavior is the same.

pfalcon, 2020-09-02:
Ah, forgot to mention that:

> Download progress: 7015896 Bytes

is exactly the size of the file being downloaded, http://archive.ubuntu.com:80/ubuntu/dists/xenial/main/installer-amd64/current/images/hd-media/vmlinuz

jukkar, 2020-09-02:
Not a high issue as there is a workaround.

pfalcon, 2020-09-03:
To confirm, the issue also happens with qemu_x86.

PR #28011 (2020-09-03): net: tcp2: Fix connection close ack values
If we receive a TCP segment with FIN | ACK | PSH flags, then
update the ack values properly.

Fixes #27982
commit 9d064e6a (2020-09-03): net: tcp2: Fix connection close ack values

If we receive a TCP segment with FIN | ACK | PSH flags, then
update the ack values properly.

Fixes #27982
closed 2020-09-03 by nashif