zephyrproject-rtos/zephyr #27985: change in device initialization behavior

pabigot, 2020-09-02:
There appears to be an unintended behavioral difference resulting from merging #24873: in the past some attempts to retrieve the device pointer for a device that had not yet been initialized succeeded, while they now fail.

The case is demonstrated for `tests/drivers/eeprom` on `native_posix`.  The I2C device initializes at `POST_KERNEL` with priority `CONFIG_I2C_INIT_PRIORITY=60`, while the EEPROM initializes at `POST_KERNEL` with priority `CONFIG_KERNEL_INIT_PRIORITY_DEVICE=50`.  After the merge the attempt by the EEPROM init function to retrieve its parent bus fails because the returned pointer is null as the I2C controller has not yet been initialized.  (The EEPROM driver does not use the I2C bus during initialization.)

I don't think this ever should have worked, and a fix will be to correct the init priorities in this specific case, but it suggests that this will occur in other cases and may be difficult to detect.

pabigot, 2020-09-02:
@tbursztyka @carlescufi @nashif for awareness.

carlescufi, 2020-09-04:
@MaureenHelm @pabigot should we try and fix those before 2.4 is released?

pabigot, 2020-09-08:
Triage: Ensure this change is documented in the release notes.  Proposal was to also `__ASSERT()` if `device_get_binding()` is invoked before the device had been initialized, however the device initialization state is a boolean that cannot distinguish "not yet initialized" from "initialization failed", so we can't do that.

Formerly it was a functionally boolean that could not distinguish "not yet initialized" from "initialization succeeded", but that's a more dangerous confusion.

pabigot, 2020-09-10:
Ropened due to heightened concern about the impact of the change in behavior.

pabigot, 2020-09-14:
Since we keep playing whack-a-mole with this, and some of the solutions (e.g. #28339) have their own weaknesses, I've pushed #28369 as a potential solution for 2.4.

I think we should spend some 2.5 development effort on maintaining device state with higher fidelity.  We should be able to get a handle for a device before it's been initialized, to break dependency loops.  We should be able to delay initializing (`probe()` in Linux-speak) a device until it's going to be used (relates to #19448).  We should be able to shutdown and restart devices, potentially as part of device power management (also relates to #19448).

All of this is IMO appropriate to be part of device power management.

tbursztyka, 2020-09-15:
@pabigot I was going to propose the same thing, let's keep the old behavior in 2.4, and revert for 2.5 in order to fix found priority issues.

PR #28198 (2020-09-09): doc: release: 2.4: note behavior change in device_get_binding
Document that some initialization orders that used to work (and shouldn't have) will no longer work.

Fixes #27985
PR #28369 (2020-09-15): kernel: device: invert sense of ready bit
The device status ready bit replaced the previous hack of clearing the device API pointer when initialization of the device failed.  It inadvertently changed the behavior of device_get_binding() when invoked before the device was initialized: previously that succeeded for uninitialized devices, after the change it failed.

Multiple driver initializations rely on being able to get a device pointer for something they're going to depend on in their init function, even if that device has not yet been initialized.  Although this is wrong, and would cause fa [... cut]
closed 2020-09-15 by carlescufi