zephyrproject-rtos/zephyr #27992: stm32f7: usb: Bursting HID Get and Set report requests leads to unresponding Control endpoint.

AntoineGL, 2020-09-02:
**Describe the bug**
There are 2 alternatives for this bug:
 - Hard bug: the board no longer responds to any Get or Set report request (timeout on host), a reset is needed. Disconnect/Reconnect the USB cable without resetting will finally put the USB status to "reset" and never "configured".
Note that the application still runs and it is still possible to exchange packets through interrupt endpoint.
 - Light bug: the board occasionally doesn't respond to the current Set report request (timeout on host).

Bursting alternatively Get -> Set -> Get -> Set .... report requests will trigger the Hard bug if the set report size is large enough (>=109 bytes in my case). It will trigger the Light bug otherwise.
Bursting only Set report request seems to trigger only the Light bug.
Bursting only Get report seems to work right.

Enabling USB driver and USB device debug logging will trigger both bugs much faster.

**To Reproduce**
Ressources on branch: https://github.com/AntoineGL/zephyr/tree/usb_hid_issue
1. Enable USB driver and USB device debug logging (with high internal buffer) to trigger the bug faster.
2. Pick the given sample (hid_issue_sample/) and flash it. You can adjust the Get report request report size by editing "#define GET_REPORT_SIZE      128"
3. Use a host application to burst the device with get/set report requests: you can pick the given host application (hid_issue_sample/host_application/) working with Linux hidraw driver:
-> Locate your device hidraw location (/dev/hidrawX) and run the application:
$ cd samples/subsys/usb/hid_issue_sample/host_application/
$ make
$ ./hid-issue /dev/hidrawX alt 128
(You can adjust the Set report size with the third argument. You can change the pattern 'alt' (for alternate), with 'get' (resp. 'set') to only burst get (resp. set) report requests) 
-> Fast enough, you should see some timeout errors.
-> Press ctrl-c to stop the application and see how many errors occurred on the total of requests.

**Expected behavior**
The device should respond to all get/set report requests.

**Impact**
The device no longer responds to get/set report requests, and should be reset for the Hard bug.

**Logs and console output**

For a working Set + Get logs (with get/set report size = 128):

```
[00:00:02.723,000] <dbg> usb_dc_stm32.HAL_PCD_SetupStageCallback: 
[00:00:02.723,000] <dbg> usb_device.usb_handle_control_transfer: ep 0x00, status 0x00
[00:00:02.723,000] <dbg> usb_dc_stm32.usb_dc_ep_read_wait: ep 0x00, 8 bytes, 0+8, 0x20021f98
[00:00:02.723,000] <dbg> usb_dc_stm32.usb_dc_ep_start_read: ep 0x00, len 64
[00:00:02.723,000] <dbg> usb_dc_stm32.usb_dc_ep_start_read: ep 0x00, len 128
[00:00:02.723,000] <dbg> usb_dc_stm32.HAL_PCD_DataOutStageCallback: epnum 0x00, rx_count 64
[00:00:02.724,000] <dbg> usb_device.usb_handle_control_transfer: ep 0x00, status 0x01
[00:00:02.724,000] <dbg> usb_dc_stm32.usb_dc_ep_read_wait: ep 0x00, 128 bytes, 0+64, 0x20020558
[00:00:02.724,000] <dbg> usb_dc_stm32.usb_dc_ep_start_read: ep 0x00, len 64
[00:00:02.724,000] <dbg> usb_dc_stm32.HAL_PCD_DataOutStageCallback: epnum 0x00, rx_count 64
[00:00:02.724,000] <dbg> usb_device.usb_handle_control_transfer: ep 0x00, status 0x01
[00:00:02.724,000] <dbg> usb_dc_stm32.usb_dc_ep_read_wait: ep 0x00, 64 bytes, 0+64, 0x20020598
[00:00:02.724,000] <dbg> usb_dc_stm32.usb_dc_ep_start_read: ep 0x00, len 64
[00:00:02.724,000] <dbg> usb_device.class_handler: bRequest 0x09, wIndex 0x0000
[00:00:02.724,000] <dbg> usb_hid.hid_class_handle_req: Class request:bRequest 0x09, bmRequestType 0x21 len 128
[00:00:02.724,000] <wrn> main: Set cb 128
[00:00:02.724,000] <dbg> usb_device.usb_handle_control_transfer: >> usb_data_to_host(2)
[00:00:02.724,000] <dbg> usb_dc_stm32.usb_dc_ep_write: ep 0x80, len 0
[00:00:02.724,000] <dbg> usb_dc_stm32.HAL_PCD_DataInStageCallback: epnum 0x00
[00:00:02.724,000] <dbg> usb_device.usb_handle_control_transfer: ep 0x80, status 0x02
[00:00:02.724,000] <dbg> usb_dc_stm32.HAL_PCD_SetupStageCallback: 
[00:00:02.724,000] <dbg> usb_device.usb_handle_control_transfer: ep 0x00, status 0x00
[00:00:02.724,000] <dbg> usb_dc_stm32.usb_dc_ep_read_wait: ep 0x00, 8 bytes, 0+8, 0x20021f98
[00:00:02.724,000] <dbg> usb_dc_stm32.usb_dc_ep_start_read: ep 0x00, len 64
[00:00:02.724,000] <dbg> usb_device.class_handler: bRequest 0x01, wIndex 0x0000
[00:00:02.724,000] <dbg> usb_hid.hid_class_handle_req: Class request:bRequest 0x01, bmRequestType 0xa1 len 4095
[00:00:02.724,000] <wrn> main: Get cb
[00:00:02.724,000] <dbg> usb_dc_stm32.usb_dc_ep_write: ep 0x80, len 128
[00:00:02.724,000] <dbg> usb_dc_stm32.usb_dc_ep_start_read: ep 0x80, len 0
[00:00:02.724,000] <dbg> usb_dc_stm32.HAL_PCD_DataInStageCallback: epnum 0x00
[00:00:02.724,000] <dbg> usb_device.usb_handle_control_transfer: ep 0x80, status 0x02
[00:00:02.724,000] <dbg> usb_dc_stm32.usb_dc_ep_write: ep 0x80, len 64
[00:00:02.724,000] <dbg> usb_dc_stm32.usb_dc_ep_start_read: ep 0x80, len 0
[00:00:02.724,000] <dbg> usb_device.usb_data_to_host: ZLP, requested 4095 , length 128 
[00:00:02.724,000] <dbg> usb_dc_stm32.HAL_PCD_DataInStageCallback: epnum 0x00
[00:00:02.724,000] <dbg> usb_device.usb_handle_control_transfer: ep 0x80, status 0x02
[00:00:02.724,000] <dbg> usb_dc_stm32.usb_dc_ep_write: ep 0x80, len 0
[00:00:02.724,000] <dbg> usb_dc_stm32.HAL_PCD_DataInStageCallback: epnum 0x00
[00:00:02.724,000] <dbg> usb_device.usb_handle_control_transfer: ep 0x80, status 0x02
[00:00:02.724,000] <dbg> usb_dc_stm32.HAL_PCD_DataOutStageCallback: epnum 0x00, rx_count 0
[00:00:02.724,000] <dbg> usb_device.usb_handle_control_transfer: ep 0x00, status 0x01
[00:00:02.724,000] <dbg> usb_dc_stm32.usb_dc_ep_read_wait: ep 0x00, 0 bytes, 0+0, 0x0800934c
[00:00:02.724,000] <dbg> usb_dc_stm32.usb_dc_ep_start_read: ep 0x00, len 64
```

On a Set report triggering the Hard bug (size = 128):
```
[00:00:02.731,000] <dbg> usb_dc_stm32.HAL_PCD_DataOutStageCallback: epnum 0x00, rx_count 136
[00:00:02.731,000] <dbg> usb_device.usb_handle_control_transfer: ep 0x00, status 0x01
[00:00:02.731,000] <dbg> usb_dc_stm32.usb_dc_ep_read_wait: ep 0x00, 0 bytes, 0+136, 0x0800934c
```
(Nothing more, the request timeouts on host. The following requests don't produce any logs and they timeout as well)

----------------------------------------------------------------------------------------------
If we summarize the beginning of a 128 bytes Set report handling:
when everything goes well (the first 13 lines):
- Setup packet callback (8 bytes)
- DataOUT packet callback (64 bytes)
- DataOUT packet callback (64 bytes)

When the Hard bug triggers:
- DataOUT packet callback (136 bytes)

This seems to be wrong, multiple packets are likely merged into the same buffer. Furthermore, when the bug occurs, we can observe that the HAL doesn't issue a USB_DC_EP_SETUP event (the call to HAL_PCD_SetupStageCallback is missing).

**Environment**
 Device:
- Board nucleo F767zi
- Zephyr OS version 2.3.99

Host:
- Ubuntu 18.04.5 LTS
- Kernel: 4.15.0-112-generic
- USB HID driver: hidraw
- Toolchain: Zephyr SDK

jfischer-phytec-iot, 2020-09-02:
@AntoineGL thanks for the report.

jfischer-phytec-iot, 2020-09-07:
I do not have nucleo_f767zi or an other STM32 board right now, so I tested it on nRF52840 and FRDM_K64F.

First, I only observed the errors for HIDIOCGFEATURE, therefore I changed hid-issue.c:

```diff
diff --git a/samples/subsys/usb/hid_issue_sample/host_application/hid-issue.c b/samples/subsys/usb/hid_issue_sample/host_application/hid-issue.c
index 66227dcbdb..a88b36f90b 100644
--- a/samples/subsys/usb/hid_issue_sample/host_application/hid-issue.c
+++ b/samples/subsys/usb/hid_issue_sample/host_application/hid-issue.c
@@ -84,17 +84,16 @@ int main(int argc, char **argv)
                usage();
                return 1;
        }
-       
-       if (type == SET || type == ALT) {
-               set_size = strtol(argv[3], NULL, 10);
-               if (set_size < 1 || set_size > SEND_REPORT_SIZE) {
-                       printf("Set size is incorrect\n");
-                       usage();
-                       return 1;
-               }
-               for (i = 0; i < SEND_REPORT_SIZE + 1; i++) {
-                       send_report[i] = (uint8_t)i;
-               }
+
+       set_size = strtol(argv[3], NULL, 10);
+       if (set_size < 1 || set_size > SEND_REPORT_SIZE) {
+               printf("Set size is incorrect\n");
+               usage();
+               return 1;
+       }
+
+       for (i = 0; i < sizeof(send_report); i++) {
+               send_report[i] = (uint8_t)i;
        }
 
        fd = open(argv[1], O_RDWR);
@@ -119,7 +118,7 @@ int main(int argc, char **argv)
                }
                if (type == GET || type == ALT) {
                        recv_report[0] = 0;
-                       if (ioctl(fd, HIDIOCGFEATURE(sizeof(recv_report)),
+                       if (ioctl(fd, HIDIOCGFEATURE(set_size + 1),
                                  recv_report) < 0) {
                                perror("Get feature report");
                                error_count++;
```

With the change above I have **not** observed any errors on nRF52840 and K64F.

Minor changes:
```diff

diff --git a/samples/subsys/usb/hid_issue_sample/main.c b/samples/subsys/usb/hid_issue_sample/main.c
index 556264dfa4..2d59e88ec1 100644
--- a/samples/subsys/usb/hid_issue_sample/main.c
+++ b/samples/subsys/usb/hid_issue_sample/main.c
@@ -8,7 +8,7 @@
 
 #include <stdint.h>
 
-#define LOG_LEVEL LOG_LEVEL_WRN
+#define LOG_LEVEL LOG_LEVEL_INF
 LOG_MODULE_REGISTER(main);
 
 #define GET_REPORT_SIZE      128
@@ -17,7 +17,7 @@ LOG_MODULE_REGISTER(main);
 #define HID_MI_FEATURE       0xB1
 #define HID_GI_REPORT_COUNT2 0x96 /* Report count on 2 bytes */
 
-static struct device *hdev;
+static const struct device *hdev;
 
 const uint8_t hid_report_desc[] = {
 HID_GI_USAGE_PAGE, USAGE_GEN_DESKTOP,              /*USAGE_PAGE (Gen Desktop) */
@@ -34,7 +34,7 @@ HID_MI_COLLECTION_END,                             /*END_COLLECTION           */
 
 const uint16_t hid_report_desc_size = sizeof(hid_report_desc);
 
-static const uint8_t ret_report[GET_REPORT_SIZE]= {0};
+static uint8_t ret_report[GET_REPORT_SIZE]= {0};
 
 static void status_cb(enum usb_dc_status_code status, const uint8_t *param)
 {
@@ -50,20 +50,24 @@ static void status_cb(enum usb_dc_status_code status, const uint8_t *param)
        }
 }
 
-int test_get_report(struct usb_setup_packet *setup, int32_t *len,
+static int test_get_report(struct usb_setup_packet *setup, int32_t *len,
                uint8_t **data)
 {
-       LOG_DBG("Get cb");
+       LOG_INF("Get cb");
        *data = ret_report;
        *len = GET_REPORT_SIZE;
-       return 0;
 
+       return 0;
 }
 
-int test_set_report(struct usb_setup_packet *setup, int32_t *len,
+static int test_set_report(struct usb_setup_packet *setup, int32_t *len,
                uint8_t **data)
 {
-       LOG_DBG("Set cb %d", *len);
+       size_t rsp_len = MIN(*len, sizeof(ret_report));
+
+       LOG_DBG("Set cb, length %u", rsp_len);
+       memcpy(ret_report, *data, rsp_len);
+
        return 0;
 }
 
@@ -82,7 +86,7 @@ void main(void)
 
 }
 
-static int composite_pre_init(struct device *dev)
+static int composite_pre_init(const struct device *dev)
 {
        hdev = device_get_binding("HID_0");
        if (hdev == NULL) {
```

```diff
diff --git a/samples/subsys/usb/hid_issue_sample/prj.conf b/samples/subsys/usb/hid_issue_sample/prj.conf
index 12453f9c56..883e98a76e 100644
--- a/samples/subsys/usb/hid_issue_sample/prj.conf
+++ b/samples/subsys/usb/hid_issue_sample/prj.conf
@@ -4,5 +4,5 @@ CONFIG_USB_DEVICE_HID=y
 CONFIG_USB_DEVICE_PRODUCT="Zephyr HID issue sample"
 
 CONFIG_LOG=y
-CONFIG_USB_DRIVER_LOG_LEVEL_DBG=y
-CONFIG_USB_DEVICE_LOG_LEVEL_DBG=y
+CONFIG_USB_DRIVER_LOG_LEVEL_WRN=y
+CONFIG_USB_DEVICE_LOG_LEVEL_WRN=y

```

jfischer-phytec-iot, 2020-09-07:
Works for me on **nucleo_wb55rg**.

jfischer-phytec-iot, 2020-09-09:
Works on nucleo_f413zh. @AntoineGL please retest with the correct size for `ioctl(fd, HIDIOCGFEATURE(...)..)`

AntoineGL, 2020-09-09:
Hi @jfischer-phytec-iot , tested with getting 129 bytes (including 1 for report ID), doesn't change anything.
Here's the usbmon trace showing that it requires and sends 128 bytes:
```
# Working SET then GET
ffff93cc20a406c0 1416157716 S Co:1:029:0 s 21 09 0300 0000 0080 128 = 01020304 05060708 090a0b0c 0d0e0f10 11121314 15161718 191a1b1c 1d1e1f20
ffff93cc20a406c0 1416158176 C Co:1:029:0 0 128 >
ffff93cc20a406c0 1416158207 S Ci:1:029:0 s a1 01 0300 0000 0080 128 <
ffff93cc20a406c0 1416158601 C Ci:1:029:0 0 128 = 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
# First timeout on SET then GET
ffff93cc20a406c0 1416158636 S Co:1:029:0 s 21 09 0300 0000 0080 128 = 01020304 05060708 090a0b0c 0d0e0f10 11121314 15161718 191a1b1c 1d1e1f20
ffff93cc20a406c0 1421276078 C Co:1:029:0 -2 64 >
ffff93cb66c77d80 1421276269 S Ci:1:029:0 s a1 01 0300 0000 0080 128 <
ffff93cb66c77d80 1426396207 C Ci:1:029:0 -2 0
```
Note that:
- On the light bug, I never have a timeout on a GET 
- On the hard bug, GETs only timeout after a timeout on a SET. (It is always a SET that triggers the hard bug)
I see that you disabled usb driver and usb device debug log, you should not. In my case, it helps a lot triggering the bug.

jfischer-phytec-iot, 2020-09-09:
@AntoineGL well it is then specific to your SoC support in usb_dc_stm32 driver.

carlescufi, 2020-09-09:
@erwango this seems related to the USB device driver for STM32.

erwango, 2020-09-09:
^^ @ydamigos @loicpoulain would one of you have time to look at this issue ?

carlescufi, 2021-02-09:
> ^^ @ydamigos @loicpoulain would one of you have time to look at this issue ?

@erwango it doesn't look like Yannis or Loic are going to be able to look at it. Is this something you could try to reproduce?

loicpoulain, 2021-02-10:
Is it reproducible with an existing (master) sample?

mbittan, 2021-03-07:
@loicpoulain It seems that the tests 14 and 21 from testusb are both able to trigger this bug. Indeed, these tests often result in timeouts when using the following testusb command:
```
sudo ./testusb -D /dev/usb/xxx/yyy -v 512 -t 14
```

Another way to reproduce this issue is by using an updated version of the sample written by @AntoineGL. Here is the associated patch: https://github.com/mbittan/zephyr/commit/6a468b851aa1ce41c69588ad3a01a30cbffdcd97.patch

I have found that the tool provided with the sample can trigger the bug often and relatively quickly, using the following parameters:
```
sudo ./hid-issue  /dev/hidrawX alt 64
```
I've been able to reproduce this issue on the nucleo_f767zi and another board based on the STM32F070 MCU.

closed 2021-08-01 by github-actions[bot]

erwango, 2021-08-23:
@simonguinot don't hesitate to reopen if required