zephyrproject-rtos/zephyr #27999: Add QSPI tests for microchip mec1521 board, drivers are in zephyr/drivers/spi, to be tested on mec15xxevb_assy6853

ashaannajoy, 2020-09-03:
**Is your enhancement proposal related to a problem? Please describe.**
A clear and concise description of what the problem is.
We are finding issues when we use qspi apis in our app. So requesting for qspi tests to be added for microchip mec1521 under drivers/tests/spi

henrikbrixandersen, 2020-09-03:
Could you please elaborate a bit on which APIs you are using and what errors you are seeing?

ashaannajoy, 2020-09-17:
Suggest to test spi_configure and spi_transceive sending some flash commands.

See example on samples SAF for an way to use SPI API on EVB, send commands and performing read & write. 
https://github.com/zephyrproject-rtos/zephyr/pull/27532/commits/a57ec86701f4769a30b643b9d52d19876a0f7be0

chen-png, 2020-09-24:
@ashaannajoy   could you take a look of this #17902, the QSPI enhancement discussion

chen-png, 2021-02-04:
the patch is merged, and the new testcase "tests/boards/mec15xxevb_assy6853/qspi" worked well on microchip board, so closed this.

closed 2021-02-04 by chen-png