zephyrproject-rtos/zephyr #28012: tests/net/lib/mqtt_subscriber/net.mqtt.subscriber fails to build on cc3220sf_launchxl

galak, 2020-09-03:
Running:

`./scripts/sanitycheck -p cc3220sf_launchxl -s tests/net/lib/mqtt_subscriber/net.mqtt.subscriber`

I get:

```
[ 89%] Building C object modules/ti/simplelink/CMakeFiles/..__modules__hal__ti__simplelink.dir/kernel/zephyr/dpl/dpl.c.obj
In file included from /home/galak/git/zephyr/include/device.h:22,
                 from /home/galak/git/modules/hal/ti/simplelink/kernel/zephyr/dpl/dpl.c:12:
/home/galak/git/zephyr/include/init.h:89:11: error: initialization of 'int (*)(const struct device *)' from incompatible pointer type 'int (*)(struct device *)' [-Werror=incompatible-pointer-types]
   89 |   .init = (_init_fn),     \
      |           ^
/home/galak/git/zephyr/include/init.h:136:2: note: in expansion of macro 'Z_INIT_ENTRY_DEFINE'
  136 |  Z_INIT_ENTRY_DEFINE(Z_SYS_NAME(_init_fn), _init_fn, NULL, _level, _prio)
      |  ^~~~~~~~~~~~~~~~~~~
/home/galak/git/modules/hal/ti/simplelink/kernel/zephyr/dpl/dpl.c:63:1: note: in expansion of macro 'SYS_INIT'
   63 | SYS_INIT(dpl_zephyr_init, POST_KERNEL, CONFIG_KERNEL_INIT_PRIORITY_DEFAULT);
      | ^~~~~~~~
/home/galak/git/zephyr/include/init.h:89:11: note: (near initialization for '__init_sys_init_dpl_zephyr_init2.init')
   89 |   .init = (_init_fn),     \
      |           ^
/home/galak/git/zephyr/include/init.h:136:2: note: in expansion of macro 'Z_INIT_ENTRY_DEFINE'
  136 |  Z_INIT_ENTRY_DEFINE(Z_SYS_NAME(_init_fn), _init_fn, NULL, _level, _prio)
      |  ^~~~~~~~~~~~~~~~~~~
/home/galak/git/modules/hal/ti/simplelink/kernel/zephyr/dpl/dpl.c:63:1: note: in expansion of macro 'SYS_INIT'
   63 | SYS_INIT(dpl_zephyr_init, POST_KERNEL, CONFIG_KERNEL_INIT_PRIORITY_DEFAULT);
      | ^~~~~~~~
cc1: all warnings being treated as errors
````

pfalcon, 2020-09-04:
Also at https://ci.linaro.org/jenkins/job/zephyr-upstream/PLATFORM=cc3220sf_launchxl,ZEPHYR_TOOLCHAIN_VARIANT=zephyr,label=docker-xenial-amd64-13/5520/display/redirect . Apparently result of "device constify" patchset recently merged.

closed 2020-09-08 by carlescufi

vanti, 2020-09-08:
Thanks @pabigot for the fix. Looks like the "device constify" patchset was merged during my absence last week and led to the failure.

PR #28136 (2020-09-08): west.yml: update hal_ti for const device
An init signature needs a const qualifier.

Fixes #28012