zephyrproject-rtos/zephyr #28031: samples/subsys/mgmt/mcumgr/smp_svr/sample.mcumg.smp_svr.bt fails to build

galak, 2020-09-03:
When I run:

`./scripts/sanitycheck -p nrf52840dk_nrf52840 -s samples/subsys/mgmt/mcumgr/smp_svr/sample.mcumg.smp_svr.bt`

I get:

```
[ 77%] Building C object zephyr/subsys/mgmt/mcumgr/CMakeFiles/subsys__mgmt__mcumgr.dir/buf.c.obj
[ 77%] Building C object zephyr/subsys/mgmt/mcumgr/CMakeFiles/subsys__mgmt__mcumgr.dir/smp.c.obj
[ 78%] Building C object zephyr/subsys/mgmt/mcumgr/CMakeFiles/subsys__mgmt__mcumgr.dir/smp_bt.c.obj
[ 78%] Building C object zephyr/subsys/mgmt/mcumgr/CMakeFiles/subsys__mgmt__mcumgr.dir/smp_shell.c.obj
/home/galak/git/zephyr/subsys/mgmt/mcumgr/smp_shell.c: In function 'smp_shell_tx_raw':
/home/galak/git/zephyr/subsys/mgmt/mcumgr/smp_shell.c:145:38: error: initialization of 'const struct shell_uart * const' from incompatible pointer type 'const struct shell *' [-Werror=incompatible-pointer-types]
  145 |  const struct shell_uart *const su = shell_backend_uart_get_ptr();
      |                                      ^~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
gmake[2]: *** [zephyr/subsys/mgmt/mcumgr/CMakeFiles/subsys__mgmt__mcumgr.dir/build.make:122: zephyr/subsys/mgmt/mcumgr/CMakeFiles/subsys__mgmt__mcumgr.dir/smp_shell.c.obj] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:3448: zephyr/subsys/mgmt/mcumgr/CMakeFiles/subsys__mgmt__mcumgr.dir/all] Error 2
gmake: *** [Makefile:104: all] Error 2
```

galak, 2020-09-03:
@carlescufi / @de-nordic not sure exactly who to assign this to, so added you guys as a start.

de-nordic, 2020-09-04:
Taking care of it.

de-nordic, 2020-09-04:
> 
> 
> @carlescufi / @de-nordic not sure exactly who to assign this to, so added you guys as a start.

@galak  If something is wrong with mcumgr or smp you may assign me to it. I will delegate if it is not for me.

PR #28048 (2020-09-04): mgmt: smp: Fix compilation problem
Fix problem with compilation due to incompatible type assignments.

Fixes #28031
commit 1d0fb461 (2020-09-04): mgmt: smp: Fix compilation problem

Fix problem with compilation due to incompatible type assignments.

Fixes #28031
closed 2020-09-04 by carlescufi