zephyrproject-rtos/zephyr #28033: rand32_ctr_drbg.c fails to build

galak, 2020-09-03:
When I try:

`./scripts/sanitycheck -p frdm_kw41z -s samples/net/sockets/websocket_client/sample.net.sockets.websocket_client`

I get:

```
/home/galak/git/zephyr/subsys/random/rand32_ctr_drbg.c: In function 'ctr_drbg_initialize':
/home/galak/git/zephyr/subsys/random/rand32_ctr_drbg.c:72:9: error: passing argument 3 of 'mbedtls_ctr_drbg_seed' discards 'const' qualifier from pointer target type [-Werror=discarded-qualifiers]
   72 |         entropy_driver,
      |         ^~~~~~~~~~~~~~
In file included from /home/galak/git/zephyr/subsys/random/rand32_ctr_drbg.c:19:
/home/galak/git/modules/crypto/mbedtls/include/mbedtls/ctr_drbg.h:278:26: note: expected 'void *' but argument is of type 'const struct device *'
  278 |                    void *p_entropy,
      |                    ~~~~~~^~~~~~~~~

```

dleach02, 2020-09-04:
Broke with the big PR to constify things... according to bisect (c8906fef79cefce3)

pfalcon, 2020-09-04:
> Would that not have been caught with CI?

Definitely was caught by the Linaro CI (I was just slow to triage failures, thanks @galak for beating me on that):
https://ci.linaro.org/jenkins/job/zephyr-upstream/PLATFORM=frdm_kw41z,ZEPHYR_TOOLCHAIN_VARIANT=zephyr,label=docker-xenial-amd64-13/5520/display/redirect

galak, 2020-09-04:
> Broke with the big PR to constify things... according to bisect ([c8906fe](https://github.com/zephyrproject-rtos/zephyr/commit/c8906fef79cefce3fe8c24cd0d01d164614d579f))
> 
> Would that not have been caught with CI?

Shows up in the nightly CI.

PR #28043 (2020-09-08): random: rand32_ctr_drbg: Fix missed device constify change
Device constify PR missed changes needed within the mbedtls conditional code.

Fixes #28033
commit a775d735 (2020-09-08): random: rand32_ctr_drbg: Fix missed device constify change

Device constify PR missed changes needed within the mbedtls
conditional code.

Fixes #28033
closed 2020-09-08 by carlescufi