zephyrproject-rtos/zephyr #28049: nucleo_wb55rg: test/spi/spi_loopback build failure

erwango, 2020-09-04:
**Describe the bug**
When building `test/spi/spi_loopback` with nucleo_wb55rg, build failure happens:
```
zephyr/drivers/dma/libdrivers__dma.a(dmamux_stm32.c.obj): in function `dmamux_stm32_init':
/local/mcu/zephyrproject/zephyr/drivers/dma/dmamux_stm32.c:155: undefined reference to `k_malloc'
```

**To Reproduce**
Steps to reproduce the behavior:
`sanitycheck -b -p nucleo_wb55rg -T tests/drivers/spi`

**Expected behavior**
Test should build

FRASTM, 2020-09-04:
The pb comes from the CONFIG_HEAP_MEM_POOL_SIZE not defined correctly
which comes from the `drivers/dma/Kconfig.stm32`

FRASTM, 2020-09-04:
config of the HEAP_MEM_POOL_SIZE depends on DMAMUX_STM32

PR #28065 (2020-09-04): drivers: dma: dmamux init needs k_malloc
The dmamux requires HEAP size definition, so that k_malloc
is valid. The HEAP size config is defined in the common for
any stm32 soc instead of specific to dma Kconfig

Fixes #28049
closed 2020-09-04 by MaureenHelm