zephyrproject-rtos/zephyr #28052: metairq_dispatch sample fails on nrf9160dk_nrf9160

PerMac, 2020-09-04:
**Describe the bug**
When calling sanitycheck to run metairq_dispatch sample on nrf9160dk_nrf9160 it hangs during the execution and finishes with a timeout. It works on other nrf platforms correctly. 

**To Reproduce**
Steps to reproduce the behavior:
1. go to zephyr dir
2. call `./scripts/sanitycheck --device-testing -T samples/scheduler/metairq_dispatch/ -p nrf9160dk_nrf9160 --device-serial /dev/ttyACM0 -v -v` 
3. See error

**Expected behavior**
The test passes on nrf9160dk_nrf9160

**Impact**
Unknown 

**Logs and console output**
<details><summary>console output (click to show)</summary>
<p>

```
INFO    - Building initial testcase list...
INFO    - 1 test configurations selected, 0 configurations discarded due to filters.

Device testing on:

| Platform          | ID   | Serial device   |
|-------------------|------|-----------------|
| nrf9160dk_nrf9160 |      | /dev/ttyACM0    |

INFO    - Adding tasks to the queue...
DEBUG   - Running cmake on /home/maciej/zephyrproject2/zephyr/samples/scheduler/metairq_dispatch for nrf9160dk_nrf9160
DEBUG   - Calling cmake with arguments: ['-B/home/maciej/zephyrproject2/zephyr/sanity-out/nrf9160dk_nrf9160/samples/scheduler/metairq_dispatch/sample.kernel.metairq_dispatch', '-S/home/maciej/zephyrproject2/zephyr/samples/scheduler/metairq_dispatch', '-DEXTRA_CFLAGS="-Werror"', '-DEXTRA_AFLAGS="-Wa,--fatal-warnings', '-DEXTRA_LDFLAGS="-Wl,--fatal-warnings"', '-GUnix Makefiles', '-DBOARD=nrf9160dk_nrf9160']
DEBUG   - DONE FEEDING
DEBUG   - Finished building /home/maciej/zephyrproject2/zephyr/samples/scheduler/metairq_dispatch for nrf9160dk_nrf9160
DEBUG   - build test: nrf9160dk_nrf9160/samples/scheduler/metairq_dispatch/sample.kernel.metairq_dispatch
DEBUG   - Building /home/maciej/zephyrproject2/zephyr/samples/scheduler/metairq_dispatch for nrf9160dk_nrf9160
DEBUG   - run test: nrf9160dk_nrf9160/samples/scheduler/metairq_dispatch/sample.kernel.metairq_dispatch
DEBUG   - Using serial device /dev/ttyACM0
DEBUG   - Flash command: ['make', '-C', '/home/maciej/zephyrproject2/zephyr/sanity-out/nrf9160dk_nrf9160/samples/scheduler/metairq_dispatch/sample.kernel.metairq_dispatch', 'flash']
DEBUG   - DEVICE: *** Booting Zephyr OS build zephyr-v2.3.0-2531-g4eba946e2bb1  ***
DEBUG   - DEVICE: I: Starting Thread0 at priority -2
DEBUG   - DEVICE: I: Starting Thread1 at priority -1
DEBUG   - DEVICE: I: StarI: M1 T1 mirq 2 disp 12 proc 1 real 1
DEBUG   - DEVICE: I: M3 T1 mirq 2 disp 221 proc 1 real 1
DEBUG   - DEVICE: I: M9 T1 mirq 2 disp 336 proc 2 real 2
DEBUG   - DEVICE: I: M17 T1 mirq 4 disp 453 proc 2 real 2
DEBUG   - DEVICE: I: M20 T1 mirq 2 disp 604 proc 0 real 0
DEBUG   - DEVICE: I: M22 T1 mirq 3 disp 754 proc 3 real 3
DEBUG   - DEVICE: I: M0 T0 mirq 2 disp 8 proc 3 real 3
DEBUG   - DEVICE: I: M5 T0 mirq 2 disp 1156 proc 3 real 3
DEBUG   - DEVICE: I: M6 T0 mirq 2 disp 1309 proc 3 real 3
DEBUG   - DEVICE: I: M7 T0 mirq 2 disp 1465 proc 0 real 0
DEBUG   - DEVICE: I: M10 T0 mirq 2 disp 1601 proc 2 real 2
DEBUG   - DEVICE: I: M11 T0 mirq 2 disp 1760 proc 0 real 0
DEBUG   - DEVICE: I: M13 T0 mirq 4 disp 1915 proc 2 real 2
DEBUG   - DEVICE: I: M14 T0 mirq 2 disp 2079 proc 1 real 1
DEBUG   - DEVICE: I: M16 T0 mirq 2 disp 2231 proc 0 real 0
DEBUG   - DEVICE: I: M18 T0 mirq 2 disp 2388 proc 1 real 1
DEBUG   - DEVICE: ting Thread2 at priority 0
DEBUG   - DEVICE: I: M4 T2 mirq 2 disp 2740 proc 0 real 0
DEBUG   - DEVICE: I: M8 T2 mirq 2 disp 2870 proc 1 real 1
DEBUG   - make: Entering directory '/home/maciej/zephyrproject2/zephyr/sanity-out/nrf9160dk_nrf9160/samples/scheduler/metairq_dispatch/sample.kernel.metairq_dispatch'
make[1]: Entering directory '/home/maciej/zephyrproject2/zephyr/sanity-out/nrf9160dk_nrf9160/samples/scheduler/metairq_dispatch/sample.kernel.metairq_dispatch'
make[2]: Entering directory '/home/maciej/zephyrproject2/zephyr/sanity-out/nrf9160dk_nrf9160/samples/scheduler/metairq_dispatch/sample.kernel.metairq_dispatch'
make[3]: Entering directory '/home/maciej/zephyrproject2/zephyr/sanity-out/nrf9160dk_nrf9160/samples/scheduler/metairq_dispatch/sample.kernel.metairq_dispatch'
make[3]: Leaving directory '/home/maciej/zephyrproject2/zephyr/sanity-out/nrf9160dk_nrf9160/samples/scheduler/metairq_dispatch/sample.kernel.metairq_dispatch'
[  2%] Built target parse_syscalls_target
make[3]: Entering directory '/home/maciej/zephyrproject2/zephyr/sanity-out/nrf9160dk_nrf9160/samples/scheduler/metairq_dispatch/sample.kernel.metairq_dispatch'
make[3]: Leaving directory '/home/maciej/zephyrproject2/zephyr/sanity-out/nrf9160dk_nrf9160/samples/scheduler/metairq_dispatch/sample.kernel.metairq_dispatch'
[  3%] Built target kobj_types_h_target
[... 105 lines cut ...]
DEBUG   - DEVICE: I: M19 T3 mirq 2 disp 3775 proc 0 real 0
DEBUG   - Timed out while monitoring serial output on nrf9160dk_nrf9160
DEBUG   - run status: timeout
INFO    - 1/1 nrf9160dk_nrf9160         samples/scheduler/metairq_dispatch/sample.kernel.metairq_dispatch FAILED Timeout (device 63.046s)
ERROR   - see: /home/maciej/zephyrproject2/zephyr/sanity-out/nrf9160dk_nrf9160/samples/scheduler/metairq_dispatch/sample.kernel.metairq_dispatch/handler.log

DEBUG   - running footprint_reports
INFO    - 0 of 1 tests passed (0.00%), 1 failed, 0 skipped with 0 warnings in 68.44 seconds
INFO    - In total 1 test cases were executed on 1 out of total 284 platforms (0.35%)
INFO    - 1 tests executed on platforms, 0 tests were only built.

Hardware distribution summary:

| Board             | ID   |   Counter |
|-------------------|------|-----------|
| nrf9160dk_nrf9160 |      |         1 |

```
</p>
</details>

**Environment (please complete the following information):**
 - OS:  Ubuntu 18.04
 - Toolchain Zephyr SDK 11.3
 - Commit SHA 4eba946e2b

vanti, 2020-09-11:
Same issue is seen on cc1352r1_launchxl as well.

PerMac, 2020-09-15:
I haven't seen this test failing for some time. Was it fixed somewhere else in the meantime? @vanti How does it look now on your platform?

carlescufi, 2020-09-15:
@ioannisg can you do a run of this test on multiple boards? I will do reel_board

carlescufi, 2020-09-15:
Passes on reel_board

vanti, 2020-09-15:
@PerMac I haven't been able to reproduce it with the latest code today.

PerMac, 2020-09-16:
I've run the test with the latest master and it passes on all platforms in my setup (nrf52dk_nrf52832, nrf52840dk_nrf52840, nrf9160dk_9160 and nrf5340dk_nrf5340_cpuapp)

nashif, 2020-09-16:
lets close this then, reopen if you see this again.

closed 2020-09-16 by nashif