zephyrproject-rtos/zephyr #28053: Eclipse broken build ability

tolezk, 2020-09-04:
Starting from 79b1f89 I get error on any build attempt if I try to use Eclipse.
Attempts to build using the command line are successful.

Eclipse's console build log doesn't contain any error:
```
----- Generating CMake files for board myboard in build
d:\tools\windows\msys64\mingw64\bin\cmake.exe -DBOARD=myboard -G Ninja -DCMAKE_EXPORT_COMPILE_COMMANDS=ON D:\projects\Zephyr\myproject
-- Application: D:/projects/Zephyr/myproject
-- Zephyr version: 2.3.99 (D:/projects/Zephyr/zephyr)
-- Found Python3: D:/tools/windows/Python/Python37/python.exe (found suitable exact version "3.7.7") found components: Interpreter 
-- Found west (found suitable version "0.7.3", minimum required is "0.7.1")
-- Board: myboard
-- Found dtc: D:/tools/windows/msys64/usr/bin/dtc.exe (found suitable version "1.6.0", minimum required is "1.4.6")
-- Found toolchain: gnuarmemb (D:/tools/windows/GNU Arm Embedded Toolchain/9.3.1 2020-q2)
-- Found BOARD.dts: D:/projects/Zephyr/myproject/module/boards/arm/myboard/myboard.dts
-- Generated zephyr.dts: D:/projects/Zephyr/myproject/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: D:/projects/Zephyr/myproject/build/zephyr/include/generated/devicetree_unfixed.h
Parsing D:/projects/Zephyr/zephyr/Kconfig
Loaded configuration 'D:/projects/Zephyr/myproject/module/boards/arm/myboard/myboard_defconfig'
Merged configuration 'D:/projects/Zephyr/myproject/prj.conf'
Configuration saved to 'D:/projects/Zephyr/myproject/build/zephyr/.config'
Kconfig header saved to 'D:/projects/Zephyr/myproject/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 9.3.1
-- The CXX compiler identification is GNU 9.3.1
-- The ASM compiler identification is GNU
-- Found assembler: D:/tools/windows/GNU Arm Embedded Toolchain/9.3.1 2020-q2/bin/arm-none-eabi-gcc.exe
-- Cache files will be written to: C:\Users\ai\AppData\Local/.cache/zephyr
-- Configuring done
-- Generating done
-- Build files have been written to: D:/projects/Zephyr/myproject/build
----- Done generating CMake files for board myproject in build
```

However, I get the error message window containing next info:
```
Errors occurred during the build.
Errors running builder 'Zephyr Application Builder' on project 'prj'.
java.lang.NullPointerException
```

mbolivar-nordic, 2020-09-04:
I don't know if I'm the right assignee since the error is a Java NPE.

But is this fixed by https://github.com/zephyrproject-rtos/zephyr/pull/27944?

dcpleung, 2020-09-05:
This should be fixed by https://github.com/zephyrproject-rtos/eclipse-plugin/pull/33 on the plugin side.

mbolivar-nordic, 2020-09-08:
@carlescufi @MaureenHelm I think this should be closed then as it doesn't seem to be a Zephyr build system bug.

closed 2020-09-08 by mbolivar-nordic

dsilva-vd, 2020-09-16:
@dcpleung  @tolezk  Is there a workaround until the fix is released? 
Thanks!

dcpleung, 2020-09-16:
@dsilva-vd Could you update the plugin and try? A new version of plugin has been pushed. It should show up in the Eclipse Marketplace shortly.

dsilva-vd, 2020-09-16:
@dcpleung The new version (Version: 0.2.99.202009161801) on Eclipse 2020-09 resolved my problem and a zephyr.elf was successfully linked! :)