zephyrproject-rtos/zephyr #28057: TCP2: client side receives EOF before all pending data is fed into it

pfalcon, 2020-09-04:
This is follow up to #27982, #28011. 

Steps top reproduce: `samples/net/sockets/big_http_download`, `qemu_x86`. With master 05972fff5af2ea05ea20af980d64447c4a07d185 (with #28011 merged), the output is:

~~~
Preparing HTTP GET request for http://archive.ubuntu.com:80/ubuntu/dists/xenial/main/installer-amd64/current/images/hd-media/vmlinuz
[00:00:00.100,000] <inf> net_config: Initializing network
[00:00:00.100,000] <inf> net_config: IPv4 address: 192.0.2.1
[00:00:00.100,000] <inf> net_config: Running dhcpv4 client...
addrinfo @0x142008: ai_family=1, ai_socktype=1, ai_protocol=6, sa_family=1, sin_port=5000

Iteration 1 of INF
sock = 0
Download progress: 7015450 Bytes; 6851 KiB; 6 MiB
Hash: 32367d329dfad28f4351357d7a119428ac82fd7d2da76e84ad44f347e3e96d32
HASH MISMATCH!
Total downloaded so far: 6 MiB

Iteration 2 of INF
sock = 0
Download progress: 7015450 Bytes; 6851 KiB; 6 MiB
Hash: 32367d329dfad28f4351357d7a119428ac82fd7d2da76e84ad44f347e3e96d32
HASH MISMATCH!
Total downloaded so far: 13 MiB

Iteration 3 of INF
sock = 0
Download progress: 7015450 Bytes; 6851 KiB; 6 MiB
Hash: 32367d329dfad28f4351357d7a119428ac82fd7d2da76e84ad44f347e3e96d32
HASH MISMATCH!
Total downloaded so far: 20 MiB
~~~

Mind the **HASH MISMATCH!** errors. Checking closer, the downloaded file size reported is also not right: http://archive.ubuntu.com:80/ubuntu/dists/xenial/main/installer-amd64/current/images/hd-media/vmlinuz has 7015896 bytes, i.e. 446 bytes missing. As can be seen, this picture is stable.

And trying to cut the `vmlinuz` file to size of 7015450 produced the SHA256 reported by the app above:

~~~
$ dd if=vmlinuz of=vmlinuz.1 count=7015450 bs=1
$ sha256sum vmlinuz.1 
32367d329dfad28f4351357d7a119428ac82fd7d2da76e84ad44f347e3e96d32  vmlinuz.1
~~~

So, it looks like FIN packet handling isn't exactly right. It looks like if FIN is received, EOF (read size == 0) is reported to an app immediately. But instead, any pending buffered data from peer should be fed first, and only the EOF communicated. (Just a wild guess, it may be a different issue.)

pfalcon, 2020-09-04:
The same behavior is on frdm_k64f too.

jukkar, 2020-09-04:
After quick investigation, the following patch might help here (did not test it yet). If you have time, please try it.

```
diff --git a/subsys/net/ip/tcp2.c b/subsys/net/ip/tcp2.c
index 5c0cbcd8a2..bdce31ed9f 100644
--- a/subsys/net/ip/tcp2.c
+++ b/subsys/net/ip/tcp2.c
@@ -1181,6 +1181,10 @@ next_state:
                        break;
                } else if (th && FL(&fl, ==, (FIN | ACK | PSH),
                                    th_seq(th) == conn->ack)) {
+                       if (len) {
+                               tcp_data_get(conn, pkt);
+                       }
+
                        conn_ack(conn, + len + 1);
                        tcp_out(conn, FIN | ACK);
                        next = TCP_LAST_ACK;
```

pfalcon, 2020-09-04:
@jukkar: Yes, that was it, thanks! Please post a patch.

PR #28063 (2020-09-04): net: tcp2: Make sure all incoming data is given to app at eof
When the connection is terminated, make sure that any pending
data is feed to the application.

Fixes #28057
commit 3e5a4d79 (2020-09-04): net: tcp2: Make sure all incoming data is given to app at eof

When the connection is terminated, make sure that any pending
data is feed to the application.

Fixes #28057