zephyrproject-rtos/zephyr #28061: nrf52840 can't boot up after power plug in,unless it was connected to JLINKRTTVIEWER with a JTAG

holmes81, 2020-09-04:
**Describe the bug**
It's strange,
It's seem like my program needs J-LINK RTT VIEWER to trigger a startup

My nrf52840 can't boot up with Zephyr OS after power plug in,unless it was connected to J-LINK RTT VIEWER by JTAG,
I try to run the example which was provided by NRFSDK,it's normal.So soc is OK,it should be a SW bug.

If I disconnected the J-LINK RTT VIEWER, the device will stop running,
Connect again and it will start running again.

If I connected with J-LINK RTT VIEWER but not disconnected it and just close the J-LINK RTT VIEWER application,
Soc will keep running.

**To Reproduce**
Steps to reproduce the behavior:
1.NRF52840QIAA SOC
2. Try to run a sample which was included by Zephyr 
3.Plug in the power supply,SOC will not run the sample
4. Connect SOC with J-LINK RTT VIEWER by JTAG,sample will be run by SOC,log will be printed by J-LINK RTT VIEWER
5. Disconnect and reconnect with J-LINK RTT VIEWER,sometimes sample wii be restarted.

**Expected behavior**
Plug in the power supply,SOC will run the sample

**Impact**
I can't run my application after plug in the power supply unless I connected it with J-LINK RTT VIEWER by a JTAG

**Logs and console output**
Zephyr OS will be reboot after I disconnect and reconnect with J-LINK RTT VIEWER 

[image]

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context about the problem here.

holmes81, 2020-09-04:
[image]

carlescufi, 2020-09-04:
@simtind does this ring a bell?

holmes81, 2020-09-07:
And If I disable logging over RTT,SOC will not run even I connected it to RTT Viewer

simtind, 2020-09-07:
Looks like you may be locking up due to some side effect of debug power not being properly disabled. This is a known issue in nrfjprog that we aim to fix in the next release. 
Internal reference: MDK-1973

holmes81, 2020-09-07:
@simtind 

HI,is there a workaround solution available for this?

carlescufi, 2020-09-08:
Closing this since this is not Zephyr-related but rather an issue with the Nordic nrfjprog tool suite.

@simtind thanks for your input. Is there a workaround that @holmes81 can use in the meantime?

closed 2020-09-08 by carlescufi

holmes81, 2020-09-09:
Thanks all so much,
the reason is CONFIG_BOARD_ENABLE_DCDC  default enabled,after I disabled it,the issue was fixed

real-tintin, 2024-09-07:
> Thanks all so much, the reason is CONFIG_BOARD_ENABLE_DCDC default enabled,after I disabled it,the issue was fixed

Life saver.