zephyrproject-rtos/zephyr #28068: Crash in USB device when turning HFXO off

pdunaj, 2020-09-04:
**Describe the bug**

The crash was replicated during nrf_desktop usual tests after the last upmerge.

```
[00248874] <inf> event_manager: e: hid_report_subscription_event report 0x1 notification disabled by 0x20001b68
--
[00248882] <inf> pmw3360: Set sample time to 100 ms
[00253403] <inf> usb_hid: Device suspended
[00253404] <wrn> usb_state: USB suspend
[00253405] <inf> usb_state: USB HID 0x20002440 disabledASSERTION FAIL [err >= 0] @ WEST_TOPDIR/zephyr/drivers/usb/device/usb_dc_nrfx.c:771
[00253410] <err> os: r0/a1:  0x00000004  r1/a2:  0x00000303  r2/a3:  0x00000000
[00253410] <err> os: r3/a4:  0x0000000b r12/ip:  0x00000000 r14/lr:  0x00014bb9
[00253410] <err> os:  xpsr:  0x61000000
[00253410] <err> os: Faulting instruction address (r15/pc): 0x0000fb5c
[00253410] <err> os: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
[00253411] <err> os: Current thread: 0x200019e8 (unknown)
[00635097] <err> os: Halting system
```

**To Reproduce**
Steps to reproduce the behavior:
Replicated by automated tests.
```
DUT bonded with Tester (pc-ble-driver)
Tester enables HVN in DUT
DUT USB plugged in
Tester disables HVN in DUT
DUT USB plugged out
```
It seems it should be possible to crash the mouse by connecting/disconnecting the USB while BLE is active.

**Expected behavior**
Crash should not happen

**Impact**
Showstopper

**Logs and console output**
N/A

**Environment (please complete the following information):**
master

**Additional context**
N/A

pdunaj, 2020-09-04:
@nordic-krch @emob-nordic

emob-nordic, 2020-09-07:
@nordic-krch Isn't this caused by dedicated path for enabling hfclk for bluetooth?

nordic-krch, 2020-09-07:
i suspect that it is caused by custom clock_control implementation in nrf connect sdk. I'm looking a that now.

nordic-krch, 2020-09-09:
@pdunaj can you close if https://github.com/nrfconnect/sdk-nrf/pull/2870 helped? It seems like not zephyr downstream issue.

closed 2020-09-09 by pdunaj