zephyrproject-rtos/zephyr #28090: bluetooth: build error with extended advertising

pabigot, 2020-09-06:
When extended advertising is enabled without `CONFIG_BT_CTLR_PRIVACY=y` the following build error is produced:

```
/mnt/devel/zp/zephyr/subsys/bluetooth/controller/ll_sw/ull_adv_aux.c: In function 'ull_adv_aux_hdr_set_clear':
/mnt/devel/zp/zephyr/subsys/bluetooth/controller/ll_sw/ull_adv_aux.c:502:25: error: 'struct ll_adv_set' has no member named 'own_addr_type'
  502 |   sec_pdu->tx_addr = adv->own_addr_type & 0x1;
```

Unless using extended advertising features **requires** `BT_CTLR_PRIVACY` by specification this should be fixed.  (I really hope it isn't required, because my use case is sensor beacons that are well-served with the default id from FICR, and very poorly served by ids that change.)

cvinayak, 2020-09-07:
@andrzej-kaczmarek Please address this. And, if there is no address in the previous primary PDU, then with/without privacy the device address has to be supplied by the caller of the `ull_adv_aux_hdr_set_clear` function in the `value` length encoded parameter.

galak, 2020-09-08:
See this on nightly build:

```
INFO    -  1/12 nrf51dk_nrf51422          tests/bluetooth/init/bluetooth.init.test_controller_dbg_ll_sw_split FAILED Build failure (build)
 ```

closed 2020-09-10 by carlescufi