zephyrproject-rtos/zephyr #28092: Make SPI speed of SDHC card configurable

lowlander, 2020-09-06:
The SD-card-spi driver has a hardcoded 400kHz and 4Mhz SPI bus speed, it would be good to make this configurable (maybe via DT?)

EminYagmahan, 2020-09-07:
This PR #26319 contains already a working fix, but is still blocked due to some conflicts.

lowlander, 2020-09-07:
@EminYagmahan  thanks for pointing that PR out, I kind of overlooked that, if that ends up in master this issue can be closed.

lowlander, 2020-09-08:
Closing this because things seem to be fixed in the mentioned PR and with the 2.4.rc1 stage it might take a while for that to be merged, no reason to keep open this bug report for that long (there are already enough others).

closed 2020-09-08 by lowlander

thelittlebug, 2021-01-16:
ok, and now?
the PR is closed and this ticket is closed. nothing changed.

PR #33684 (2021-03-26): disk: sdhc: Switch to clock frequency from DTS
The "spi-max-frequency" property already exists, but was unused. This
now sets the SPI clock frequency to this value (limited to 24MHz) once
initialisation is complete.

Due to the nature of the SPI API, it is necessary to have two separate
configuration structures to switch clock speed as some SPI drivers only
compare pointers to detect changes.

Fixes: #32996, Fixes: #28092, #22906

Improves issue reported in #26513, but does not implement CMD6 to
switch to High Speed mode.