zephyrproject-rtos/zephyr #28093: SDHC card driver can't reprogram SPI bus speed

lowlander, 2020-09-06:
The SDHC card driver uses two SPI bus speeds, 400kHz to start and later switches to 4MHz. Or at least it tries to switch to 4MHz. To do that it updates the `data->cfg.frequency` and `&data->cfg` is than later passed to the SPI transfer functions. 

That doesn't work, at least not on the STM32SPI driver, because it only re-configures the SPI bus if the `cfg` pointers are different.

This is extremely counter intuitive, and also isn't easy to work around because every user of the SPI driver will need to have at least two cfg structs and one has to keep track of which one was used last (else there still is no change of pointers). 

This is apperently so error prone that even the documentation tries to warn for it;
> config: Pointer to a valid spi_config structure instance. **Pointer-comparison may be used to detect changes from previous operations.**
 
I know the STM32SPI driver has this "problem" and a quick scan trough the other drivers seems they all use the same scheme to compare pointers, and not content.

Possible solution: Change the `config` member of the `spi_context` form `const struct spi_config*` to `struct spi_config` and copy and compare the actual content and not just the pointer.

lowlander, 2020-09-07:
@henrikbrixandersen can you confirm (since you did a lot of work on the sdcard driver) that the sdhc spi driver isn't able to upgrade the spi bus speed? And that that might cause the slow read/write speeds like in issue #22906

henrikbrixandersen, 2020-09-07:
> @henrikbrixandersen can you confirm (since you did a lot of work on the sdcard driver) that the sdhc spi driver isn't able to upgrade the spi bus speed? And that that might cause the slow read/write speeds like in issue #22906

The current way of attempting to change the SCL frequency will not work. See my comment on https://github.com/zephyrproject-rtos/zephyr/pull/26319#issuecomment-688465092.

lowlander, 2020-09-08:
> The current way of attempting to change the SCL frequency will not work. See my comment on [#26319 (comment)](https://github.com/zephyrproject-rtos/zephyr/pull/26319#issuecomment-688465092).

I missed that PR (filtered on the wrong labels I guess), I updated this bug report to at least updating the documentation.

henrikbrixandersen, 2020-09-09:
@lowlander I have read through the SPI API documentation again. From my point of view the documentation is clear on what can be changed in the struct between consecutive calls: https://docs.zephyrproject.org/latest/reference/peripherals/spi.html#c.spi_config - but I might be biases, since I have used the SPI API extensively.

The reasoning behind this (the pointer compare, the fields you are allowed to change between invocations) is that the `struct spi_config` is supposed to be used within a SPI peripheral driver. A SPI peripheral normally runs at one fixed bitrate and has one fixed CS line. The `SPI_HOLD_ON_CS` and `SPI_LOCK_ON` flags can safely be changed from the driver between invocations as they should not affect the SPI controller configuration.

The SPI bus can have multiple SPI peripherals, each with their own driver (or at least their own driver instance in case of multiple peripherals on the same SPI bus using the same driver) and each of these will call `spi_transceive()` with their own, private `struct spi_config`. Thus, a simple pointer compare should be sufficient for establishing whether the SPI controller needs to be reconfigured for the given transaction.

The SPI SDHC disk access driver is the exception here. It uses two different SPI controller configurations, each with their own bitrate. This can be solved by using two different `struct spi_config` instances as suggested in #26319.

Could you please suggest a change to the API documentation to address the issue, you are seeing?

lowlander, 2020-09-09:
@henrikbrixandersen I might also be a bit biased because I strongly dislike the pointer compares :-) 

> Note
Only cs_hold and lock_on can be changed between consecutive transceive call. Rest of the attributes are not meant to be tweaked.

> Warning
Most drivers use pointer comparison to determine whether a passed configuration is different from one used in a previous transaction. **Changes to fields in the structure may not be detected.**

`cs_hold` and `lock_on` are fields in the structure. 

```
struct spi_config cfg;

...

cfg.cs_hold = 1;
spi_transceiver(&cfg .....);

cfg.cs_hold = 0;
spi_transceiver(&cfg .....);

```

Is the code above suppose to work or not ? If the pointer compare warning is to be believed it should not work ...

```
struct spi_config cfg_hold;
struct spi_config cfg_no_hold;

...

cfg_hold.cs_hold = 1;
spi_transceiver(&cfg .....);

cfg_no_hold.cs_hold = 0;
spi_transceiver(&cfg .....);

```

... and it should be like that.

If I understand you correctly the first piece of code is OK. Does the second piece of code even work? Or does the second piece of code reset the CS line? I believe the SPI API is extremely error prone and needs more clarification (or better a redesign).

How about something like this (please don't copy grammar and spelling mistakes :-) )

> Warning
Only cs_hold and lock_on can be changed between consecutive transceive calls. Other members of struct spi_config may not be changed. To change for example the speed one needs to create two spi_config variables and switch between them as follows;  `` code example ``

5frank, 2021-12-07:
I also run into this issue. IMO the pointer compare is acceptable even though not very "intuitive".  If this will remain, I think adding a `spi_configure()` method or a `SPI_FORCE_RECONFIG` flag  would be a good idea. This would make temporary spi configurations easier to write and less obscure to read.