zephyrproject-rtos/zephyr #28095: Doc: Getting Started Guide: reel board blinky gif is outdated

DSc-de, 2020-09-06:
**Describe the bug**
If setting up Zephyr according to the Getting Started Guide including build and flash of blinky sample on reel_board / real_board_v2, one can see blinking of red LED D3 (GPIO P0.11) at the front side. However, the Getting Starting Guide shows blinking of another led: LED D13 ( P0.13) at the back side.

**To Reproduce**
Steps to reproduce the behavior:
1. build and flash blinky sample for reel_board or reel_board_v2
2. compare blinking led on the board with blinking led in the Getting Started Guide gif
3. observe difference: board blinking on front side, gif blinging on back side

**Expected behavior**
Gif shall show blinking of front LED D3 (GPIO P0.11).

**Impact**
Could confuse new users setting up Zephyr on reel_board / reel_board_v2 according to Getting Started Guide.

**Environment:**
 - OS: Windows
 - Toolchain: GNU ARM embedded
 - Version used: 2.4.0-rc1

**Additional context**
Root cause: blinky controls led0. led0 has been reallocated on the reel_board from the back side led to the front side led. For example, version of 18 Sep 2019 still has "old" mapping which matches the Getting Started Guide blinking.

jfischer-phytec-iot, 2020-09-08:
> Root cause: blinky controls led0. led0 has been reallocated on the reel_board from the back side led to the front side led.

That is not quite true, the LED on the back was removed and GPIO reused for the buzzer. Actually documentation issue not a bug.

carlescufi, 2020-09-08:
@DSc-de or @jfischer-phytec-iot will you update the .gif please?

DSc-de, 2020-09-08:
Below please find the updated gif. If acceptable, please commit to /doc/getting_started/img . Please let me know if I shall do a pull request instead (never done this before but the great zephyr documentation will tell me how to).

[image]

PR #28340 (2020-09-14): doc: getting_started: updated reel board blinky gif
Fixes: #28095
commit d04c328d (2020-09-14): doc: getting_started: updated reel board blinky gif

Fixes: #28095
closed 2020-09-14 by carlescufi