zephyrproject-rtos/zephyr #28105: SMP: next thread determination is not atomic with respect to context switch

andrewboie, 2020-09-06:
**Describe the bug**
For some time now we have seen weird behavior on x86_64 where a thread object that was not suspended was being resumed, or the same thread was being scheduled on two CPUs at the same time. A typical backtrace:

```
START - test_fs_read
E: Page fault at address 0xb9 (error code 0x10)
E: Linear address not present in page tables
E: Access violation: supervisor thread not allowed to execute
E: PTE: not present
E: RAX: 0x0000000000104522 RBX: 0x0000000000125780 RCX: 0x000000000000000c RDX: 0x0000000000000138
E: RSI: 0x000000000014e000 RDI: 0x0000000000138000 RBP: 0x000000000013a3f8 RSP: 0x0000000000140e80
E:  R8: 0x000000000014b578  R9: 0x0000000000000000 R10: 0x0000000000000000 R11: 0x0000000000000000
E: R12: 0x0000000000000000 R13: 0x000000000014b578 R14: 0x0000000000104522 R15: 0x0000000000000000
E: RSP: 0x0000000000140e80 RFLAGS: 0x0000000000000246 CS: 0x0018 CR3: 0x0000000000150000
E: RIP: 0x00000000000000b9
E: Attempt to resume un-suspended thread object
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 1
E: Current thread: 0x0000000000126220 (main)
E: Halting system
```

AFAICT (I'd say I'm 95% sure at this point) this is a race condition between the determination of the next thread to run in `next_up()` and performing the actual context switch. This has to be atomic with respect to the `sched_spinlock`, as this lock's scope includes any changes in thread state.

Currently, in both x86_64's interrupt exit handling path, and the core kernel implementation of `do_swap()`, some API (either `z_get_next_switch_handle()` or `z_get_next_ready_thread()`) is called to determine the next thread to schedule, both end up invoking `next_up()`. This is then used to perform the context switch in `arch_switch()`. The `sched_spinlock` is not held in between these two events, only local CPU interrupts are disabled. The thread state could change in the meantime due to activity on some other CPU.

I am not yet completely sure the best way to adjust the interrupt exit path on x86_64. But for the C code in `do_swap()` I had an idea we could replace this bit:

```
	if (is_spinlock) {
		k_spin_release(lock);
	}
```

with an operation to atomically re-queue whatever lock was provided, to the sched_spinlock. If the sched_spinlock was already provided, do nothing and keep it locked. If local interrupts were just locked (z_swap_irqlock() for example), just grab the sched_spinlock. Whatever the case, hold sched_spinlock through the determination of the next ready thread and the subsequent context switch call to the end of do_swap().

I expect this is also an issue on ARC HS, our other platform which supports SMP.

andrewboie, 2020-09-08:
Next step is to confirm the bug by identifying some specific thread state change scenarios across CPUs that will reproduce some of these "attempt to resume unsuspended thread object" faults. Need to make extra sure that the scheduler framework doesn't have some other way of compensating for an obsolete scheduling decision being executed, just in case I missed something.

andrewboie, 2020-09-21:
Found another potential cause of these crashes, in how every IPC object wakes up threads.

Example from `k_sem`:
```
void z_impl_k_sem_give(struct k_sem *sem)
{
        k_spinlock_key_t key = k_spin_lock(&lock);
        struct k_thread *thread;

        sys_trace_semaphore_give(sem);
        thread = z_unpend_first_thread(&sem->wait_q);

        if (thread != NULL) {
                arch_thread_return_value_set(thread, 0);
                z_ready_thread(thread);
        } else {
                sem->count += (sem->count != sem->limit) ? 1U : 0U;
                handle_poll_events(sem);
        }

        z_reschedule(&lock, key);
        sys_trace_end_call(SYS_TRACE_ID_SEMA_GIVE);
}
```

The wait queue is protected by `sched_spinlock`, and the individual calls to `z_unpend_first_thread()` and `z_ready_thread()` hold it. But if the thread state changes (like a `k_thread_abort()` on another CPU) we could be readying a dead thread.

All the IPC objects have this problem, AFAICT. This is only a problem on SMP because in uniprocessor spinlocks just disable interrupts, and in all cases we are holding a different spinlock while all of this is going on.

The solution I am trying out adds a new private scheduling API for this use-case:

```
/**
 * Wake up a thread pending on the provided wait queue
 *
 * Given a wait_q, wake up the highest priority thread on the queue. If the
 * queue was empty just return false.
 *
 * Otherwise, do the following, in order,  holding sched_spinlock the entire
 * time so that the thread state is guaranteed not to change:
 * - if the callback function is provided, run the callback function with
 *   the provided object and data pointers
 * - Set the thread's return value to swap_retval
 * - un-pend and ready the thread
 *
 * @param wait_q Wait queue to wake up the highest prio thread
 * @param swap_retval Swap return value for woken thread
 * @param cb Callback function, or NULL
 * @param obj Kernel object associated with this wake operation, passed to cb
 * @param data data value passed to the callback function
 * @retval true If a thread waa woken up
 * @retval false If the wait_q was empty
 */
bool z_wake_one(_wait_q_t *wait_q, unsigned int swap_retval,
		z_thread_wake_cb_t cb, void *obj, void *data);
```

This API is basically a complement to `z_pend_curr()` and if some operation needs to be performed on the thread that was woken up, that can be implemented in the callback function which runs with `sched_spinlock` held.

From a higher-level perspective, we can close these races inside the kernel with the right locking, but I am starting to think we really do need a process ID abstraction for operations on threads from applications; i.e. `k_tid_t` is a handle and not the thread object pointer.

andrewboie, 2020-10-27:
I spoke offline with @andyross and we think the context switch in the bug description may not be a problem, but we agree that https://github.com/zephyrproject-rtos/zephyr/issues/28105#issuecomment-696317757 is very likely a bug. I am looking deeper at it.

andrewboie, 2020-10-30:
I have a patch now to close the potential race when un-pending threads from IPC wait queues.
Unfortunately, these crashes are still reproducible with that applied, samples/userspace/shared_mem still eventually always crashes for me if you let it run long enough. More digging needed.

MrVan, 2020-11-17:
@andrewboie  any progress on this issue?

andrewboie, 2020-11-17:
> @andrewboie any progress on this issue?

No. I am testing https://github.com/zephyrproject-rtos/zephyr/pull/29668 but I already know it does not completely solve this issue.

@MrVan patches are welcome if you would like to help debug this effort.

pabigot, 2021-01-21:
Note reproducer available in #30360 if you add `CONFIG_TEST=y` so that code.

andyross, 2021-02-04:
I'm guessing I'm not going to get a fix in for this before the release, so adding some quick notes just so I don't forget anything once I get dragged away to other things.  My current test rig is below, it's a trimmed down and minified version of Peter's code, that eventually swapped in a semaphore for the original queue.

The core problem is with repeated preemption of spinning threads.  The circumstances that are needed are:

1. A high priority thread on CPU0 blocks, leaving the CPU idle.
2. On CPU1, a lower priority thread makes the high priority thread runnable but continues running itself
3. The high priority thread then preempts CPU1 and CPU0 would be expected to pick up the low priority thread
4. But instead what happens is they both get the same thread somehow

There's not much code in there, and I'm actually working right now on minimizing it farther (via directly hooking the scheduler inside the test and eliminating semaphore too).  But the locking all looks correct as far as I see it.

But the priority switch is definitely the key: you have to preempt a spinning thread in a context where another CPU is idle and available to run something.  Swap the priorities and it doesn't happen.  I can also comment out the z_reschedule() to prevent preemption and suppress it too.

```c
#include <zephyr.h>

K_SEM_DEFINE(sem, 0, 99);

volatile int pending;

static void thread_aux(void *p1, void *p2, void *p3)
{
	while (true) {
		(void)k_sem_take(&sem, K_FOREVER);
		pending = false;
	}
}

/* Note priority: this thread must be higher than main() to exercise
 * the bug, otherwise the threads pin themselves to specific cpus and
 * everything works. It has to be able to preempt the current thread
 * and cause the two threads to swap processors.
 */
K_THREAD_DEFINE(auxthread, 4096,
		thread_aux, NULL, NULL, NULL, -1, 0, 0);

void main(void)
{
	while (1) {
		pending = true;
		k_sem_give(&sem);
		while (pending) {
		}
	}
}
```

pabigot, 2021-02-04:
Thanks, this is interesting.  Two questions:

First, can you reproduce this on real hardware, or just qemu?  When I ran my reproducer on upsquared there were no failures.

Second,
> 2. On CPU1, a lower priority thread makes the high priority thread runnable but continues running itself
> 3. The high priority thread then preempts CPU1 and CPU0 would be expected to pick up the low priority thread

That seems like a sub-optimal solution.  Is it feasible to detect that CPU0 is idle and make it pick-up the high priority thread, instead of swapping CPUs and all related context?

andyross, 2021-02-04:
I tried it briefly on my up2 and didn't see the problem, but I believe for sure this is real and not an emulator bug.  If I had to guess the effective latency for an interprocessor interrupt on qemu is much higher (it's a unix signal delivered by the host OS to a separate thread vs. a few clock cycles of transit through an interrupt controller) so gives more window for things to go wrong (which in this case means the local CPU gets preempted -- that probably happens faster on hardware and the other CPU picks up the thread before the local one can make its way to z_swap()).

As far as performance: that's going to be application dependent really.   Always running the new thread on the idle CPU gets it to run faster (except maybe on qemu, see above) and makes for better parallelism, but if it involves work with cache-resident data on the local CPU (it probably does if you just woke it up) it might actually end up doing the high priority work with lower latency even though the low priority thread sees better throughput.  Some apps are going to choose one, some the other.

In general apps with really severe requirements like that should be managing their own CPU assignments and workloads.  We have APIs for that too.  But the default seems not completely wrong.

PR #32089 (2021-02-14): Fix race and deadlock in SMP context switch paths
Two bugs found by one exerciser in #28105 that repeatedly hammered the thread preemption paths.

The first was simple enough, some needless defensive coding (not wanting to change a preexisting switch handle) could be seen by code spinning in wait_for_switch() as a completed context switch and cause the other to start running the current thread before it had spilled its context.

The second was an even rarer (and more interesting) deadlock exposed when that was fixed, in a similar area of code.  Crazy timing (possible seemingly only under qemu) [... cut]
commit dd432215 (2021-02-14): kernel/sched: Fix race with switch handle

The "null out the switch handle and put it back" code in the swap
implementation is a holdover from some defensive coding (not wanting
to break the case where we picked our current thread), but it hides a
subtle SMP race: when that field goes NULL, another CPU that may have
selected that thread (which is to say, our current thread) as its next
to run will be spinning on that to detect when the field goes
non-NULL.  So it will get the signal to move on when we revert the
value, when clearly we are still running on the stack!

In practice this was found [... cut]
closed 2021-02-14 by nashif