zephyrproject-rtos/zephyr #28108: EEPROM shell MPU Fault when performing a write command with more than 9 bytes

nandojve, 2020-09-07:
**Describe the bug**
Using EEPROM console to write more than 9 bytes on an AT24 gives MPU fault. The read is performing up to 256 (max allowed). When create a mock for `.transfer` with a `return 0` was noted the same behavior. This indicate that problem can be outside drivers.

**To Reproduce**
1. build hello world app
2. make menuconfig and enable an I2C + EEPROM AT24 + Console with 256 bytes buffer + Log + UART backend
3. execute: eeprom write <device> 0 8 7 6 5 4 3 2 1 0 1 2 3
4. See error

**Expected behavior**
EEPROM Shell working up to 256 bytes for read and write

**Impact**
EEPROM Shell doesn't execute a write command with more than 9 bytes.

**Logs and console output**

```
uart:~$ eeprom write EEPROM_CAPE 10 8 7 6 5 4 3 2 1 0 1 2 3
Writing 9 bytes to EEPROM...
Verif[00:00:09.516,000] <err> os: ***** MPU FAULT *****
[00:00:09.516,000] <err> os:   Data Access Violation
[00:00:09.516,000] <err> os:   MMFAR Address: 0xceab
[00:00:09.516,000] <err> os: r0/a1:  0x00000000  r1/a2:  0x20000370  r2/a3:  0x00000002
[00:00:09.516,000] <err> os: r3/a4:  0x0000cea7 r12/ip:  0x000016ee r14/lr:  0x00007f99
[00:00:09.516,000] <err> os:  xpsr:  0x21000000
[00:00:09.516,000] <err> os: r4/v1:  0x00000000  r5/v2:  0x0000cc05  r6/v3:  0x0000f49b
[00:00:09.516,000] <err> os: r7/v4:  0x0000f597  r8/v5:  0x0000e8ed  r9/v6:  0x00000000
[00:00:09.516,000] <err> os: r10/v7: 0x00000000  r11/v8: 0x00000000    psp:  0x20003970
[00:00:09.517,000] <err> os: Faulting instruction address (r15/pc): 0x00003480
[00:00:09.517,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:09.517,000] <err> os: Current thread: 0x20000370 (shell_uart)
[00:00:09.604,000] <err> os: Halting system
```

Looking at LST file the that fault happen at `shell->ctx->state = state;` command:
```
	if ((flag_last_nl_get(shell) == 0U) ||
    3440:	f412 4f7f 	tst.w	r2, #65280	; 0xff00
    3444:	d005      	beq.n	3452 <shell_process+0xf6>
	return shell->ctx->internal.flags.last_nl;
    3446:	f8d0 2638 	ldr.w	r2, [r0, #1592]	; 0x638
    344a:	f3c2 2207 	ubfx	r2, r2, #8, #8
    344e:	4293      	cmp	r3, r2
    3450:	d1e0      	bne.n	3414 <shell_process+0xb8>
	shell->ctx->internal.flags.last_nl = val;
    3452:	f8d0 2638 	ldr.w	r2, [r0, #1592]	; 0x638
    3456:	f363 220f 	bfi	r2, r3, #8, #8
    345a:	f8c0 2638 	str.w	r2, [r0, #1592]	; 0x638
				if (!shell->ctx->cmd_buff_len) {
    345e:	8e42      	ldrh	r2, [r0, #50]	; 0x32
    3460:	b99a      	cbnz	r2, 348a <shell_process+0x12e>
	shell->ctx->internal.flags.history_exit = val ? 1 : 0;
    3462:	f8d0 3638 	ldr.w	r3, [r0, #1592]	; 0x638
    3466:	f362 1386 	bfi	r3, r2, #6, #1
    346a:	f8c0 3638 	str.w	r3, [r0, #1592]	; 0x638
	shell_history_mode_exit(shell->history);
    346e:	68e0      	ldr	r0, [r4, #12]
    3470:	f009 fb9b 	bl	cbaa <shell_history_mode_exit>
	shell_raw_fprintf(shell->fprintf_ctx, "\n");
    3474:	497b      	ldr	r1, [pc, #492]	; (3664 <shell_process+0x308>)
    3476:	6960      	ldr	r0, [r4, #20]
    3478:	f009 f875 	bl	c566 <shell_raw_fprintf>
	shell->ctx->state = state;
    347c:	68a3      	ldr	r3, [r4, #8]
    347e:	2202      	movs	r2, #2
    3480:	711a      	strb	r2, [r3, #4]
```

**Environment (please complete the following information):**
 - OS: Linux
 - Zephyr SDK 0.11.3
 - 2.4.0-rc1

**Additional context**
List file for MPU Enabled: [zephyr.lst](https://github.com/zephyrproject-rtos/zephyr/files/5180615/zephyr.txt)

__

jakub-uC, 2020-09-08:
I doubt that pointed instruction is a root cause of the issue. It is called after each and every command execution.
In my opinion, the problem can be in the `static int cmd_write(const struct shell *shell, size_t argc, char **argv)` function inside drivers/eeprom/eeprom_shell.c.

Would you mind adding some `k_msleep(10);`  in the command to let the shell print everything?
After:
`shell_print(shell, "Writing %d bytes to EEPROM...", len); `
`shell_print(shell, "Verifying...");` and one after 
`err = eeprom_read(eeprom, offset, rd_buf, len); ` 
and one before
`shell_print(shell, "Verify OK"); ` ?

henrikbrixandersen, 2020-09-08:
Which board is this?

nandojve, 2020-09-08:
Hi @jakub-uC , I agreed with you!, I'll check your suggestion at my night.

@henrikbrixandersen It is a personal board that uses SAM4L + AT24C256, as reference you can check SAM4L-EK. There is no currently I2C driver for this SoC on main. Because of that, my first thing was mock the I2C driver API and check on HW. For my surprise, I could reproduce the exactly same behavior.

nandojve, 2020-09-08:
Hi @jakub-uC 
Changed the code as
```
	shell_print(shell, "Writing %d bytes to EEPROM...", len);
	k_msleep(10);
	err = eeprom_write(eeprom, offset, wr_buf, len);
	if (err) {
		shell_error(shell, "EEPROM write failed (err %d)", err);
		return err;
	}

	shell_print(shell, "Reading...");
	k_msleep(10);
	err = eeprom_read(eeprom, offset, rd_buf, len);
	if (err) {
		shell_error(shell, "EEPROM read failed (err %d)", err);
		return err;
	}
	shell_print(shell, "Verifying (memcmp)...");
	k_msleep(10);
	if (memcmp(wr_buf, rd_buf, len) != 0) {
		shell_error(shell, "Verify failed");
		return -EIO;
	}

	shell_print(shell, "Verify OK");
	k_msleep(10);
```
This is the result
```
[00:00:00.000,000] <inf> i2c_sam_twim: Device TWIM_1 initialized
*** Booting Zephyr OS build v2.4.0-rc1-12-gea5f15c72563  ***
Hello World! sam4l_wm400_cape

uart:~$ eeprom write EEPROM_CAPE 0 8 7 6 5 4 3 2 1 0 1 2 3
Writing 9 bytes to EEPROM...
Reading...
Verifying (memcmp)...
Verify OK
[00:02:32.535,000] <err> os: ***** MPU FAULT *****
[00:02:32.535,000] <err> os:   Data Access Violation
[00:02:32.535,000] <err> os:   MMFAR Address: 0x8d61
[00:02:32.535,000] <err> os: r0/a1:  0x00000000  r1/a2:  0x0000000e  r2/a3:  0x00000002
[00:02:32.535,000] <err> os: r3/a4:  0x00008d5d r12/ip:  0x00000000 r14/lr:  0x00002985
[00:02:32.535,000] <err> os:  xpsr:  0x01000000
[00:02:32.535,000] <err> os: Faulting instruction address (r15/pc): 0x00002bd4
[00:02:32.535,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:02:32.535,000] <err> os: Current thread: 0x200002d0 (shell_uart)
[00:02:32.597,000] <err> os: Halting system
[00:00:00.000,000] <inf> i2c_sam_twim: Device TWIM_1 initialized
*** Booting Zephyr OS build v2.4.0-rc1-12-gea5f15c72563  ***
Hello World! sam4l_wm400_cape

uart:~$ eeprom read EEPROM_CAPE 0 256
Reading 256 bytes from EEPROM, offset 0...
00000000: 08 07 06 05 04 03 02 01  00 ff ff ff ff ff ff ff |........ ........|
00000010: ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff |........ ........|
00000020: ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff |........ ........|
00000030: ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff |........ ........|
00000040: ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff |........ ........|
00000050: ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff |........ ........|
00000060: ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff |........ ........|
00000070: ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff |........ ........|
00000080: ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff |........ ........|
00000090: ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff |........ ........|
000000A0: ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff |........ ........|
000000B0: ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff |........ ........|
000000C0: ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff |........ ........|
000000D0: ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff |........ ........|
000000E0: ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff |........ ........|
000000F0: ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff |........ ........|
uart:~$ 
```
List file, same point
```
	shell->ctx->state = state;
    2bd0:	68a3      	ldr	r3, [r4, #8]
    2bd2:	2202      	movs	r2, #2
    2bd4:	711a      	strb	r2, [r3, #4]
```
Mock the SPI transfer method will fail at memcmp for obvious reasons but we should not see MPU Fault
```
uart:~$ [00:00:00.000,000] <inf> i2c_sam_twim: Device TWIM_1 initialized
*** Booting Zephyr OS build v2.4.0-rc1-12-gea5f15c72563  ***
Hello World! sam4l_wm400_cape

uart:~$ eeprom write EEPROM_CAPE 0 8 7 6 5 4 3 2 1 0 1 2 3
Writing 9 bytes to EEPROM...
Reading...
Verifying (memcmp)...
Verify fail
[00:00:13.103,000] <err> os: ***** MPU FAULT *****
[00:00:13.103,000] <err> os:   Data Access Violation
[00:00:13.103,000] <err> os:   MMFAR Address: 0x8c65
[00:00:13.103,000] <err> os: r0/a1:  0xfffffffb  r1/a2:  0x0000000e  r2/a3:  0x00000002
[00:00:13.103,000] <err> os: r3/a4:  0x00008c61 r12/ip:  0x00000000 r14/lr:  0x00002985
[00:00:13.103,000] <err> os:  xpsr:  0x01000000
[00:00:13.103,000] <err> os: Faulting instruction address (r15/pc): 0x00002bd4
[00:00:13.103,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:13.103,000] <err> os: Current thread: 0x200002d0 (shell_uart)
[00:00:13.165,000] <err> os: Halting system
```

I try test an old version and noted that is not ok. On that version, I got errors on subsys/shell/shell_fprintf.c at out_func. The error happen at `sh_fprintf->ctrl_blk->buffer_cnt == sh_fprintf->buffer_size`.

henrikbrixandersen, 2020-09-09:
What is your `CONFIG_EEPROM_SHELL_BUFFER_SIZE` set to? Is your main stack large enough to hold the buffer? Can you try lowering the EEPROM shell buffer size or increasing the stack size?

jakub-uC, 2020-09-09:
@nandojve I observed in the past similar problems with the shell when the thread stack was too low. I agree with @henrikbrixandersen proposals.

nandojve, 2020-09-09:
Hi @henrikbrixandersen , I already showed a output from a 256 bytes read. I'm not sure if this is an heap/stack problem.
Yesterday I spend lot of time and noted that can be failing at shell_fprintf.c at out_func function when one of those variables are checked `sh_fprintf->ctrl_blk->buffer_cnt == sh_fprintf->buffer_size`.  How is possible this line of code give us MPU Fault?

I remember that I have a similar problem with STM32 and a at25 like EEPROM. I'll try to double check with that too.
I'm start to be convinced that it rely can be something with shell.

nandojve, 2020-09-09:
Hi @henrikbrixandersen , same thing with STM32F767ZI and a ST M95640 (SPI AT25 compatible)

Board: nucleo_f767zi
shield: steval-fki868v2

```
uart:~$ eeprom read M95640 1024 256
Reading 256 bytes from EEPROM, offset 1024...
00000000: 01 02 03 04 ff ff ff ff  ff ff ff ff ff ff ff ff |........ ........|
00000010: ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff |........ ........|
00000020: ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff |........ ........|
00000030: ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff |........ ........|
00000040: ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff |........ ........|
00000050: ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff |........ ........|
00000060: ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff |........ ........|
00000070: ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff |........ ........|
00000080: ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff |........ ........|
00000090: ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff |........ ........|
000000A0: ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff |........ ........|
000000B0: ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff |........ ........|
000000C0: ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff |........ ........|
000000D0: ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff |........ ........|
000000E0: ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff |........ ........|
000000F0: ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff |........ ........|
uart:~$ eeprom write M95640 1536 8 7 6 5 4 3 2 1 0 1 2 3 4 5 6 7 8
Writing 9 bytes to EEPROM...
Verifying...
[00:00:38.564,000] <err> os: ***** BUS FAULT *****
[00:00:38.564,000] <err> os:   Imprecise data bus error
[00:00:38.564,000] <err> os: r0/a1:  0x00000000  r1/a2:  0x0000000e  r2/a3:  0x00000002
[00:00:38.564,000] <err> os: r3/a4:  0x0edc66ac r12/ip:  0x00000000 r14/lr:  0x080029a5
[00:00:38.564,000] <err> os:  xpsr:  0x41000000
[00:00:38.564,000] <err> os: Faulting instruction address (r15/pc): 0x08002492
[00:00:38.564,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:38.564,000] <err> os: Current thread: 0x20020368 (unknown)
[00:00:38.566,000] <err> os: Halting system
```

jakub-uC, 2020-09-10:
@nandojve would you mind pushing some branch where you mocked the I2C driver?
If I would be able to reproduce this I could investigate a bit more.

henrikbrixandersen, 2020-09-10:
@nandojve This looks like it could be a generic problem with the shell. Can you confirm that increasing `CONFIG_SHELL_ARGC_MAX` to e.g. 20 allows you to pass in more bytes without crashing?

henrikbrixandersen, 2020-09-10:
According to my Git bisect, the bug was introduced in 512de5ecac62de202cf17676d57c7c9d466dc3e2.

henrikbrixandersen, 2020-09-10:
@nandojve The reason I could not initially reproduce this was that I increased `CONFIG_SHELL_ARGC_MAX` in order to be able to pass more than 9 arguments in to the EEPROM write command. 

This bug is not specific to the EEPROM shell; problem can be reproduced just by typing in a command with more arguments than `CONFIG_SHELL_ARGC_MAX` (e.g. `1 2 3 4 5 6 7 8 9 10 11 12 13 14` with the standard `CONFIG_SHELL_ARGC_MAX=12`).

jakub-uC, 2020-09-10:
@henrikbrixandersen Thank you for finding - we will fix it.

andrewboie, 2020-09-10:
> I observed in the past similar problems with the shell when the thread stack was too low.

FWIW, an MPU fault due to a stack overflow should be reported as a stack overflow on the ">>> ZEPHYR FATAL ERROR" line. If it ever doesn't do that and reports a generic CPU exception instead, can consider that a bug.

PR #28301 (2020-09-14): Fix shell fault when too many arguments in the command were provided
Added fix and test case.

Fixes #28108.
closed 2020-09-14 by carlescufi