zephyrproject-rtos/zephyr #28112: timer/scheduler problem (STM32F407)

lowlander, 2020-09-07:
I have a high prio task (the highest prio in the system, higher than anything else) that should run every 20ms, to  check that I toggle a LED that I measure with a scope. 

At random intervals (sometimes every second, but sometimes it takes 15 seconds, or sometimes twice in a row) it does not wake up after 20 ms but after 100ms (it always seems to be 100ms).  This system is not halted at that time, IRQ handlers do run just that all waits seem delayed 100ms. 

When I disable tickless (`CONFIG_TICKLESS_KERNEL=n CONFIG_TICKLESS_IDLE=n`) the problem goes away, and the task always (tested several hours) starts after 20ms.

The task is scheduled like follows;
```
K_TIMER_DEFINE(t, NULL, NULL);

void task(void* a) {
  k_timer_start(&t, K_MSEC(20), K_MSEC(20));

  while (1) {
    k_timer_status_sync(&t);
    led_on();

    ....

    led_off();
  }
}

```

carlescufi, 2020-09-07:
@lowlander is this a duplicate of https://github.com/zephyrproject-rtos/zephyr/issues/28118

lowlander, 2020-09-07:
> @lowlander is this a duplicate of #28118?

No they are 2 different problems, #28118 seems to be that the system has the wrong idea of what 1ms is, and this issue is that at "random" times the timers are not expired for up to 100ms (this only seems to happen when tickless is used)

pabigot, 2020-09-08:
@lowlander Could you try the fix in #28034 to see if that fixes it?  The behavior here seems similar.

lowlander, 2020-09-08:
Bug #28118 was not Zephyrs fault, and I (with some help) found a plausible explanation for the behavior. This "bug" might also not be zephyrs fault I although the behavior is a bit strange, since it doesn't miss 1 tick but 100ms. This is weird because there is a 1 ms task in my system so even with tickless it should do something every 1ms.

I will do more testing tomorrow and report back.

lowlander, 2020-09-11:
I have not been able to reproduce this, after working around #28118 my code has changed to much. I tried to make a test case on a Olimex board, but never could reproduce the error. I think this might be a combination of the long `irq_lock` in my code and the load of the system (lot of CAN , Ethernet, and SPI IRQs). 

Since my problem "disappeared" and nobody else has seen this problem I am closing this bug report.

closed 2020-09-11 by lowlander