zephyrproject-rtos/zephyr #28113: Embed precise Zephyr version & platform name in sanitycheck output .xml

abrodkin, 2020-09-07:
Currently .xml files produced by the `sanitycheck` script only have information about tests being executed:
```xml
<testsuites>
    <testsuite errors="0" failures="0" name="nsim_em" skipped="0" tests="1" time="1.550905">
        <testcase classname="sample.basic" name="sample.basic.helloworld" time="1.550905" />
    </testsuite>
</testsuites>
```

This is good but for automatic parsing of those results it might be very convenient to have additional data built-in. In particular:
1. Version of Zephyr source tree being used
2. Platform on which tests were run

That suggestion came-up during discussion of test results submission automation here https://github.com/zephyrproject-rtos/test_results/pull/68.

With that new information in place we'll be able to do at least automated verification of PRs with test results:
 * If .xml files were put in the proper folder structure (Zephyr version & platfrorm)
 * If that Zephyr version is at all used for results collection or if the source tree was clean (I guess we don't want results from "dirty" trees with unknown changes).

@maksimmasalski, @hakehuang, @wangnuannuan

PerMac, 2020-09-28:
Isn't the platform name already included in the testsuite attribute: "name="nsim_em"?

hakehuang, 2020-09-28:
> Isn't the platform name already included in the testsuite attribute: "name="nsim_em"?

version is missing

PerMac, 2020-09-28:
I quickly tried to get some more info about JUnit style XML reports and from what I found it seems that there is no "official" format. I think this page might be useful: https://llg.cubic.org/docs/junit/ 
I remember that we had a conclusion that JUnit doesn't support some of the attributes required by us (e.g. version). 
It seems that we can "abuse" the <property> attribute, where any name (e.g environment related stuff like version) can be used with a corresponding value.

abrodkin, 2020-09-28:
Oh, so what stops us from adding another item called for example "version" for each test similarly to what we already have for the platform?

PerMac, 2020-09-28:
I guess that nothing stops us. I assume we didn't know that this `property` attribute can be used? I think we will just have to add a way for the sanitycheck script to know internally which version (or SHA?) of zephyr it is using, as I don't see this info available anywhere during the execution.

abrodkin, 2020-09-29:
@PerMac probably not during execution, but at build time we already define `BUILD_VERSION`, see https://github.com/zephyrproject-rtos/zephyr/blob/master/cmake/git.cmake#L16. Another question now is how to pass it to the sanitycheck script :)

Or given `ZEPHYR_BASE` is known in the `sanitycheck` script we may just run `git describe` in `ZEPHYR_BASE` and record the output.

hakehuang, 2020-09-30:
add more property is a general requirements for sanityscript output, I suppose we will require more and more properity over time.

closed 2020-10-26 by nashif