zephyrproject-rtos/zephyr #28114: subsys: OSDP forces SERIAL=y

erwango, 2020-09-07:
**Describe the bug**
OSDP subsys forces SERIAL Kconfig symbol in its definition:
```
menuconfig OSDP
	bool "Open Supervised Device Protocol (OSDP) driver"
	select RING_BUFFER
	select SERIAL
	select SERIAL_SUPPORT_INTERRUPT
	help
	  Add support for Open Supervised Device Protocol (OSDP)
```

One of the consequence is that sanitycheck filtering deos not operate and sample run on platform not supporting SERIAL:
```
tests:
  sample.mgmt.osdp:
    tags: osdp
    filter: dt_compat_enabled_with_alias("gpio-leds", "led0") and CONFIG_SERIAL
    harness: osdp
```

When running sanitychcek, test is ran on stm32h747i_disco_m4, which does not support SERIAL by default
:
```
$ sanitycheck -b -p stm32h747i_disco_m4 -T samples/subsys/mgmt/osdp
Deleting output directory /local/mcu/zephyrproject/zephyr/sanity-out
INFO    - JOBS: 16
INFO    - Building initial testcase list...
INFO    - 1 test configurations selected, 0 configurations discarded due to filters.
INFO    - Adding tasks to the queue...

ERROR   - stm32h747i_disco_m4       samples/subsys/mgmt/osdp/sample.mgmt.osdp          FAILED: Build failure
ERROR   - see: /local/mcu/zephyrproject/zephyr/sanity-out/stm32h747i_disco_m4/samples/subsys/mgmt/osdp/sample.mgmt.osdp/build.log
INFO    - Total complete:    1/   1  100%  skipped:    0, failed:    1
INFO    - 0 of 1 tests passed (0.00%), 1 failed, 0 skipped with 0 warnings in 6.44 seconds
INFO    - In total 1 test cases were executed on 1 out of total 284 platforms (0.35%)
INFO    - 0 tests executed on platforms, 1 tests were only built.
```

**To Reproduce**
```
$ sanitycheck -b -p stm32h747i_disco_m4 -T samples/subsys/mgmt/osdp
```

**Expected behavior**
OSDP should follow the approach used for similar subsy, such as modem:
```
config MODEM_RECEIVER
	bool "Enable modem receiver helper driver"
	depends on SERIAL_SUPPORT_INTERRUPT
	select UART_INTERRUPT_DRIVEN
	select RING_BUFFER
	help
```

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context about the problem here.

erwango, 2020-09-07:
^^ @cbsiddharth

cbsiddharth, 2020-09-07:
Thank you. I will send PR for this.

galak, 2020-09-08:
Bumped to high since this fails in nightly build CI.

cbsiddharth, 2020-09-08:
@galak, for now disabling OSDP tests on this platform as it needs CONFIG_SERIAL. Please check PR #28130.

closed 2020-09-14 by MaureenHelm