zephyrproject-rtos/zephyr #28117: CoAP/LWM2M: Clean Packet Retransmission Concept

broglep-koubachi, 2020-09-07:
**Summary**
Currently CoAP packet retransmission is not explicitly modelled in the code base. In the CoAP layer the logic about re-transmission is done with [next_timeout](https://github.com/zephyrproject-rtos/zephyr/blob/c6bb8b191d0159b415d79addf76b3dee25215710/subsys/net/lib/coap/coap.c#L1204) and [coap_pending_cycle](https://github.com/zephyrproject-rtos/zephyr/blob/c6bb8b191d0159b415d79addf76b3dee25215710/subsys/net/lib/coap/coap.c#L1220). A second [re-transmission concept](https://github.com/zephyrproject-rtos/zephyr/blob/d431b56587a9afbee091180d3511257a148a359f/subsys/net/lib/lwm2m/lwm2m_object.h#L426) is only partially implemented in LWM2M layer. There is a need for clean packet retransmission handling

**Details**
`next_timeout` doubles the previous timeout 3 times, after that it signal end of re-transmission by returning the same timeout as before. This implementation rather hides how re-transmissions are done behind timeout generation. Further the current approach makes it difficult to fix the already existing TODO about `random generated initial ACK timeout`.  In the LWM2M layer there is a [send_attempts](https://github.com/zephyrproject-rtos/zephyr/blob/d431b56587a9afbee091180d3511257a148a359f/subsys/net/lib/lwm2m/lwm2m_object.h#L426) for `lwm2m_message`, but it not used actually (just being increment at the moment, never read)

**Preferred Solution**

- CoAP Layer:  the number of times a packet has been sent is tracked in `coap_pending` struct. `next_timeout` uses the number of transmission for its calculation of timeout. `coap_pending_cycle` should ideally only handle timeout, decision to stop retransmission happens on a higher level (in [CoAP server](https://github.com/zephyrproject-rtos/zephyr/blob/c6bb8b191d0159b415d79addf76b3dee25215710/samples/net/sockets/coap_server/src/coap-server.c#L954) and [LWM2M engine](https://github.com/zephyrproject-rtos/zephyr/blob/d431b56587a9afbee091180d3511257a148a359f/subsys/net/lib/lwm2m/lwm2m_engine.c)). If `coap_pending_cycle` should also handle stop of retransmissions, it should use the number of times a packet is sent to return if transmission should be stopped.

- LWM2M Layer: LWM2M Engine properly uses it own re-transmission on top of CoAP. e.g. let CoAP re-transmit 3 times, if it fails, re-transmit lwm2m_message once. Or if not deemed usefully, completely remote the LWM2M message re-transmission concept (remove send_attemps from lwm2m_message)

**Motivation**
A clean retransmission concept across layers facilitates understanding of the code base and how retransmissions works in the zephyr client. Further it helps implementing of more advanced features like custom retransmission schemes for different LWM2M messages (e.g. bootstrap requests)

rlubos, 2020-09-10:
I agree that the retransmission logic in CoAP could be more explicit. Especially, that currently the retransmission count is hardcoded  (in the aforementioned `next_timeout()` function) and with a proper retransmission counter it could be made configurable (CoAP spec does not put a hard requirement on the MAX_RETRANSMIT value).

> LWM2M Engine properly uses it own re-transmission on top of CoAP

Could you clarify what did you mean? According to LwM2M spec, it does not specify any retransmission mechanism, other than the one at the CoAP level (please correct me if I'm wrong).

In general, my feeling is that Zephyr would benefit from an additional CoAP layer, built on top of current APIs, that would take care of the mechanics defined in the CoAP RFC (CON retransmission, request/response matching, message deduplication). Currently, the burden of implementing these specific CoAP features is on the application (or the LwM2M lib), while the implementation could be common for all CoAP users.

broglep-work, 2020-09-14:
> > LWM2M Engine properly uses it own re-transmission on top of CoAP
> 
> Could you clarify what did you mean? According to LwM2M spec, it does not specify any retransmission mechanism, other than the one at the CoAP level (please correct me if I'm wrong).

While the LwM2M spec does not specify any retransmission, it could be useful if LwM2M lib has its own mechanism for re-transmission if the underlying CoAP transmission does not work. Something like this is present in the bootstrapping client, it creates one lwm2m message, if the corresponding CoAP packet transmission fails, it restart the bootstrap process (by restarting the whole lwm2m engine) and creates a new lwm2m message. You could also have this without an engine restart where one could configure to retry sending a lwm2m message. I assume `send_attempts` was meant for this. But if the CoAP retransmission is flexible enough, there is no need for having a retransmission concept a level higher on LwM2M

PR #31406 (2021-01-19): net: coap: Rework packet retransmission concept
* **Rework pending retransmission logic**

    Introduce retransmission counter to the coap_pending structure. This allows to simplify the retransmission logic and allows to keep track of the number of remaining retranmissions.
  
    Additionally, extend the `coap_pending_init()` function with `retries` parameter, which allows to set the retransmission count individually for each confirmable transaction.
  
    Fixes #28117

* **Randomize initial ACK timeout** 

    Add Kconfig option to randomize the initial ACK timeout, as specified in RFC 725 [... cut]
commit 538e19ee (2021-01-19): net: coap: Rework pending retransmission logic

Introduce retransmission counter to the coap_pending structure. This
allows to simplify the retransmission logic and allows to keep track of
the number of remaining retranmissions.

Additionally, extend the `coap_pending_init()` function with `retries`
parameter, which allows to set the retransmission count individually for
each confirmable transaction.

Fixes #28117
closed 2021-01-19 by nashif