zephyrproject-rtos/zephyr #28118: timers strange rounding errors

lowlander, 2020-09-07:
I have the following setup, for a task with the highest (highest cooperative prio) ;

```
K_TIMER_DEFINE(t, NULL, NULL);

void task(void* a) {
  k_timer_start(&t, K_MSEC(20), K_MSEC(20));

  while (1) {
    k_timer_status_sync(&t);
    led_on();

    ....

    led_off();
  }
}
```

My expectation is that the task will run every 20ms, but it doesn't. The timing is highly depended on the `CONFIG_SYS_CLOCK_TICKS_PER_SEC` setting and also on if the system is tickless or not. Here the times I measured for tick settings form 1000 to 10000.

```
       tickless     with tick
1000   20           20
2000   20.5         20.5
3000   20.66-21     20.666
4000   21           20.75
5000   21           20.8
6000   21-21.3      20.825
10000  21-21.3      20.9
```

Here is an example what that looks like on a scope (tick == 3000) ; 

[image]

It seems the time that is waited it  `x + (1ms - ticktime)`  for example tick 3000 => ticktime 0.3333 ms  gives `20 + (1 - 0.333) = 20.666` 

So the more accurate the delay could be calculated the more it is wrong. 

When tickless is selected it seems to do the same thing, just at higher tick values it starts to jitter and between certain values.

This is on a STM32F407 running at 168MHz. 

**expected behavior:** On a realtime OS 20ms should be 20ms and not 20ms+~1ms especially since the timer resolution is available.

lowlander, 2020-09-07:
Segger Systemview tells me the 20ms is exactly 20ms, at the same time my scope tells me it is not. Sounds like the system time isn't correctly calculated? So the scheduling works fine, just that the 20ms isn't 20ms.

lowlander, 2020-09-07:
After some more testing it seems to be related how the system is reset'ed, at power on (POR) things work OK, but after a reset with the debugger (J-link/Ozone) the timing is wrong. Will investigate more and try to reproduce it on one of my Nucleo boards.

lowlander, 2020-09-08:
OK I did some more debugging and the difference between POR and debugger-reset is that our bootloader is skipped. After some more searching I found that our bootloader enables the STM32 FLASH Icache and Dcache. Zephyr itself does not seem to enable those, so after a debugger reset the cache is disabled. To check I added code to enable the Icache and Dcache in Zephyr and now the 20ms is really 20ms. 

Some very big questions remain, that I hope some STM32 expert (@erwango) here can answer; why does the timing depend so much on the FLASH settings? It really makes no sense to me. I understand that code execution is slower, almost twice as slow, my LED-on time (low voltage in the scope image) goes from 1.2ms to 700us when flash cache is enabled, but why is Zephyr not able to get a stable 20ms timer? And why does it still depend on the number of ticks per second? 

I will try to make a reproducible test that works on one of the nucleo boards.

lowlander, 2020-09-08:
I couldn't find a 407 Nucleo so I used an Olimex E407, with the following main loop;
```
	k_timer_start(&t, K_MSEC(20), K_MSEC(20));

	while (1) {
		k_timer_status_sync(&t);

		gpio_pin_set(dev, PIN, 1);

		flags = irq_lock();

		k_busy_wait(600); // safety function 

		irq_unlock(flags);

		gpio_pin_set(dev, PIN, 0);
	}
```
The big busy_wait is a functional safety check that runs every 20ms to check if the system is still operating correctly (RAM unchanged for critical areas, flash unchanged,  certain devices still operational, etc.)  this may not be interrupted at all.

This setup above with the 600us busywait works fine, but as soon as the busywait goes above 1ms things go wrong with the timing. Than the 20ms delay depends on the tick count as shown in the table above.

Disabling the FLASH caches causes the time needed by our safety function to  go from 600us to about 1.2ms, and with that the 20ms starts to end up being wrong. 

So when I set the tick to 1000 - 10000 and check what the max irq block time maybe I get the following table;
```
 1000 1985us 
 2000  985us
 3000  650us
 4000  485us
 5000  385us
10000  185us
```
With a bit of overhead removed that means the irq block may not be more than `2 * tick_time`. 

Not sure if this can be prevented because with the default 10000 tick/sec the 200us isn't that absurdly long to block irqs. 

Next I will try to reproduce PR #28112 to see if I can reproduce that in the same way.

ABOSTM, 2020-09-08:
@lowlander ,
Disabling interrupts for 600us or 1.2ms is quite long. And it looks to me like it causes drift in zephyr tick count:
CONFIG_SYS_CLOCK_TICKS_PER_SEC is generally 10000 which means 100us.
STM32 implementation use systick to generate an interrupt every tick, so every 100us in this case.
So if interrupts are blocked for 1.2ms you may loose about 12 ticks.
It explains that with lower CONFIG_SYS_CLOCK_TICKS_PER_SEC = 1000, tick duration is 1ms.
And so blocking interrupts for 600ms is not much of an issue (at least for systick).
Blocking for 1.2ms there is a small probability you miss 1 tick.
So it also explain the influence of cache: the faster the code run, the less tick you miss.

From my point of view, you should not block interrupts for more than a single tick. Even `2 * tick_time` is too much.
In the example below if IRQ are blocked just a bit more than `tick_time` then 1st IRQ is lost
```
IRQ blocked  <-------------->
             |              |
           TICK            TICK+1
```

lowlander, 2020-09-08:
@ABOSTM would it be possible for Zephyr to detect that it missed a tick? For example by looking at a free running counter to see more time has passed than N ticks (and so N-1 ticks were missed). Do you have any idea if it would be worth it for me to look into that looking into that ?

ABOSTM, 2020-09-08:
@lowlander, with current implementation based on systick, it is not possible to detect missed TICK. because there is no hardware track of this, and thus software is enable to know interrupts are missed.
Theoretically, it could be possible to track missed interrupt if TICK is port on STM32 hardware timer peripheral instead of systick (taking advantage of timer instance supporting repetition counter), but that means full rework of driver. So I don't advice that.

Instead I would suggest:
* Is it possible to split the safety operations and release IRQ in between? For example: execute RAM check, then release IRQ, then execute FLASH check, ...
Thus IRQ are blocked for shorter time.
* fine tune  CONFIG_SYS_CLOCK_TICKS_PER_SEC (find an acceptable compromise)
* Why is it so important to execute safety exactly at 20ms ? what if it is executed a bit more often, like each 19ms (+-1) ?

lowlander, 2020-09-08:
> @lowlander, with current implementation based on systick, it is not possible to detect missed TICK. because there is no hardware track of this, and thus software is enable to know interrupts are [...]

If we have the need for more exact timing we can use a hardware timer and ISR to do that, I was just wondering if there was an easy "fix" that could also benefit others.

> 
> Instead I would suggest:
> 
> * Is it possible to split the safety operations and release IRQ in between? For example: execute RAM check, then release IRQ, then execute FLASH check, ...
>   Thus IRQ are blocked for shorter time.

We basically already do that, but some parts are hard to split up. But with the FLASH cache enabled things seem fast enough.

> * fine tune  CONFIG_SYS_CLOCK_TICKS_PER_SEC (find an acceptable compromise)

1000 seems OK, I haven't been able to measure any skips.

> * Why is it so important to execute safety exactly at 20ms ? what if it is executed a bit more often, like each 19ms (+-1) ?

It isn't that big a deal, the problem was that I didn't know the cause, and that worried me. 

@ABOSTM thank you for your time and advice, and since Zephyr isn't at fault here I will close this bug report.

closed 2020-09-08 by lowlander