zephyrproject-rtos/zephyr #28124: Linking POSIX API against external lib

pablogs9, 2020-09-07:
Hello, I'm trying to use Zephyr POSIX API inside an external lib. In order to do that for example I test the `sample/application_development/external_lib/` with the following changes:

```
diff --git a/samples/application_development/external_lib/mylib/src/mylib.c b/samples/application_development/external_lib/mylib/src/mylib.c
index 3f18508c35..7d6b851fe6 100644
--- a/samples/application_development/external_lib/mylib/src/mylib.c
+++ b/samples/application_development/external_lib/mylib/src/mylib.c
@@ -6,9 +6,13 @@
 
 #include "mylib.h"
 #include <stdio.h>
+#include <time.h>
 
 int mylib_hello_world(void)
 {
+       struct timespec tp;
+       clock_gettime(CLOCK_MONOTONIC, &tp);
+
        printf("mylib says: Hello World!\n");
        return 0;
 }
diff --git a/samples/application_development/external_lib/prj.conf b/samples/application_development/external_lib/prj.conf
index 8b042db540..f274e81052 100644
--- a/samples/application_development/external_lib/prj.conf
+++ b/samples/application_development/external_lib/prj.conf
@@ -1 +1,4 @@
 CONFIG_STDOUT_CONSOLE=y
+
+CONFIG_POSIX_API=y
+CONFIG_APP_LINK_WITH_POSIX_SUBSYS=y
```

And I get the following error:

```
src/mylib.c: In function 'mylib_hello_world':
src/mylib.c:14:2: warning: implicit declaration of function 'clock_gettime' [-Wimplicit-function-declaration]
   14 |  clock_gettime(CLOCK_MONOTONIC, &tp);
      |  ^~~~~~~~~~~~~
src/mylib.c:14:16: error: 'CLOCK_MONOTONIC' undeclared (first use in this function)
   14 |  clock_gettime(CLOCK_MONOTONIC, &tp);
      |                ^~~~~~~~~~~~~~~
```

If I put the same couple of lines inside the `main.c` of the app it builds. Can you provide me some help with the required configuration for this requirement?

pabigot, 2020-09-07:
Zephyr's POSIX infrastructure doesn't support the clock_gettime() API, and does not have the various `CLOCK_*` sources.  There was a vision for an enhancement that could support that (#19030) but it didn't go anywhere.

pabigot, 2020-09-07:
> Zephyr's POSIX infrastructure doesn't support the clock_gettime() API, and does not have the various `CLOCK_*` sources. 

No, I'm wrong about that.

I applied the patch you give above to the external library function and am able to build for that platform.  This may be a recent fix due to #27939.  Please check the current master branch, or 2.4.0-rc1.

pablogs9, 2020-09-08:
I guess that I have yesterday's branch of Zephyr. Here you have a list of commands for replicating it:

```
mkdir test_zephyr
cd test_zephyr
west init .
west update
pip3 install --user -r zephyr/scripts/requirements.txt

cd zephyr
git fetch; git checkout master; git pull;
git apply patch.txt # Find patch file attached below
cd ..

export ZEPHYR_TOOLCHAIN_VARIANT=zephyr
export ZEPHYR_SDK_INSTALL_DIR=[YOUR SDK]
west build -b olimex_stm32_e407 -p auto zephyr/samples/application_development/external_lib/
```

[patch.txt](https://github.com/zephyrproject-rtos/zephyr/files/5186238/patch.txt)

pabigot, 2020-09-08:
Thanks; apparently I had patched the version in my personal development area then built in my nordic development.  I've reproduced the problem now.

Based on the [description in the documentation](https://docs.zephyrproject.org/latest/guides/portability/posix.html#posix-support) this should probably work, but it doesn't.

The problem is that `<time.h>` when building the external library resolves to the toolchain `<time.h>`.  It needs to resolve to `${ZEPHYR_BASE}/include/posix/time.h`.  This can be done by modifying the source to include `<posix/time.h>` instead.

To do it without modifying the source you'd have to somehow communicate the `posix_sys` interface library flags down into the externally built library.  That can be done with:

```
--- a/samples/application_development/external_lib/CMakeLists.txt
+++ b/samples/application_development/external_lib/CMakeLists.txt
@@ -20,7 +20,7 @@ zephyr_get_compile_definitions_for_lang_as_string(       C definitions)
 zephyr_get_compile_options_for_lang_as_string(           C options)
 
 set(external_project_cflags
-  "${includes} ${definitions} ${options} ${system_includes}"
+  "${includes} -I${ZEPHYR_BASE}/include/posix ${definitions} ${options} ${system_includes}"
   )
 
 include(ExternalProject)
```

but this doesn't work because Zephyr's `<posix/time.h>` includes `<kernel.h>` which includes a generated header for kernel system calls that isn't available when the external library is being built.

To make progress, change the source to include `<posix/time.h>`.

Fixing the larger issue of making the Zephyr POSIX subsystem able to be used by an external library is likely to be a major effort.  I'm going to mark this a bug so we can discuss it in triage today, but it'll probably get converted to an enhancement.

pablogs9, 2020-09-08:
Ok, thank you very much @pabigot

closed 2020-11-22 by github-actions[bot]

pablogs9, 2021-02-02:
@pabigot @MaureenHelm any update on this?

pabigot, 2021-02-02:
Not to my knowledge.