zephyrproject-rtos/zephyr #28131: Crash while serving large files via HTTP with TCP2

armandciejak, 2020-09-07:
**Describe the bug**
While testing TCP2 with the _dumb_http_server_ sample application on the MIMXRT1050_EVK board I experimented several crashes when increasing the size of the served file.

**To Reproduce**
Steps to reproduce the behavior:
1. Apply patch **ADD LINK**
2. west build -b mimxrt1050_evk zephyr/samples/net/sockets/dumb_http_server
3. west flash
4. ping the board on a terminal `ping -A 192.168.1.241`
5. load the page with curl on another terminal `while true; do time curl http://192.168.1.241:8080 --output test || break; done`

**Expected behavior**
No crash.

**Impact**
Since we need to serve "large" files via HTTP this problem prevents us to deliver stable firmware to our customers. So far we are using TCP1 but we have even more problems with it, therefore we are hoping to switch to TCP2 in order to solve them.

**Logs and console output**
Here are several examples of crash log.

```
Connection #150 from 192.168.1.146
Err[00:00:55.861,000] <err> net_pkt: Data buffer (1500) allocation failed.
[00:00:55.861,000] <err> net_tcp: conn: 0x80007888 packet allocation failed, len=1452
[00:01:34.637,000] <err> os: ***** USAGE FAULT *****
[00:01:34.637,000] <err> os:   Attempt to execute undefined instruction
[00:01:34.637,000] <err> os: r0/a1:  0x80005010  r1/a2:  0x0f0e070d  r2/a3:  0x0000000b
[00:01:34.637,000] <err> os: r3/a4:  0x80000000 r12/ip:  0x0f0e070c r14/lr:  0x6000c4dd
[00:01:34.637,000] <err> os:  xpsr:  0x01000000
[00:01:34.637,000] <err> os: Faulting instruction address (r15/pc): 0x20000004
[00:01:34.637,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:01:34.637,000] <err> os: Current thread: 0x80001810 (unknown)
[00:01:34.655,000] <err> os: Halting system
```

```
Connection #0 from 192.168.1.146
Err[00:00:07.607,000] <err> net_pkt: Data buffer (1500) allocation failed.
[00:00:07.607,000] <err> net_tcp: conn: 0x80007888 packet allocation failed, len=1452
[00:00:07.607,000] <err> os: ***** BUS FAULT *****
[00:00:07.607,000] <err> os:   Instruction bus error
[00:00:07.607,000] <err> os: r0/a1:  0x80005058  r1/a2:  0x0d0d0f0f  r2/a3:  0x071d0d0d
[00:00:07.607,000] <err> os: r3/a4:  0x80000000 r12/ip:  0x0d0d0f0e r14/lr:  0x6000c4dd
[00:00:07.607,000] <err> os:  xpsr:  0xa1000000
[00:00:07.607,000] <err> os: Faulting instruction address (r15/pc): 0x071d0d0c
[00:00:07.607,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:07.607,000] <err> os: Current thread: 0x80001810 (unknown)
[00:00:07.651,000] <err> os: Halting system
```

```
Connection #1 from 192.168.1.146
Err[00:00:08.420,000] <err> net_pkt: Data buffer (1500) allocation failed.
[00:00:08.420,000] <err> net_tcp: conn: 0x80007888 packet allocation failed, len=1452
[00:00:08.421,000] <err> os: ***** BUS FAULT *****
[00:00:08.421,000] <err> os:   Instruction bus error
[00:00:08.421,000] <err> os: r0/a1:  0x80004f80  r1/a2:  0x4f0f0d0d  r2/a3:  0x0f2f0b1f
[00:00:08.421,000] <err> os: r3/a4:  0x80000000 r12/ip:  0x4f0f0d0c r14/lr:  0x6000c4dd
[00:00:08.421,000] <err> os:  xpsr:  0x81000000
[00:00:08.421,000] <err> os: Faulting instruction address (r15/pc): 0x0f2f0b1e
[00:00:08.421,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:08.421,000] <err> os: Current thread: 0x80001810 (unknown)
[00:00:08.465,000] <err> os: Halting system
```

```
$ZEPHYR_SDK_INSTALL_DIR/arm-zephyr-eabi/bin/arm-zephyr-eabi-addr2line -e build/zephyr/zephyr.elf 0x6000c4dd 
build/zephyr/include/generated/syscalls/kernel.h:168
```

`kernel.h:168` is the location of the `k_yield` function.

**Environment (please complete the following information):**
 - OS: Kubuntu 20.04
 - Toolchain: Zephyr SDK 0.11.4, west 0.7.3
 - Zephyr: v2.4.0-rc1

armandciejak, 2020-09-07:
I cannot reproduce it if the  network has latency (tried with PLC adapters between my PC and the eval board).

armandciejak, 2020-09-07:
I'm wondering if that could be the Ethernet driver but I do not have another eval board at end. If someone can try to reproduce it with a different SOC that would be interesting.

jukkar, 2020-09-07:
> Steps to reproduce the behavior:
> 
>     1. Apply patch in [test-tcp2.zip](https://github.com/zephyrproject-rtos/zephyr/files/5182994/test-tcp2.zip)
>     2. west build -b mimxrt1050_evk zephyr/samples/net/sockets/dumb_http_server
>     3. west flash
>     4. ping the board on a terminal `ping -A 192.168.1.241`
>     5. load the page with curl on another terminal `while true; do time curl http://192.168.1.241:8080 --output test || break; done`
> 
> ****

The test-tcp2.zip has this change `CONFIG_NET_BUF_TX_COUNT=256`. This means that we are able to send max 32kB file because each net_buf is 128 bytes long by default. Normally this should not be an issue but the `dumb_http_server` is very dumb as the name implies and tries to send the max buffer in one go regardless of the size. So this will fail for those bigger files you have in the zip file. The application should be changed to send data in smaller chunks or the amount of net_buf's should be increased.

> I'm wondering if that could be the Ethernet driver but I do not have another eval board at hand. If someone can try to reproduce it with a different SOC that would be interesting.

BTW, have you tried the app with qemu or native_posix. If there is no crash there, then it could indicate issue in the driver.

jukkar, 2020-09-07:
I tried this with qemu_x86 and native_posix boards. Both are unable to send larger files, I tried the 34kb one. The application fails but does not crash in these boards. I also tried with valgrind when using native_posix and it did not report any NULL pointer access or similar. I have some ARM based boards with Ethernet and can try them next.

armandciejak, 2020-09-08:
> The test-tcp2.zip has this change `CONFIG_NET_BUF_TX_COUNT=256`. This means that we are able to send max 32kB file because each net_buf is 128 bytes long by default. Normally this should not be an [...]

To me there is a problem with the behaviour of send. It should not be a problem to try sending a large amount of data since send returns how much it was able to process, you can use that information to send the remaining data, like it is done in the `dumb_http_server`. To me the implementation in `dumb_http_server` is perfectly fine. The problem is that the call fails when the memory buffers are exhausted instead of blocking.

I think that send must behave like explained in the send man page:
> When the message does not fit into the send buffer of the socket, send() normally blocks, unless the socket has been placed in nonblock‐ing I/O mode.

Note: I tried to send data in smaller chunks by modifying the send loop in the `dumb_http_server` but that didn't change anything. The only hack which helped was to add a lengthy operation (e.g. printk) after the call to send.

jukkar, 2020-09-08:
Clarification how the system works atm:

* Application can send x amount of data, x can be quite large
* Net stack checks the MTU, and then allocates enough net_buf's so that the MTU size packet can be send. The data to be sent, is passed to TCP module for sending. TCP clones the packet so that we can re-send any data that is lost.
* Depending on how fast the network is and how fast the application sends the data, it might happen that application sends so much data that TCP sending queue fills up and uses all the memory.

It looks like this is the scenario in your case. If you slow the sending (like putting printk), we have time to receive the ACK to the sent data and release the data pending in TCP send queue and then send more data.

In any case, there should be no crashes etc in the system.

armandciejak, 2020-09-09:
I opened another issue #28216 for the problem regarding the behavior of send, so that only the crash problem is addressed in this issue.

jukkar, 2020-09-09:
I have been investigating this and have following observations:

* I can replicate a crash but only with fast device and only randomly. I have been using Atmel sam-e70 and even with that the bug disappers easily if I enable TCP debugging. I was not able to reproduce this with frdm_k64f which also has Ethernet support.
* In my initial analysis, it looks like something is overwriting `accept_cb` field in `tcp2_priv.c:struct tcp`. Sometimes the accept_cb is set to NULL, sometimes it contains an invalid pointer. Both ot these lead to a crash as we try to call this callback.
* I then tried to attach debugger to this. Unfortunately gdb returns `Remote replied unexpectedly to 'vMustReplyEmpty': vMustReplyEmpty` so was not able to see who overwrites the callback pointer.

@armandciejak If you are able to connect gdb to the device, could you try to attach a watchpoint to the accept_cb field in conn struct. In my run the device is crashing in tcp2.c:`tcp_recv()`  `conn_old->accept_cb()`. Can you check if the crashing is happening in the same location in your device?

armandciejak, 2020-09-21:
I've tested v2.4.0-rc2 and I'm still experimenting illegal data accesses on the MIMXRT1050-EVK board. I'm using the test setup described in #28216 as well as the MPU hack for catching NULL pointer access.

Here is the fault that got reported:

```
Connection #2889 from 192.168.1.170
Connection from 192.168.1.170 closed
Connection #2890 from 192.168.1.170
[00:05:55.465,000] <err> os: ***** MPU FAULT *****
[00:05:55.465,000] <err> os:   Data Access Violation
[00:05:55.465,000] <err> os:   MMFAR Address: 0xc
[00:05:55.465,000] <err> os: r0/a1:  0x80000a6c  r1/a2:  0x800079d0  r2/a3:  0x80000a6c
[00:05:55.465,000] <err> os: r3/a4:  0x80008150 r12/ip:  0x00000000 r14/lr:  0x600123f7
[00:05:55.465,000] <err> os:  xpsr:  0x81000000
[00:05:55.465,000] <err> os: Faulting instruction address (r15/pc): 0x60011fa8
[00:05:55.465,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:05:55.465,000] <err> os: Current thread: 0x80001998 (rx_workq)
[00:05:55.504,000] <err> os: Halting system
```

The instruction causing the fault is in `ethernet_recv`:

```
60011f9c <ethernet_recv>:
{
60011f9c:       e92d 43f8       stmdb   sp!, {r3, r4, r5, r6, r7, r8, r9, lr}
        return pkt->frags->data;
60011fa0:       690b            ldr     r3, [r1, #16]
60011fa2:       4602            mov     r2, r0
60011fa4:       460c            mov     r4, r1
60011fa6:       689d            ldr     r5, [r3, #8]
        uint16_t type = ntohs(hdr->type);
60011fa8:       89ab            ldrh    r3, [r5, #12]
60011faa:       021e            lsls    r6, r3, #8
60011fac:       ea46 2613       orr.w   r6, r6, r3, lsr #8
        switch (type) {
60011fb0:       f248 63dd       movw    r3, #34525      ; 0x86dd
        uint16_t type = ntohs(hdr->type);
60011fb4:       b2b6            uxth    r6, r6
```

jukkar, 2020-09-21:
Thanks for the info. I was not able to see the crash but according to your backtrace, the fix in #28568 should help

armandciejak, 2020-09-22:
@jukkar With #28568 I cannot reproduce the crash anymore :+1:

PR #28568 (2020-09-22): net: Drop incoming packet if there is no data in it
If the network driver for some reason did not set the data in
the network packet properly, then just drop it as we cannot do
anything with just plain net_pkt.

Fixes #28131
commit c54a511d (2020-09-22): net: Drop incoming packet if there is no data in it

If the network driver for some reason did not set the data in
the network packet properly, then just drop it as we cannot do
anything with just plain net_pkt.

Fixes #28131
closed 2020-09-22 by carlescufi