zephyrproject-rtos/zephyr #28133: using nrf52dk_nrf52832 with serial disabled

fromeijn, 2020-09-07:
**Describe the bug**
When using the blink example on nrf52dk_nrf52832 with `CONFIG_SERIAL=n` in `prj.conf`, the board starts but hangs. Tried also with serial enabled and than at  runtime set uart in power off mode with `device_set_power_state(dev, DEVICE_PM_OFF_STATE, NULL, NULL);` but more or less the same result. Also tried on the nrf52840dk_nrf52840 but this device doenst have the same issues.

**To Reproduce**
Steps to reproduce the behavior:
1. cd zephyr/samples/basic/blinky
2. west build -p auto -b nrf52dk_nrf52832
3. west flash
4. flashing LED1
5. add `CONFIG_SERIAL=n` in `prj.conf`
6. west build -p auto -b nrf52dk_nrf52832
8. west flash
9. no flashing LED1 (also LED2 is lighting up for some reason)

**Expected behavior**
I would expect the device to run also without uart, because it is not used at all in the blink example. Also saves power if not used.

**Impact**
Unable to save power used by uart.

**Logs and console output**
NA

**Environment (please complete the following information):**
 - OS: Linux (Ubuntu 20.04 64bit)
 - Toolchain Zephyr SDK (zephyr-sdk-0.11.3), V2.3-branch
 - Commit c0d3c8004422b359fd2b6257a2c8e226eff54dd6

**Additional context**
NA

carlescufi, 2020-09-07:
Have you tried setting the `status` to `disabled` in the `.dts` file or in an overlay of your own?

fromeijn, 2020-09-07:
Yes I have tried that, also has same result.

pabigot, 2020-09-07:
Using the unmodified `samples/basic/blinky` in current Zephyr development master with `nrf52dk_nrf52832`:

I can't duplicate setting `CONFIG_SERIAL=n` causing LED1 to not flash and LED2 to light up.  LED1 behaves as expected.

As far as power differences: Measuring current on that board fed by a 3V3 battery directly to VDD/GND during the periods when the LED is off I get:

With uCurrent:
* Standard config: 466 uA
* Add `CONFIG_SERIAL=n`: 290 uA

With CurrentRanger:
* Standard config: 475 uA
* Add `CONFIG_SERIAL=n`: 295 uA

With nRF-PPK (which IIUC only measures the MCU current) connected to the nRF52-DK powered via USB  I get:
* Standard config: 309 uA
* Add `CONFIG_SERIAL=n`: 1.5 uA

pabigot, 2020-09-07:
Retesting with v2.3.0 release I don't see any difference.

fromeijn, 2020-09-07:
mmm stange, thanks for checking out! btw im using the correct board, right? `PCA10040` (if the version matters, V0.9.0, 2015.42)

pabigot, 2020-09-07:
Yes, pca10040 is what I call it too (now nrf52dk_nrf52832).  The hardware version might be part of the problem: I have and tested on v1.1.1 2017.1 and v2.0.0 2020.9.  Yours appears to be pretty old, and might even be a PDK which IIRC used experimental silicon with serious problems.

fromeijn, 2020-09-08:
I have ordered a new dev board, I will retry tomorrow!

fromeijn, 2020-09-10:
Tested with a new board, problem was indeed that board not zephyr, thanks for helping out. I will close this issue

closed 2020-09-10 by fromeijn