zephyrproject-rtos/zephyr #28138: No more able to flash board on windows

ABOSTM, 2020-09-07:
**Describe the bug**
I could no more flash boards with openocd on windows (nucleo_l476rg, nucleo_f746zg, ...):

> Error: couldn't open C:zephyrzephyr_projectzephyuildzephyrzephyr.hex

Clearly there is an issue with path where` \ `have been removed.
Note: I am working on windows and have recently perform a west update.
I could workaround by replacing `\` by `/` in hex_name (in openocd.py).
```
    def do_flash(self, **kwargs):
        self.hex_name = self.hex_name.replace('\\', '/')
        . . .
```
But I am not sure it is the right place. Specially I don't know if it is specific to openocd or if the same occurs for other runners (pyocd, ...).

After a bisect, with 'west update' between each step, I found the guilty commit:
sha1: 3124c02987071b045b33bbfa2d12606b2ff0a757

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p auto -b nucleo_l476rg samples/basic/blinky
2. west flash
3.  See error

**Expected behavior**
Successful flashing over openocd

**Logs and console output**

> -- west flash: using runner openocd
> -- runners.openocd: Flashing file: C:\zephyr\zephyr_project\zephyr\build\zephyr\zephyr.hex
> Open On-Chip Debugger 0.10.0 (2020-04-08) [https://github.com/sysprogs/openocd]
> Licensed under GNU GPL v2
> libusb1 09e75e98b4d9ea7909e8837b7a3f00dda4589dc3
> For bug reports, read
>         http://openocd.org/doc/doxygen/bugs.html
> Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
> Info : clock speed 500 kHz
> Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
> Info : STLINK v2.1 JTAG v35 API v2 M26 VID 0x0483 PID 0x374B
> Info : using stlink api v2
> Info : Target voltage: 3.246689
> Info : Unable to match requested speed 500 kHz, using 480 kHz
> Info : Stlink adapter speed set to 480 kHz
> Info : stm32l4x.cpu: hardware has 6 breakpoints, 4 watchpoints
> Info : Listening on port 3333 for gdb connections
>     TargetName         Type       Endian TapName            State
> --  ------------------ ---------- ------ ------------------ ------------
>  0* stm32l4x.cpu       hla_target little stm32l4x.cpu       running
> 
> Info : Unable to match requested speed 500 kHz, using 480 kHz
> Info : Stlink adapter speed set to 480 kHz
> Info : Unable to match requested speed 500 kHz, using 480 kHz
> target halted due to debug-request, current mode: Thread
> xPSR: 0x01000000 pc: 0x08000f34 msp: 0x20000790
> Error: couldn't open C:zephyrzephyr_projectzephyuildzephyrzephyr.hex
> auto erase enabled
> FATAL ERROR: command exited with status 1: 'C:/Users/frq07334/AppData/Roaming/GNU MCU Eclipse/OpenOCD/OpenOCD-20200408-0.10.0/bin/openocd.exe' -s 'C:/zephyr/zephyr_project/zephyr/boards/arm/nucleo_l476rg\support' -f 'C:/zephyr/zephyr_project/zephyr/boards/arm/nucleo_l476rg\support\openocd.cfg' -c init -c targets -c 'reset halt' -c 'flash write_image erase C:\zephyr\zephyr_project\zephyr\build\zephyr\zephyr.hex' -c 'reset halt' -c 'verify_image C:\zephyr\zephyr_project\zephyr\build\zephyr\zephyr.hex' -c 'reset run' -c shutdown

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain : gnuarmemb
 - Commit SHA: 3124c02987071b045b33bbfa2d12606b2ff0a757

ABOSTM, 2020-09-07:
^^ @mbolivar-nordic @carlescufi @erwango

sdaglish, 2020-09-08:
I can confirm the same issue when flashing to a stm32f429i_disc1 board.

Modifiying openocd.py as mentioned also fixed the issue, allowing the board to be flashed.

**Environment (please complete the following information):**

* OS: Windows 10
* Toolchain : gnuarmemb

PR #28201 (2020-09-09): scripts: runners: fix openocd on Windows
Commit 3124c02 ("cmake: flash/debug: refactor runner configuration")
changed the way the hex_file and elf_file inputs in the RunnerConfig
object are created. In particular, they are now host-style paths.

This breaks flashing with openocd on Windows, which doesn't handle that
properly. Fix that by "casting" the internal hex_file and elf_file
attributes to POSIX paths.

Fixes: #28138
commit 1eef1e3c (2020-09-09): scripts: runners: fix openocd on Windows

Commit 3124c02 ("cmake: flash/debug: refactor runner configuration")
changed the way the hex_file and elf_file inputs in the RunnerConfig
object are created. In particular, they are now host-style paths.

This breaks flashing with openocd on Windows, which doesn't handle that
properly. Fix that by "casting" the internal hex_file and elf_file
attributes to POSIX paths.

Fixes: #28138
closed 2020-09-09 by ioannisg