zephyrproject-rtos/zephyr #28145: nRF52840 Dongle cannot scan LE Coded PHY devices

kasapreyizz, 2020-09-08:
I am working with Zephyr version 2.4 on nRF52840-Dongle for adding ability to scan CODED PHY devices. My host is Raspberry Pi3 with kernel  version 5.4.51 v7+,
HCI_USB sample is used with following configuration:

CONFIG_STDOUT_CONSOLE=y
CONFIG_GPIO=y
CONFIG_SERIAL=y
CONFIG_UART_INTERRUPT_DRIVEN=y

CONFIG_BT=y
CONFIG_BT_H4=y
CONFIG_BT_CTLR=y
CONFIG_BT_HCI=y

CONFIG_USB=y
CONFIG_USB_DEVICE_STACK=y
CONFIG_USB_DEVICE_BLUETOOTH=y
CONFIG_USB_DEVICE_BLUETOOTH_VS_H4=y

CONFIG_MAIN_STACK_SIZE=1024
CONFIG_SYSTEM_WORKQUEUE_STACK_SIZE=1024

CONFIG_BT_CTLR_RX_BUFFERS=18
CONFIG_BT_CTLR_TX_BUFFERS=19
CONFIG_BT_CTLR_TX_BUFFER_SIZE=251
CONFIG_BT_CTLR_DATA_LENGTH_MAX=251
CONFIG_BT_HCI_CMD_COUNT=20
CONFIG_BT_RX_BUF_COUNT=20
CONFIG_BT_RX_BUF_LEN=258
CONFIG_BT_DISCARDABLE_BUF_SIZE=257
CONFIG_BT_MAX_CONN=24

CONFIG_BT_CTLR_PHY_CODED=y
CONFIG_BT_PHY_UPDATE=y
CONFIG_BT_AUTO_PHY_UPDATE=y
CONFIG_BT_LLL_VENDOR_NORDIC=y
CONFIG_BT_LL_SW_SPLIT=y

I trying to scan and connect a Coded PHY device, which is nRF52-PCA10056 board running the Long Range Demo (Nordic's Solution). I can not reach the board when board is advertising via CODED PHY but I can reach it when it advertising via 1M PHY.

Since automatic PHY update is activated, I tried to change it via hcitool, following command: hcitool cmd 08 31 03 04 04 and I got the following answer at btmon:

< HCI Command: LE Set Default PHY (0x08|0x0031) plen 3    #28 [hci0] 540.203681
        All PHYs preference: 0x03
          No TX PHY preference
          No RX PHY preference
        TX PHYs preference: 0x04
          LE Coded
        RX PHYs preference: 0x04
          LE Coded
> HCI Event: Command Complete (0x0e) plen 4               #29 [hci0] 540.205957
      LE Set Default PHY (0x08|0x0031) ncmd 1
        Status: Success (0x00)

But still, no Coded PHY device cannot scannable via my Zephyr USB HCI Bluetooth controller. 
Any help will appreciated. Thanks in advance.

carlescufi, 2020-09-08:
@kasapreyizz this is not implemented yet. @andrzej-kaczmarek is currently working on it. Please join the #bluetooth channel on our Slack for more details.

closed 2020-09-08 by carlescufi

eriklins, 2022-08-16:
Hello,
came across the same issue today and hence wondering if this has been fixed?
Thanks,
ER!K

carlescufi, 2022-08-22:
> Hello, came across the same issue today and hence wondering if this has been fixed? Thanks, ER!K

@eriklins this should now work.