zephyrproject-rtos/zephyr #28148: mec15xxevb_assy6853:running samples/drivers/espi/ failed

Zhaoningx, 2020-09-08:
Describe the bug
running samples/drivers/espi/, and there are not output,and the result just block here until system hangs.

To Reproduce
Steps to reproduce the behavior:

west build -b mec15xxevb_assy6853 samples/drivers/espi/
~/software/ci/tools/usbgpio_ftdi/pre_script.sh
west flash
~/software/ci/tools/usbgpio_ftdi/post_script.sh
see error

*** Booting Zephyr OS build v2.4.0-rc1-11-g8c3efc7be5fe  ***

Environment (please complete the following information):
OS: Fedora28
Toolchain: Zephyr-sdk-0.11.3
Commit ID: g8c3efc7be5fe

maksimmasalski, 2020-09-08:
Yes, test should work on the Microchip, because in the its datasheet https://github.com/MicrochipTech/CPGZephyrDocs/blob/master/MEC1501/MEC1501_Datasheet.pdf page 165 has a description about eSPI.

albertofloyd, 2020-09-11:
@maksimmasalski  this is supported in MEC15 chips but similar to PECI, I2C, eSPI bus testing requires another device to act as eSPI host. 
While this sample can be run in EVB the HW connection to a eSPI host (Intel RVP) is not documented/supported.

Suggest to mark as not supported in that HW only in Modular card (which setup is documented)

nashif, 2020-09-11:
need to change the yaml for this sample file to get excluded. @maksimmasalski  can you please take care of that?

maksimmasalski, 2020-09-15:
> need to change the yaml for this sample file to get excluded. @maksimmasalski can you please take care of that?

I will do that

PR #28407 (2020-09-17): samples: exclude mec15xxevb_assy6853 in espi sample when sanitycheck
ESPI is supported in MEC15 chips but similar to PECI, I2C, eSPI bus
testing requires another device to act as eSPI host.
While this sample can be run in EVB the HW connection to a eSPI host
(Intel RVP) is not documented/supported.

Mark as not supported in that HW only in Modular card
(which setup is documented)

Also remove duplicate definition of the "depends_on: espi"
in the end of the sample.yaml file

Fixes #28148
closed 2020-09-17 by MaureenHelm