zephyrproject-rtos/zephyr #28150: mec15xxevb_assy6853:running samples/boards/mec15xxevb_assy6853/power_management/ failed

Zhaoningx, 2020-09-08:
Describe the bug
running samples/boards/mec15xxevb_assy6853/power_management/, and the result just keep looping until the system hangs.

To Reproduce
Steps to reproduce the behavior:

west build -b mec15xxevb_assy6853 samples/boards/mec15xxevb_assy6853/power_management/
~/software/ci/tools/usbgpio_ftdi/pre_script.sh
west flash
~/software/ci/tools/usbgpio_ftdi/post_script.sh
see error

*** Booting Zephyr OS build v2.4.0-rc1-11-g8c3efc7be5fe  ***
[00:00:00.005,279] <wrn> pwrmgmt_test: PM single-thread test started for cycles: 5, logging: 0
[00:00:00.005,279] <inf> pwrmgmt_test: About to enter light sleep
Wake from Light Sleep
PM >
[00:00:01.507,293] <inf> pwrmgmt_test: About to enter deep Sleep
[00:00:01.507,843] <inf> pwrmgmt_test: PM sleep entry latency 0.000 seconds
Wake from Deep Sleep
PM <
Wake from Light Sleep
PM >
[00:00:04.610,229] <inf> pwrmgmt_test: PM sleep residency 3.103 seconds
[00:00:04.610,229] <inf> pwrmgmt_test: About to enter light sleep
[00:00:06.112,243] <inf> pwrmgmt_test: About to enter deep Sleep
[00:00:06.730,072] <inf> pwrmgmt_test: PM sleep entry latency 0.618 seconds
[00:00:06.730,072] <wrn> pwrmgmt_test: Sleep entry latency is higher than expected
Wake from Deep Sleep
PM <
Wake from Light Sleep
PM >
[00:00:09.215,179] <inf> pwrmgmt_test: PM sleep residency 2.485 seconds
[00:00:09.215,179] <inf> pwrmgmt_test: About to enter light sleep
[00:00:10.717,193] <inf> pwrmgmt_test: About to enter deep Sleep
[00:00:11.335,021] <inf> pwrmgmt_test: PM sleep entry latency 0.617 seconds
[00:00:11.335,021] <wrn> pwrmgmt_test: Sleep entry latency is higher than expected
Wake from Deep Sleep
PM <
Wake from Light Sleep
PM >
[00:00:13.820,129] <inf> pwrmgmt_test: PM sleep residency 2.486 seconds
[00:00:13.820,129] <inf> pwrmgmt_test: About to enter light sleep
[00:00:15.322,143] <inf> pwrmgmt_test: About to enter deep Sleep
[00:00:15.939,971] <inf> pwrmgmt_test: PM sleep entry latency 0.617 seconds
[00:00:15.939,971] <wrn> pwrmgmt_test: Sleep entry latency is higher than expected
Wake from Deep Sleep
PM <
Wake from Light Sleep
PM >
.
.
.

Environment (please complete the following information):
OS: Fedora28
Toolchain: Zephyr-sdk-0.11.3
Commit ID: g8c3efc7be5fe

MaureenHelm, 2020-09-08:
cc: @scottwcpg

dcpleung, 2020-09-14:
@albertofloyd I tried going back to the commit where the warning messages were introduced, and the sleep latency is about 0.650 seconds (which is higher then the 500ms threshold). Also I got `PM state[0]` counter being 6 instead 5. Just wondering if there is any special setup required for this this test.

albertofloyd, 2020-09-14:
@dcpleung no special HW setup (other than debug pins indicated in the test case)
That discrepancy you observed is why I added the counters, intermittently the entry and exit do not match across kernel versions.
Regarding the latency that's a limitation on the system which sleep entry time is unbounded, even if all threads are stopped.
I filed this as an enhancement to guarantee a bounded time. https://github.com/zephyrproject-rtos/zephyr/issues/26889

dcpleung, 2020-09-14:
Regarding the sample (ignoring the latency and cycle counts), this is expected behavior. It runs 4 sets of scenarios, each repeating 5 times. So you will see repeating messages. After all these, it simply idles with nothing to do. So closing this as this is not a bug.

closed 2020-09-14 by dcpleung