zephyrproject-rtos/zephyr #28152: frdm_k64f: running samples/subsys/canbus/canopen/ failed

Zhaoningx, 2020-09-08:
Describe the bug
running samples/subsys/canbus/canopen/ failed, and show me time out.

To Reproduce
Steps to reproduce the behavior:

west build -b frdm_k64f samples/subsys/canbus/canopen/ -DCONFIG_BOOTLOADER_MCUBOOT=y
west flash

see error
�*** Booting Zephyr OS build v2.4.0-rc1-8-g27e1fd69f987  ***
Start sending data
ReTX complete cb [-2]
Error while sending data to ID 384 [-2]
ceiving error [-14]
Got 0 bytes in total
Receiving erreor [-14]
[00:00:02.007,000] <err> isotp: Reception of next FC has timed out
[00:00:02.007,000] <err> isotp: Reception of next FC has timed out
TX complete cb [-2]
TX complete cb [-2]
Error while sending data to ID 384 [-2]
Receiving error [-14]
Got 0 bytes in total
Receiving erreor [-14]
[00:00:18.050,000] <err> isotp: Reception of next FC has timed out
[00:00:18.050,000] <err> isotp: Reception of next FC has timed out
TX complete cb [-2]
Error while sending data to ID 384 [-2]
Receiving error [-14]
Got 0 bytes in total
Receiving erreor [-14]
[00:00:20.055,000] <err> isotp: Reception of next FC has timed out
[00:00:20.055,000] <err> isotp: Reception of next FC has timed out
TX complete cb [-2]
Error while sending data to ID 384 [-2]
Receiving error [-14]
Got 0 bytes in total
Receiving erreor [-14]
[00:00:22.061,000] <err> isotp: Reception of next FC has timed out
[00:00:22.061,000] <err> isotp: Reception of next FC has timed out
TX complete cb [-2]

.
.
.

Environment (please complete the following information):
OS: Fedora28
Toolchain: Zephyr-sdk-0.11.3
Commit ID: g27e1fd69f987

henrikbrixandersen, 2020-09-08:
Your west command refers to the CANopen sample, but your console output shows output from the ISO-TP sample. Did you flash the board correctly?

Zhaoningx, 2020-09-09:
I'm not familiar with the k64f board, I just ran commands '**west build -b frdm_k64f samples/subsys/canbus/canopen/ -DCONFIG_BOOTLOADER_MCUBOOT=y**' and the ran '**west flash**', and as you see, the failure will be reported, so do you think that is there right?

Zhaoningx, 2020-09-09:
Hello @henrikbrixandersen , I'm not familiar with the test case, I just ran commands 'west build -b frdm_k64f samples/subsys/canbus/canopen/ -DCONFIG_BOOTLOADER_MCUBOOT=y' and then ran 'west flash', and as you see, the failure will be reported, so do you think that is this right?

henrikbrixandersen, 2020-09-09:
@Zhaoningx So you ran those commands manually and not via sanitycheck?

The output, you have posted, cannot come from the `samples/subsys/canbus/canopen/` sample. It looks like it is from the `samples/subsys/canbus/isotp` sample.

Zhaoningx, 2020-09-09:
Let me check.

Zhaoningx, 2020-09-09:
Hi @henrikbrixandersen , I double checked it just now, the output is normal, and as follow:

*** Booting Zephyr OS build v2.4.0-rc1-16-g11f44c11aac5  ***
[00:00:00.006,000] <inf> fs_nvs: 2 Sectors of 4096 bytes
[00:00:00.006,000] <inf> fs_nvs: alloc wra: 0, ef0
[00:00:00.006,000] <inf> fs_nvs: data wra: 0, 1f8
[00:00:00.006,000] <dbg> canopen_storage.canopen_settings_set: restored object dictionary EEPROM entries
[00:00:00.006,000] <dbg> canopen_driver.CO_CANmodule_init: rxSize = 13, txSize = 9
[00:00:00.007,000] <inf> app: CANopen stack initialized

Thanks for your comments. I'll close this issue later.

closed 2020-09-09 by Zhaoningx