zephyrproject-rtos/zephyr #28153: reel_board: running samples/subsys/shell/fs/ failed

Zhaoningx, 2020-09-08:
Describe the bug
running samples/subsys/shell/fs/, and the result just keep waiting for some inputs until the system hangs.
And do you all think this test case shouldn't be ran at reel_board in daily report.

To Reproduce
Steps to reproduce the behavior:

west build -b reel_board samples/subsys/shell/fs/
west flash

see error

*** Booting Zephyr OS build v2.4.0-rc1-8-g27e1fd69f987  ***

uart:~$ 

Environment (please complete the following information):
OS: Fedora28
Toolchain: Zephyr-sdk-0.11.3
Commit ID: g27e1fd69f987

nashif, 2020-09-08:
Not sure what exactly fails here, this is a sample that expects input, so it needs a harness which would exclude it from automated testing.

pabigot, 2020-09-08:
```
uart:~$ help
Please press the <Tab> button to see all available commands.
You can also use the <Tab> button to prompt or auto-complete all commands or its subcommands.
You can try to call commands with <-h> or <--help> parameter for more information.
Shell supports following meta-keys:
Ctrl+a, Ctrl+b, Ctrl+c, Ctrl+d, Ctrl+e, Ctrl+f, Ctrl+k, Ctrl+l, Ctrl+n, Ctrl+p, Ctrl+u, Ctrl+w
Alt+b, Alt+f.
Please refer to shell documentation for more details.
uart:~
```
A harness for the above command and some of its output would cover this.

nashif, 2020-09-08:
not a reel_board issue

Zhaoningx, 2020-09-09:
Hi @nashif @pabigot , I totally agree with you all, but you know, daily testing is just testing automatically, and when running this test case, it will be blocked here 'uart:~$ ' to wait input until time out. And it will be reported as a failure in the daily report. Is there OK? And could you all provide a solution to solve this problem?

Zhaoningx, 2020-09-09:
BTW, When using command '**sanitycheck**' to run this test case, human-computer interaction cannot be performed and report a timeout failure directly.

PR #28290 (2020-09-10): samples: shell: fs: require keyboard harness
The test for this sample requires special setup and keyboard
interactivity.

Fixes #28153
commit 994374ec (2020-09-10): samples: shell: fs: require keyboard harness

The test for this sample requires special setup and keyboard
interactivity.

Fixes #28153
closed 2020-09-10 by MaureenHelm